esp8266/Arduino #3230: Disabling RX and TX on hardware UART ( D9 and D10)

bpham817, 2017-05-10:
This is not in my league. I'd like to configure D9 and D10 as digital inputs. 
i need to disable RX and TX on hardware UART pin D9 and D10. Can anyone show me how to do that?

### Basic Infos

#### Hardware
Hardware:			ESP-12E v1.0
Core Version:      	2.3.0

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

merlokk, 2017-05-12:
#define RX_PIN 3   // GPIO3
#define TX_PIN 1   // GPIO1

void setup() {
// dont Serial.begin(74880)!!!!!!!!!!!!!!!!!!!
  pinMode(RX_PIN, INPUT);    
  pinMode(TX_PIN, INPUT);   
}

bpham817, 2017-05-12:
I did that but then I am losing my capability to debug my program through the Serial Monitor because the Serial.begin(74880) was commented out. That's bad.

is there a way to keep the Serial Console enabled for debugging purposes while keeping both pins as INPUT.

JamesGKent, 2017-05-12:
well no?
the serial console is connected to those pins, to use the serial console at least one of those pins must be an output to send data to the computer.
you could use SoftwareSerial to output your debugging info on some other pins... but you can't have one pin as both serial output and digital input.

merlokk, 2017-05-12:
Put here Serial1.begin(74880)

debug via Serial1.println("debug here");

bpham817, 2017-05-12:
ok i will look into debugging via telnet. I am running out of pins. Using Serial1 will require a USB to serial adapter. this is what I've heard so far from the crowd.
https://www.sparkfun.com/products/12977

is there a library out there that can be using for debugging over UDP/IP

bpham817, 2017-05-25:
ok. i've got a chance to look into FullWithCommands example using xlogger library. I have the sketched uploaded over OTA. 
at least i can see two radio buttons choose file and Update. i can upload the binary successfully. I can ping to the nodemcu.
how do i test the FullWithCommands example?

merlokk, 2017-05-26:
Connect to it via telnet. (I like "putty"  client)

bpham817, 2017-06-01:
ok got it. i like "putty" too.

Based on the existing codes, i've added and extended the codes for my own usage as following:

Added:
 #include <Ticker.h>
Ticker broadCastMsg;

from the setup() function, added this line at the end of the routine:
broadCastMsg.attach(5, SndBCastMsg);

added SndBCastMsg function:
void SndBCastMsg()
{
	DEBUG_PRINTLN(F("Starting..."));
	DEBUG_PRINT(F("Hardware id:"));
	DEBUG_PRINTLN(HARDWARE_ID);
	DEBUG_PRINT(F("MAC address: "));
	DEBUG_PRINTLN(WiFi.macAddress());
}

from the loop(), commented out some log messages...

used Putty to connect via Telnet. The debug msg from SndBcastMsg did show up on the Telnet session few times then it looks like the software crashed.

i am wondering if there is any race condition or an overrun buffer causing the software to crash?

bpham817, 2017-06-01:
error from PuTTY, fatal Error: Network error: Software caused connection abort

merlokk, 2017-06-08:
May you put somethere full sketch?

bpham817, 2017-06-08:
#include <xlogger.h>
#include <ArduinoOTA.h>
#include <Ticker.h>

#define               PROGRAM_VERSION   "1.00"

#define DEBUG

#ifdef DEBUG
  // log level: info
  #define             DEBUG_PRINT(...)    logger.print(__VA_ARGS__)
  #define             DEBUG_PRINTLN(...)  logger.println(__VA_ARGS__)
  // log level: warning
  #define             DEBUG_WPRINT(...)   logger.print(llWarning, __VA_ARGS__)
  #define             DEBUG_WPRINTLN(...) logger.println(llWarning, __VA_ARGS__)
  // log level: error
  #define             DEBUG_EPRINT(...)   logger.print(llError, __VA_ARGS__)
  #define             DEBUG_EPRINTLN(...) logger.println(llError, __VA_ARGS__)
#else
  #define             DEBUG_PRINT(...)
  #define             DEBUG_PRINTLN(...)
  #define             DEBUG_WPRINT(...)
  #define             DEBUG_WPRINTLN(...)
  #define             DEBUG_EPRINT(...)
  #define             DEBUG_EPRINTLN(...)
#endif

xLogger               logger;
char                  HARDWARE_ID[7] = {0};
Ticker broadCastMsg;

///////////////////////////////////////////////////////////////////////////
//   NTP
///////////////////////////////////////////////////////////////////////////
boolean syncEventTriggered = false;   // True if a time even has been triggered
NTPSyncEvent_t ntpEvent;              // Last triggered event

void processSyncEvent(NTPSyncEvent_t ntpEvent) {
  if (ntpEvent) {
    DEBUG_EPRINT(F("NTP sync error: "));
    switch (noResponse) {
      case noResponse: DEBUG_EPRINTLN(F("NTP server not reachable")); break;
      case invalidAddress: DEBUG_EPRINTLN(F("Invalid NTP server address")); break;
      default: DEBUG_EPRINTLN(F(""));
    }
  } else {
    DEBUG_PRINT(F("Got NTP time: "));
    DEBUG_PRINTLN(NTP.getTimeDateString(NTP.getLastNTPSync()));
  }
}

///////////////////////////////////////////////////////////////////////////
//   ESP
///////////////////////////////////////////////////////////////////////////
/*
  Function called to restart the switch
*/
void restart() {
  DEBUG_PRINTLN(F("Restart..."));
  ESP.reset();
  delay(1000);
}

/*
  Function called to reset the configuration of the switch
*/
void reset() {
  DEBUG_PRINTLN(F("Reset..."));
  WiFi.disconnect();
  delay(1000);
[... 133 lines cut ...]
  DEBUG_PRINTLN(F("Debug info level"));
  DEBUG_WPRINTLN(F("Debug warning level"));
  DEBUG_EPRINTLN(F("Debug error level"));
#endif

  // time
  DEBUG_PRINT("NTP time: ");
  DEBUG_PRINTLN((timeStatus() == timeSet)? llInfo:llError, NTP.getTimeDateString());

  for(int i = 0; i < 10; i ++)
  {
    // NTP
    if (syncEventTriggered) {
      processSyncEvent(ntpEvent);
      syncEventTriggered = false;
    }

    delay(500);
  }

  logger.handle();
  ArduinoOTA.handle();
  yield();

}

bpham817, 2017-06-08:
The sketch is based on 'FullWithCommands" example

merlokk, 2017-06-21:
It seems that TCP stack have strange behavier if it calls inside the interrupt.
Its not recommended to call complex functions from interrupt like TCP or serial send.

http://www.switchdoc.com/2015/10/iot-esp8266-timer-tutorial-arduino-ide/#Comments
Look at the end of article.

I will look and maybe ill fix it, but always.... its not recommended to call from ISR (Interrupt Service Routine).

merlokk, 2017-06-22:
It hangs here
```
WiFiClient telnetClient;
...
telnetClient.print(msg)
```

so( i cant send data to socket from ISR. 
It needs to put lines to RAM and then send them in `handle()`

closed 2017-08-22 by bpham817