esp8266/Arduino #3231: Esp 8266 12e BLOCK

alejandronara, 2017-05-11:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------
### Basic Infos

  #include<ESP8266WiFi.h>;

//a0:20:a6:13:29:bb  => 192.168.0.9
//a0:20:a6:00:fa:37  => 192.168.0.11

  // especificamos el puerto

  int estado             = 0;
int estado_pin_blue = 0;
  int estadoActual     = 0;
  int cambio           = 0;
  int valWifi            = -1;
 // int continua         = 1;

  String valorEstado      = "";
  
  const char* ssid      = "Esecure";
  const char* password    = "proyectonano";

  //const char* ssid      = "fibertel1424";
  //const char* password    = "sebastian1424";

  
String web_cabecera;
String web_pie; 
String web_on;      
String web_off;
        

    
      
String web_respuesta ="";     
      
WiFiServer server(8083);

void setup() {

    web_cabecera =  "HTTP/1.1 200 OK\r\n";
       web_cabecera +="Content-Type: text/html\r\n";
       web_cabecera +="Connection: close\r\n";
       web_cabecera +="\r\n";
       web_cabecera +="<html lang=\"en\">";
     web_cabecera +="<head>";
       web_cabecera +="<meta charset=\"utf-8\"> ";
       web_cabecera +="<title>Control de Sensores - Domonavi</title> ";
      web_cabecera +="<link type=\"image/x-icon\" href=\"www.esecure.com.ar/icon/esecure_favicon.png\" rel=\"icon\" />";
      web_cabecera +="<script type=\"text/javascript\" ></script>"; 
    web_cabecera +="</head>";
    web_cabecera +="<body id=\"page1\" oonload=\"alert('hola fabio')\" style=\"background-color:#000000\">";
      web_cabecera +="<table border=\"0\">";
        web_cabecera +="<tr height=\"150px\">";
          web_cabecera +="<td colspan=\"3\">";
          web_cabecera +="<span style=\"font-size:50px;color:white;\">DOMONAVI - <span  style=\"font-size:40px;font-style: italic;\">Domótica al alcance de tus manos</span></span>";
          web_cabecera +="</td>";
        web_cabecera +="</tr>";
        web_cabecera +="<tr>";
          web_cabecera +="<td  colspan=\"3\"><span style=\"font-size:60px;color:white;border-bottom:1pt solid white;\">Estado de Sensores</span>";
          web_cabecera +="</td>";
        web_cabecera +="</tr>";
        web_cabecera +="<tr height=\"150px\">";
          web_cabecera +="<td>";
          web_cabecera +="<span style=\"font-size:40px;color:white;\">01 - Sala de Estar</span>";
         web_cabecera +="</td>";

      web_pie = "<tr>";
      web_pie += "<td colspan=\"3\">";
      web_pie +=    "<span style=\"font-size:36px;color:white;\">";
          web_pie +="Domonavi - Todos los derechos reservados.";
          web_pie +="</span>";
          web_pie +="</td>";
        web_pie +="</tr>";
        web_pie +="<table>";
      web_pie +="</body>";
      web_pie +="</html>";        

      web_on = "<td>";
          web_on +="[image] <span style=\"font-size:35px;color:white;\">[encendido]</span>"; 
        web_on +="</td>";
        web_on +="<td>";
          web_on +="<a href=\"http://190.19.211.211:8083/gpio/0\">";
            web_on +="<button> <span style=\"font-size:35px;ccolor:white;\">Apagar</span>  </button>";
          web_on +="</a>";
        web_on +="</td>";      

    web_off = "<td>";
       web_off += "[image] <span style=\"font-size:35px;color:white;\">[apagado]</span>";
       web_off += "</td>";
        web_off += "<td>";
         web_off += "<a href=\"http://190.19.211.211:8083/gpio/1\">";
          web_off +="<button> <span style=\"font-size:35px;ccolor:white;\">Encender</span>  </button>";
          web_off +="</a>";
        web_off +="</td>";
        
  pinMode(5, OUTPUT);
  pinMode(14, INPUT);

 //pinMode(16, OUTPUT); //blue

  digitalWrite(5, 0);
  
  delay(3000);
  Serial.begin(115200);
  
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  
  Serial.println("");
  Serial.println("WiFi connected");
  // inciamos el servidor web
  server.begin();
  Serial.println("Server started");
  // imprimimos la direccion IP
  Serial.println(WiFi.localIP());

}

void loop() {
   //delay(50); 
   cambio = 0;

   estado = digitalRead(14);
  if(estado == 1){
    
    if(estadoActual == 1){
      estadoActual  = 0;
    } else {
      estadoActual  = 1;
    }
    cambio = 1;     
    delay(200);
  }

  delay(200);

    WiFiClient client = server.available();
    if (client) {

         web_respuesta  ="";      
         while(!client.available()){
          delay(1);
         } 
         
         String req = client.readStringUntil('\r');
         
         //Serial.println("Client disonnected "+req);
         
         Serial.println(req);
         client.flush();
         valWifi = -1;
        
    if (req.indexOf("/gpio/0") != -1){
           valWifi = 0;
        } else if (req.indexOf("/gpio/1") != -1){
           valWifi = 1;
        }
     
        if(valWifi != -1){
      if(estadoActual != valWifi ){
        cambio = 1 ;
      } else {
        cambio = 0 ;
      }
      estadoActual = valWifi;
      
     
      
      if(valWifi == 1){
        web_respuesta =  web_cabecera;
        web_respuesta +=  web_on; 
        web_respuesta +=  web_pie; 
        
      } else { // = 0

        web_respuesta =  web_cabecera;
        web_respuesta +=  web_off; 
        web_respuesta +=  web_pie; 
        
      }
      
      
    } else {
    
      if(estadoActual == 1){
        web_respuesta =  web_cabecera;
        web_respuesta +=  web_on; 
        web_respuesta +=  web_pie; 
      } else {
        web_respuesta =  web_cabecera;
        web_respuesta +=  web_off; 
        web_respuesta +=  web_pie;
      }
    } 
    
    client.print(web_respuesta);
    delay(1);
    client.stop();
    //client = false;
  }
      
  if(cambio == 1){
    digitalWrite(5,estadoActual);
  }
}

#### Hardware
Hardware:			ESP-12e
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			qio0
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

Hello my friends, this is my first post.
I have a problem with ESP8266 12E, before 1,2 or "n" hours block and i need to reset to work again.
That is very frustating and i not know what else to do.
Some ideas?

[image]

igrr, 2017-05-15:
The sketch you have posted has a few issues:

- use of `String`s to build the response causes significant heap fragmentation. Consider placing the web page contents into PROGMEM.
- your implementation of HTTP server doesn't wait for the remote client to close the connection. Please use ESP8266WebServer library to build an HTTP server instead.

closed 2017-05-15 by igrr

alejandronara, 2017-05-16:
Thanks Ivan.