esp8266/Arduino #3232: ESP8266 hangs with levelshifter on pins D5,6,7,8

I-Connect, 2017-05-11:
Hi,

I want to connect an NodeMCU v3 to a arduino mega using SPI.

Without anything connected I can upload a blink sketch using Arduino IDE and it works.

As MEGA is 5V I added a level shifter and connected 3.3V GND and ESP pins D5,6,7 and 8 to a level shifter (I did not connect the high voltage side yet)

Now I cannot upload the blink sketch anymore . I see it "trying to connect", "an error occurred while uploading", "warning espcomm_sync failed" "espcomm_open failed" "espcomm_upload_mem failed".

When I disconnect D8 all works fine again sketch uploads and led blinks. When I now connect D8 again while sketch is running blink keeps on working. When I reset or try to upload again it fails... 

(in rest status the level shifter pulls the pin high would this be an issue for uploading or on reset?)

Any Ideas?

Thx,

igrr, 2017-05-12:
IO15 of the ESP8266 is a bootstrapping pin, and should be at low logic level on reset. Otherwise ESP8266 will not run the application.

I-Connect, 2017-05-12:
thx, good to know. 

Although it is very inconvenient as you never know what the status is of the CS pin when the ESP is slave and there is another MCU controlling it...

-Any tips on how to establish bootstrapping pins to be always low on startup?
-Out of interest... What exactly is the function of a bootstrapping pin?
-Are there more pins that have a logic level requirement at startup?

thx,
Jeroen

igrr, 2017-05-15:
This table lists the bootstrapping pins and boot modes:
https://github.com/esp8266/Arduino/blob/master/doc/boards.rst#boot-messages-and-modes

closed 2017-05-15 by igrr

I-Connect, 2017-05-15:
thx.

Am I correct that GPIO 00, 02 and 15 cannot be used as input as they need a predetermined status at boot? As you cannot determine the input status at boot of the ESP (without building electronics around it).

And if this is true then I cannot use GPIO15 as SPI CS or connected to a serial device? (without building electronics around it).

I would appreciate a confirmation on this or a tip how to circumvent it. I am currently designing/building a solution  :-)

Thx, Jeroen

igrr, 2017-05-15:
If you can't force the state of these pins at boot time (perhaps using external circuitry), then it is not safe to use these pins as inputs.

One possible way to circumvent this is to use an external buffer between the signal source and these input pins. These can be obtained as ICs (74xx125/126, if memory serves), or built from discrete transistors. You would need to reserve a pin to control this buffer though.

I-Connect, 2017-05-15:
thx for the confirmation!!

ah tri-state buffer.... good tip. thx!!
Would I need to do anything specific in code to make sure this "buffer control pin" goes low on boot, and would this also work if the esp hangs due to whatever reason and reset by a WDT?

(sorry for all the questions but I never ran into this with Arduino's)

igrr, 2017-05-15:
Yes, i suppose you need something in your code to control the buffer. You likely need to enable this pin as output in your code, and have an external pulldown on it (assuming the buffer enable signal is active high). Then if WDT happens, the pin will go low. The other question whether it will go low soon enough so that the buffer is hi-Z before ESP32 latches the input values on bootstrapping pins. I don't have an answer for this one, you may have to experiment a bit and see if that works.