esp8266/Arduino #3233: SetCAcert and LoadCACert arent accessible

robfullagar, 2017-05-11:
Hi 
I have downloaded the libraries to my arduino IDE 1.8.2 using the URL in preferences as indicated.

I see that in wifisecure.h and wifisecure.cpp, setCACert and LoadCACert are declared in the git hub repo, but not in what is downloaded, along with verifycertchain?

All are in the documentation but arent present in the download.

I am attemting to setup mutual TLS auth, with certs/keys on the SPIFFS, rather than the server cert fingerprint hardcoded, to allow certs to be revoked etc and allowing devices to to connect to multiple servers by just replacing the certs.

Thanks in advance.

Rob

#### Hardware
Hardware:			Windows 10, HP Zbook i7 32gb RAM, 2TB disk
Core Version:      	Esp wroom 02

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512MB/64k SPIFFS
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz?
Upload Using:		 SERIAL
Reset Method:		ck / nodemcu

messages here
```

JamesGKent, 2017-05-12:
to use wifisecure you need to download the git version rather than the 2.3.0 version currently available directly in the arduino IDE. follow the instructions for installing from git

robfullagar, 2017-05-13:
James,

Thanks for this, I managed to adopt the GIT method with some tweaks for my windows Arduino IDE. I now have access to the functionality I require. Many Thanks.

Rob

JamesGKent, 2017-05-13:
if you have what you need can you close this issue?
If you have any further issues you can always create a new issue or reopen this one
thanks

closed 2017-05-14 by robfullagar