esp8266/Arduino #3235: SPIFFS 3M 1M and compuler directive needed

WheresWally, 2017-05-12:
I am using NodeMCU's with Arduino core. Initially I picked 3M memory model, because I could,
I developed several projects and now I am using OTA and I have crashed "reboot-loop" 2 boards,
Now I want to swap to using 1M to allow more room for OTA BUT the file system written by 3M model isn't readable when recompiled with 1M . And that is understandable.
Now I have to be careful when fixing bugs on old projects to be sure not to use the wrong memory model for SPIFFS,

To avoid this problem, is there a compiler directive or similar method to ensure that can be in the code overrides the setting in the IDE?

 Cheers
Terry

igrr, 2017-05-13:
There isn't a directive which would allow you to override SPIFFS size setting, but you can check the size at run time like this:

https://github.com/esp8266/Arduino/blob/master/cores/esp8266/spiffs_api.cpp#L113-L119

Alternatively, if you manage to inject a `NO_GLOBAL_SPIFFS` define into CPPFLAGS (by modifying boards.txt or platform.txt), global SPIFFS object will not be created at all, and you can create it in your sketch (similar to the way it is done in the link above).

WheresWally, 2017-05-13:
Thanks for the info. I will see if I can use those variables to check which memory model is being used at least.

closed 2017-05-13 by WheresWally