esp8266/Arduino #3236: ESP8266 Library 2.3.0 does not work with Stino2017 for SublimeText

elbrinke, 2017-05-12:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ALL ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

When I select version 2.3.0 on Stino Arduino IDE for any ESP project, it fails to compile. Only version 2.1.0 works well. 

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```

[Build] /Users/eddy/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/examples/WiFiClientEvents...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[1.4%] Compiling WiFiClientEvents.ino.cpp...
[2.9%] Compiling ESP8266WiFi.cpp...
[4.3%] Compiling ESP8266WiFiAP.cpp...
[5.7%] Compiling ESP8266WiFiGeneric.cpp...
[7.1%] Compiling ESP8266WiFiMulti.cpp...
[8.6%] Compiling ESP8266WiFiScan.cpp...
[10.0%] Compiling ESP8266WiFiSTA.cpp...
[11.4%] Compiling WiFiClient.cpp...
[12.9%] Compiling WiFiClientSecure.cpp...
[14.3%] Compiling WiFiServer.cpp...
[15.7%] Compiling WiFiUdp.cpp...
[17.1%] Compiling server_drv.cpp...
[18.6%] Compiling spi_drv.cpp...
[20.0%] Compiling wifi_drv.cpp...
[21.4%] Compiling WiFi.cpp...
[22.9%] Compiling WiFiClient.cpp...
[24.3%] Compiling WiFiServer.cpp...
[25.7%] Compiling WiFiUdp.cpp...
In file included from /applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:29:0:
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.h:21:7: error: redefinition of 'class WiFiUDP'
 class WiFiUDP:public UDP
       ^
In file included from /applications/Arduino.app/Contents/Java/libraries/WiFi/src/utility/wifi_drv.h:22:0,
                 from /applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:26:
/Users/eddy/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/WiFiUdp.h:32:7: error: previous definition of 'class WiFiUDP'
 class WiFiUDP : public UDP, public SList<WiFiUDP> {
       ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In constructor 'WiFiUDP::WiFiUDP()':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:35:22: error: class 'WiFiUDP' does not have any field named '_sock'
 WiFiUDP::WiFiUDP() : _sock(NO_SOCKET_AVAIL) {}
                      ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual uint8_t WiFiUDP::begin(uint16_t)':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:45:9: error: '_sock' was not declared in this scope
         _sock = sock;
         ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:46:9: error: '_port' was not declared in this scope
         _port = port;
         ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::available()':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:56:7: error: '_sock' was not declared in this scope
   if (_sock != NO_SOCKET_AVAIL)
       ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual void WiFiUDP::stop()':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:66:8: error: '_sock' was not declared in this scope
    if (_sock == NO_SOCKET_AVAIL)
        ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:69:26: error: '_sock' was not declared in this scope
    ServerDrv::stopClient(_sock);
                          ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::beginPacket(IPAddress, uint16_t)':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:88:7: error: '_sock' was not declared in this scope
   if (_sock == NO_SOCKET_AVAIL)
       ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:90:7: error: '_sock' was not declared in this scope
   if (_sock != NO_SOCKET_AVAIL)
       ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::endPacket()':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:101:32: error: '_sock' was not declared in this scope
  return ServerDrv::sendUdpData(_sock);
                                ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual size_t WiFiUDP::write(const uint8_t*, size_t)':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:111:27: error: '_sock' was not declared in this scope
  ServerDrv::insertDataBuf(_sock, buffer, size);
                           ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::read()':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:125:23: error: '_sock' was not declared in this scope
    ServerDrv::getData(_sock, &b);
                       ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::read(unsigned char*, size_t)':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:137:31: error: '_sock' was not declared in this scope
    if (!ServerDrv::getDataBuf(_sock, buffer, &size))
                               ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::peek()':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:152:22: error: '_sock' was not declared in this scope
   ServerDrv::getData(_sock, &b, 1);
                      ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual IPAddress WiFiUDP::remoteIP()':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:166:25: error: '_sock' was not declared in this scope
  WiFiDrv::getRemoteData(_sock, _remoteIp, _remotePort);
                         ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual uint16_t WiFiUDP::remotePort()':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:176:25: error: '_sock' was not declared in this scope
  WiFiDrv::getRemoteData(_sock, _remoteIp, _remotePort);
                         ^
[Build] Error occurred.
```

igrr, 2017-05-15:
As you see from the log, the IDE is trying to compile Arduino built-in WiFi library in addition to the ESP8266WiFi library. This isn't the correct behavior, please report that to STino developers.

closed 2017-05-15 by igrr

VijayaniX, 2018-04-30:
I had the same problem and subsequently i found the solution at (https://github.com/Robot-Will/Stino/issues/445)