esp8266/Arduino #3238: SCL/SDA reversed

DrRob, 2017-05-13:
The pin diagram on this page: https://github.com/esp8266/Arduino/blob/master/doc/reference.md shows the SCL and SDA pins swapped compared to the info in the "I2C (Wire library)" section on this page: https://github.com/esp8266/Arduino/blob/master/doc/libraries.md

I'd been trying for hours to get a TSL2561 sensor to work - I tried three different TSL2561 libraries, all with the same result (returning 65535 for both values).  The sensor worked fine when I was using NodeMCU.

Turns out I need to edit the TSL261 library to initialise Wire differently:
  Wire.begin(5, 4);
and then it works.

I'm using esp8266 board support version 2.3.0, with Generic ESP8266 for an ESP 12E on a home-made board, which is compatible with NodeMCU (at least as far as the SCL and SDA pins is concerned).

Rob.

igrr, 2017-05-15:
Wire.begin function takes two arguments, SDA and SCL pin numbers, in the following order:

```
void begin(int sda, int scl);
```

So the correct way of initializing the library with SDA = GPIO4 and SCL = GPIO5 would be `Wire.begin(4, 5);`. If this doesn't work for you, that means that you have GPIO4 and GPIO5 swapped somewhere on the board.

Some ESP12 have been manufactured with an incorrect silkscreen, where GPIOs 4 and 5 have been reversed. If you have designed you home made board looking at an incorrect silk screen, you may very well have such issue.

This is an incorrect pinout:
http://img15.hostingpics.net/pics/790940ESP12SCHEMA.jpg

This is the correct one:
http://www.esp8266.com/wiki/lib/exe/fetch.php?cache=&media=esp-12_pindef.png

closed 2017-05-15 by igrr

wouterds, 2019-01-27:
I have a similar issue, but I can detect the sensor if I scan all i2c addresses. If I hook up the sensor to a nodemcu and run the code on there it works fine though.

I am clueless.

Addidis, 2019-09-07:
This comment comes up if you are googling heltec wifi kit 32 , adafruit 1306 lib  + adafruit gfx lib. 

use adafruit 1306 128x64 i2c  example 
Find this line and change ox3d to 0x3c
if(!display.begin(SSD1306_SWITCHCAPVCC, 0x3d))

then at the start of setup loop add lines

void setup() {
    Wire.begin(4,15);
    pinMode(16,OUTPUT);
    digitalWrite(16,HIGH);   //reset pin must be high for screen to run