esp8266/Arduino #3239: how to fix no member listen

hungpkav, 2017-05-13:
how to fix no member listen
[image]

igrr, 2017-05-15:
As the error message says, SoftwareSerial class on ESP8266 doesn't have member called `listen`. If you have some suggestion about this, please open an issue in SoftwareSerial repository here: https://github.com/plerup/espsoftwareserial

closed 2017-05-15 by igrr

olehs, 2017-05-23:
@igrr EspSoftwareSerial supports listen() method since 3.2.0.
Default version in ESP8266 is 3.1.0. 
Is there a chance to get an updated version?

igrr, 2017-05-23:
Uhm, sure, i didn't know there was a more recent version. In the past @plerup would make a PR to update [the version used](https://github.com/esp8266/Arduino/blob/master/package/build_boards_manager_package.sh#L64). Should I update it to 3.2.3, which is the latest?

olehs, 2017-05-23:
Yes, the latest version is compatible with AVR's one and seems to work fine

plerup, 2017-05-23:
Sorry about that, forgot :grimacing:

bruescor, 2018-05-05:
Good evening guys, I was researching on the softwareserial for nodemcu and came to stop here. I'm new here and I'm starting to program with the module and I had a problem with this listen class. I wonder if you have any library that solves this problem. Thanks in advance.