esp8266/Arduino #3241: ESP8266 12E keeps rebooting (rst cause:2, boot mode:(3,6))

SurtrTch, 2017-05-14:
Hello,
I want to use my ESP8266 12E with a keypad, I wired it this way:
1-----D0
2-----Rx
3-----SD3
4-----D3
5-----D4
6-----D5
7-----D6
8-----D7

I didn't use the D1 and D2 because I want to add an LCD I2C later.

Well, first I tried this code, same wiring it works perfectly:
`#include <Keypad.h>
 
const byte n_rows = 4;
const byte n_cols = 4;
 
char keys[n_rows][n_cols] = {
  {'1','2','3','A'},
  {'4','5','6','B'},
  {'7','8','9','C'},
  {'*','0','#','D'}
};
 
byte colPins[n_rows] = {D3, 10, 03, D0};
byte rowPins[n_cols] = {D7, D6, D5, D4};
 
Keypad myKeypad = Keypad( makeKeymap(keys), rowPins, colPins, n_rows, n_cols); 
 
void setup(){
  Serial.begin(115200);
}
 
void loop(){
  char myKey = myKeypad.getKey();
 
  if (myKey != NULL){
    Serial.print("Key pressed: ");
    Serial.println(myKey);
  }
}`

But later in my project I will need to use another commands and somethings, so I used this one, very similar
`#include <Keypad.h>
 
const byte n_rows = 4;
const byte n_cols = 4;
 
char keys[n_rows][n_cols] = {
  {'1','2','3','A'},
  {'4','5','6','B'},
  {'7','8','9','C'},
  {'*','0','#','D'}
};
 
byte colPins[n_rows] = {D3, 10, 03, D0};
byte rowPins[n_cols] = {D7, D6, D5, D4};
 
Keypad myKeypad = Keypad( makeKeymap(keys), rowPins, colPins, n_rows, n_cols); 
 
void setup(){
  Serial.begin(115200);
}
 
void loop(){
           char keypressed = myKeypad.waitForKey();
         if (keypressed != NO_KEY)
             {
             char c1 = keypressed;
              Serial.println(c1);
              }
}`

But when I upload this on the board, the board starts rebooting, and I get this in the serial monitor
 
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld

I'm powering it from my computer, and as you can see all I did is changing a function and add a variable
Thanks

SurtrTch, 2017-05-14:
Same wiring same all, the problem starts if I change the function "getkey" to "waitForKey" I don't know why, the function waitForKey stops the program and wait until a key is pressed, not like "getkey" that keeps the programme looping

RudyFiero, 2017-05-14:
Include a delay in the routine. That would allow background tasks that need to run. You can not have long blocking code with the ESP8266. It will cause the behavior you see.

SurtrTch, 2017-05-14:
Yes I noticed if I keep pressing buttons from the keypad the board doesn't reboot until I stop for few seconds.
Is there a way to increase that rebooting perdiod, like from 2seconds to 10 or more seconds, it would be great for me if it waited few seconds more before rebooting?
Thanks

WheresWally, 2017-05-14:
If you load the original sketch, is all ok then?

igrr, 2017-05-14:
Pin 10 is used to connect flash memory inside the esp12 module, I doubt it can be used for other purposes.

SurtrTch, 2017-05-14:
As you can see I kept the same wiring, even if I used the pin 10, the first code works fine but the second one keeps rebooting if I don't press buttons quickly, the problem appears if I change the function I mentionned above.

I've read about a reset period that can be modified, but I don't have further information about it. If you know please tell me, thanks.

igrr, 2017-05-14:
Ok, sorry, missed that part.

`Keypad::waitForKey` method in Keypad library is not compatible with the ESP8266 Arduino core because it uses a busy loop. This part of the code library needs to be modified, by inserting a `delay(0);` or `yield();`:
https://github.com/Chris--A/Keypad/blob/master/src/Keypad.cpp#L211-L215

SurtrTch, 2017-05-14:
Oh, I'm a beginner, I didn't understand what you meant...
Should I remove the keypad "waitforkey" function, or should I modify something to be able to use it?
Thanks

WheresWally, 2017-05-14:
The codes are :

0 -> normal startup by power on
1 -> hardware watch dog reset
2 -> software watch dog reset (From an exception)
3 -> software watch dog reset system_restart (Possibly unfed watchdog got angry)
4 -> soft restart (Possibly with a restart command)
5 -> wake up from deep-sleep
Looking at the provided code you can decide from what reason the chip is restarting.

I found that loops need a yield() to be ESP8266 friendly, to feed the watchdog before it triggers.
the line referred to by Igrr ...
`	while( (waitKey = getKey()) == NO_KEY );	// Block everything while waiting for a keypress.
`
most likely should be ...
`	while( (waitKey = getKey()) == NO_KEY ) yield();	// Block everything while waiting for a keypress.
`

igrr, 2017-05-14:
Or rather,
```
while( (waitKey = getKey()) == NO_KEY ) yield(); // DON'T block everything while waiting for a keypress.
```
:)

SurtrTch, 2017-05-14:
If it doesn't block the program, it would be like the function getKey() and this one make a little mess in my program, I really want the program to wait like 5s or more in this step before looping again.

I want the program to wait a key then store it into a char c1, then wait another key and store it into char c2 ... So I can do other things later. If I use the function "getkey" it mess with my program: I can only store one char, because the program doesn't wait. (to explain my little idea).

And how can I modify the library if I want to test your solution.

Thanks guys.

SurtrTch, 2017-05-14:
I downloaded that library by using the "library manager" method, where can I find the cpp file so I can modify it?
Thanks

lights0123, 2017-05-14:
It is in the folder under your sketchbook location, so probably Documents/Arduino/libraries/Keypad/Keypad.cpp.

SurtrTch, 2017-05-14:
No didn't find it there.
I have found that library on the arduino library "Program files - Arduino - Libraries"
Because I tried first time with an Arduino UNO, then when I used the NodeMcu I had that library problem and I downloaded it using the library manager.
"I don't know if both uses the same library, or the NodeMcu ones are saved in another place like wifi libraries ..."

SurtrTch, 2017-05-14:
Hello everyone,
So it seems it works fine,
For everyone having the same problem: I did a copy of the library (so not to ruin it), then I opened the file keypad.cpp with note++, I did the modification proposed by "WheresWally" and "igrr", then I saved the .cpp file as Keypad2 and renamed the keypad.h to keypad2.h, then I kept the same second code mentionned above, now I can store 2 keys as c1 and c2 and manipulate them with using the modified version of "waitForKey" function.

You guys thanks so much you're really helpful.

closed 2017-05-15 by igrr

qwerty12345asdf, 2018-10-03:
If we are calling the yield() function in the "Keypad.cpp" file should we not have some code associated with it too what i mean is 
void yield(){
           //something that the function does//
}
Or should we just call it and save it and hope it works in the final compilation on the arduino ide.
One more final comment : 
how should we call the yield() if the while loop is like this:=
while(some condition){
//some work//
};
Thanks.
Ps. I am facing similar problem while working with a finger print sensor library here https://github.com/sparkfun/Fingerprint_Scanner-TTL

d-a-v, 2018-10-03:
```
while (some condition) {
    yield();
    //some work//
}
```

`yield()` is a "system" function that should be called regularly *if* you are stuck in your `loop()` function.
In the arduino model we should try to not be stuck, then it usually is
```
void loop() {
    // stuff
    if (some condition) {
        //some work//
    }
    // stuff
}
```

You can consider that system is doing this:
```
    setup();
    while (true) {
        loop();
        yield();
    }
```

moorthi5, 2018-12-29:
ets Jan 8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld

 Hello guys ,anyone help me , i have used Esp12e  after uploading iot code  its continuesly rebooting ,not connect mqtt ,what to do? How to solve

pieman64, 2018-12-29:
@moorthi5 there are lots of reasons for a device reset and that's why you must create a NEW issue with all the required details.

texsean, 2019-03-08:
mine was also giving the same error in the serial monitor but kept failing to upload, I was about to give up and just throw it away when I found this page, I fixed it and wanted to share in case someone else has a esp8266 that always fails to download no matter what you try, reboot power down another port hit flash button, nothing worked. sooo.. I hooked a wire to ground and touched every pin (BUT vin and 3.3v) as that would be stupid. this somehow "fixed" my device for some reason. I suspect it was one of  the 2 reserve pins. if anyone knows why this desparate measure fixed my bricked node ecu please chime in!

javadhabibi1500, 2019-06-11:
> Include a delay in the routine. That would allow background tasks that need to run. You can not have long blocking code with the ESP8266. It will cause the behavior you see.

Adding a 1ms delay worked for me.
Thanks

BrunoPedro7, 2020-04-10:
> > Include a delay in the routine. That would allow background tasks that need to run. You can not have long blocking code with the ESP8266. It will cause the behavior you see.
> 
> Adding a 1ms delay worked for me.
> Thanks

Where did you add it?

raspiduino, 2020-07-22:
Just put this into loop():
ESP.wdtFeed(); // feeds the dog
delay(0);

raspiduino, 2020-07-22:
I did it and it worked!

tickietackie, 2021-02-18:
Actually I got this because my baud rate was set wrongly to 9600 instead of 115200 because of copy pasting ...

viveks115, 2022-07-19:
```
void dimmerLampESP8266::begin(DIMMER_MODE_typedef DIMMER_MODE, ON_OFF_typedef ON_OFF)
{
	dimMode[this->current_num] = DIMMER_MODE;
	dimState[this->current_num] = ON_OFF;
	timer_init();
	delay(10);
	ext_int_init();	
}
```
I ADDED A DELAY I BETWEEN TO INITILIZATION IT WORKED FOR ME

JamesDelfini, 2022-09-11:
> delay(0);

What the heck it also works for me

raspiduino, 2022-09-11:
> > delay(0);
> 
> What the heck it also works for me

:))

Idk
Sth just works but can't be explain