esp8266/Arduino #3242: ESP8266 12E Reset Boot Infinite loop

WheresWally, 2017-05-14:
I had asked this question in https://github.com/esp8266/Arduino/issues/1671

Has anyone found a method to recover a 12E module once it gets into an infinite reset boot loop?
I have 5 x 12E's on the same version of a small PCB. The PCB has all the required voltage reg/smoothing/pull-ups/pull-downs needed to make a 12E happy. A MCP1825 3.3V linear reg and caps. 

Now 3 of them work are working 24x7 without an issue in various circuits and applications.
BUT 2 of them just do the error text at 74880 baud as listed under DEBUG below.

The boards are stable while programming or flashing or erasing.

I tried this  [http://www.pratikpanda.com/completely-format-erase-esp8266-flash-memory/](url)
And I do get the **ets_main.c%**  error when there is blank flash.

I tried programming many different sketches via Arduino, all without exception causes the same reset loop.

I flashed the LUA system to them and that also causes the same reset loop.

I tried the AT firmware to and that also causes the reset loop
This seems to be a problem that many others have, but no one has solved, 
It maybe easier to just buy a new one. ;) 

Fresh Out Of Ideas now....
Any help will be appreciated

### Debug Messages

```
ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 2408, room 16 
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0 
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0 
tail 8
chksum 0xd8
csum 0xd8

2nd boot version : 1.6
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
no GPIO select!
jump to run user1 @ 1000

Fatal exception (0): 
epc1=0x40201cdc, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
```

WheresWally, 2017-05-15:
Ok, now a new issue. I was able to re-flash either 12E using the Flash Download Tool 2.3/4.
I saw there is a V3.4.4 of the tool so i tried that. It said comms failed.
So I tried the 2.4 version, it too failed
I swapped to the 2nd failing unit. and it flashed using 2.4.
I tried the 2nd unit with 3.4.4 and comms failed
Tried it again with 2.4 and now comms failed there too.
Now neither unit will flash with either version of the Flash Download Tool.
grrrrrr

I did grab a few screen shots of the debug from the one time the 2,4 tool did work.
EFUSE ERRORS are showing so I will have a look now why

WheresWally, 2017-05-15:
When flashing with the Flash Download Tool 2.3/4 there is a few errors in the DOS box.
I am wondering if some EFUSE is set wrong. And if it is a WRITE ONCE type then nothing can be done for it. yes?
```
EFUSE LOG:
REG0:D4E90001
REG1:4200D107
REG2:6400b000
REG3:005CCF7F
===================
EFUSE NORMAL MODE
===================
EFUSE VAL ERROR...
EFUSE VAL WARNING...
```

bobybobybob, 2017-05-21:
I have reset loop too. With easy code like   blabla = blabla + 1   serial print blabla

But if i upload a code where the esp can connect to wlan and then upload the same simple minimal counter code, it works perfect.

devyte, 2017-09-06:
Given that the error message involves EFUSE errors, that there is concensus on esspressif's bbs that blown fuses means a hardware failure (meaning there is no recovery), and that there is no real documentation available on the efuses, I don't think there is anything we can do in this repository.
Per esspressif, it is very unlikely that efuses can be blown by the user, unless there is ESD or high voltage, or the ESP somehow enters an unknown test mode.
I don't see any possible avenue of investigation to developers here. Given the current number of issues, I am closing this for now.

closed 2017-09-06 by devyte