esp8266/Arduino #3243: ็How to ArduinoIDE on windows can use Iwip with IP-FORWARD?

TridentTD, 2017-05-14:
### ArduinoIDE on Windows with IP-FORWARD

#### Wemos Mini
Hardware:			ESP-12F
Core Version:      	2.3.0

### Description
How to  ArduinoIDE on Windows with IP-FORWARD?

d-a-v, 2017-05-22:
IP-forwarding is usually for routers or cores with at least 2 IP interfaces.
You could enable IP forwarding within lwip, so a client of the AP network could talk with other hosts on the STA side. 
Is this really what you want ?

devyte, 2017-05-22:
I have already tried this with the nodemcu firmware (lua), per
areresearch's instructions. It requires a routing rule in the router, so
that it knows to forward packets destined to clients connected the SOFTAP
to the STA address. A few other minor things also need to be setup, like a
dns forwarder, but it works.

I was able to stream 30mins of netflix through the ESP without a hitch
(router <-> ESP <-> laptop).

This tells me that the ESP is fully capable of acting like a transparent
access point for a wifi extender, or as a real mesh node (e.g. with
B.A.T.M.A.N. or something).

I just wish somebody with the knowhow in lwip and network meshes could take
an interest and implement this.

In addition, I also know that somebody implemented a soft NAT for the ESP.

d-a-v, 2017-05-22:
You could try IP_FORWARD = 1
find tools/sdk/lwip/include/lwipopts.h, change IP_FORWARD to 1
then recompile from tools/sdk/lwip/src/ with `make release`
(before that you should backup tools/sdk/liblwip*.a)

about soft nat, a lwip proper patch is needed (or tell your internet modem to do so :) )

d-a-v, 2017-10-26:
Considering #1924, are we still willing to directly route between AP clients and STA peers ?
As said above, this needs a routing rule in the host network router (unless we do NAT too).

davisonja, 2017-10-26:
Do we have enough RAM for useful NAT?

d-a-v, 2017-10-26:
[This project](https://github.com/martin-ger/esp_wifi_repeater) does NAT

devyte, 2017-10-26:
@NeoCat has a branch with Napt implemented in lwip (I think). It has an api for setting things up.

d-a-v, 2017-10-26:
I had a quick look. It is indeed implemented in lwip, and it seems to be clean.
@NeoCat have you considered to propose your work to lwip guys ?

danielicy, 2018-08-07:
Hi guys. do you have an estimation when will this issue be released? I really need it

bha14ur, 2019-01-29:
> I have already tried this with the nodemcu firmware (lua), per areresearch's instructions. It requires a routing rule in the router, so that it knows to forward packets destined to clients connected [...]
Hello,
You did wifi mesh node, for that what you used?
arduino ide or SDK or other?

devyte, 2019-10-31:
As a general rule, doing custom builds of internal libs like lwip isn't meant for normal users. However, it is ok to support it for advanced users in some manner.
lwip2 currently supports being manually built with different config param than the default, including IP_FORWARD enabled. It is not automatic, and must be done from command line with a git install of the core, but it can be done, and it is documented in a README in the lwip build directory and in @d-a-v 's link layer repo.
Closing this as resolved.

closed 2019-10-31 by devyte