esp8266/Arduino #3244: I am new to Arduino and ESP8266 IDE, and have just downlaode the latest versions. I get numerous warning messages sayin "invalid library found in sub directories. What does this mean and what can I do do fix?

Johnht, 2017-05-14:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

igrr, 2017-05-15:
You have likely not followed the instructions for installing the ESP8266 Arduino core correctly. Please remove the existing installation and try again, this time following the instructions found in https://github.com/esp8266/Arduino/blob/master/README.md#installing-with-boards-manager.

If you have issues, please describe what steps you have taken.

Also please don't place the issue description into issue title. Please read https://help.github.com/articles/creating-an-issue/ about creating issues.

AdamJWood, 2017-05-16:
I had the same issue with an invalid library.  Are you suggesting there is some sort of advice in the installation instructions regarding this??

igrr, 2017-05-16:
If you follow installation instructions exactly, there shouldn't be an issue with 'invalid library'. Most often this happens if one downloads the zip archive of ESP8266 Arduino core and then uses "Import .zip" function in the library menu of the IDE, or extracts the contents of the zip file into Libraries directory in the sketchbook.

AdamJWood, 2017-05-16:
I followed the instructions exactly and I had an issue with an invalid library.  It was an easy fix of simply moving the library out of the folder but it was unrelated to the assumed issue you're referring to.  I updated an old version; it was about a year and a half old.  I think you're assuming this was the first time the Arduino IDE was installed.  Granted, you're probably right in that scenario but IDE updates bring a whole new set of unexpected issues.

igrr, 2017-05-16:
I assumed that the subject was the initial installation, based on "I am new to Arduino and ESP8266 IDE, and have just downlaode the latest version" part of the issue description.

If you had similar issue when updating the IDE, it does deserve to be reported. I wasn't aware that updating the IDE would cause such problems, and this hasn't been reported in the past. If you can provide steps to reproduce the issue, that would be appreciated.

AdamJWood, 2017-05-16:
You're right, that's a fair assessment.  Although, I've personally only written a couple dozen sketches myself and would still consider myself new to Arduino.  But if I had written that title I could see how it could be misleading.

It seems to me this issue is Arduino's incompatibility with some older libraries.  I didn't dig very deep to find the root issue but there was probably some code change Arduino made that broke some previous functionality.  There's probably a changelog somewhere that addresses said changes.

devyte, 2017-09-06:
Given that the original issue here is likely a user error due to being a new user, that the description doesn't make use of the issue template and lacks information, and that there is a lack of response from the orinal poster, I am closing this.

closed 2017-09-06 by devyte