esp8266/Arduino #3246: Arduino as IDE disconnect automatically, when esp8266 upload finished

Caprarde, 2017-05-16:
### Basic Infos

#### Hardware
Hardware:			Sunfounder esp8266 Serial

### Description

I use a Arduino Nano as IDE to flash the esp module, that works good, but when the upload status has reached 100%, the arduino is disconnected from my pc and the red led on the module ist off.ö The blue lamps on the Arduino are still on. When i plugged in the arduino again, the wifi module do nothing. Why disconnected the arduino automatically from my pc when the upload is finished?

### Settings in IDE

Module:  			Generic ESP8266 Module
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, LOW);   // Turn the LED on (Note that LOW is the voltage level
                                    // but actually the LED is on; this is because 
                                    // it is acive low on the ESP-01)
  delay(1000);                      // Wait for a second
  digitalWrite(LED_BUILTIN, HIGH);  // Turn the LED off by making the voltage HIGH
  delay(2000);                      // Wait for two seconds (to demonstrate the active low LED)
}

igrr, 2017-05-17:
Sorry, no idea about Arduino Nano — this issue tracker is about ESP8266 Arduino core. Could it be that you are powering ESP8266 from Arduino Nano onboard voltage regulator? If so, try powering it with an external power supply, with sufficient decoupling close to the ESP8266 module.

devyte, 2017-10-24:
@Caprarde will you be providing more details about your setup?

closed 2017-12-28 by igrr