esp8266/Arduino #3247: can't customize the WifiClient timeout for connect

Zonglin-Li6565, 2017-05-16:
### Basic Infos

#### Hardware
Hardware:		NodeMCUv2, ESP-12E
Core Version:      	?

### Description
The WifiClient connect function hang the program
### Settings in IDE
Doesn't matter

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {
        WiFiClient client;
        IPAddress ipAddr;
        ipAddr.fromString(IPADDR);
        Serial.println("Trying to log");
        if (client.connect(ipAddr, PORT)) {
            client.print(msg);
            Serial.println("Logged");
        }
        // Have some other time critical stuff here. The connect takes too long if the ipaddress doesn't exist. 
}

igrr, 2017-05-17:
This issue is tracked in https://github.com/esp8266/Arduino/issues/2120.

igrr, 2017-05-17:
Also https://github.com/esp8266/Arduino/issues/1420.

closed 2017-05-17 by igrr