esp8266/Arduino #3251: Network Scanner does not work from Ticker

mcmega, 2017-05-17:
Hi All!
Faced a problem.
I check the connection with the router. In the event of a failure, I run the STA+AP mode. I scan the network and try to reconnect when I see my router.
The scan function works fine if it is in a "loop" arduino sketch. But I want to run a scan every 10 seconds. I create the wifiCheck.attach task (10, checkConnect), and the scan does not work. WiFi.scanNetworks () returns 0!
What to do?
```
void checkConnect() {
  if (WiFi.status() != WL_CONNECTED) {
    digitalWrite(SLed, LOW);
    if (wifiMode == "STA") { Mode_STA_AP(); }
    if (wifiMode == "STA+AP") {
      for (int i = 0; i < WiFi.scanNetworks(); i++) {
        if (WiFi.SSID(i) == STAn) { Init_WiFi(); }
      }
      WiFi.scanDelete();
    }
  }
}
```

reaper7, 2017-05-17:
leave Your scan code inside loop
and inside ticker task set only global boolean flag for triggering,
something like this:
```
volatile bool scanFlag = false;

void setScanFlag() {
  scanFlag = true;
}

void setup() {
  //your code here
  wifiCheck.attach(10, setScanFlag)
}

void loop() {
  if (scanFlag) {
    if (WiFi.status() != WL_CONNECTED) {
      if (wifiMode == "STA") { Mode_STA_AP(); }
      if (wifiMode == "STA+AP") {
        for (int i = 0; i < WiFi.scanNetworks(); i++) {
          if (WiFi.SSID(i) == STAn) { Init_WiFi(); }
        }
        WiFi.scanDelete();
      }
    }
  }
  //your code here
}
```

mcmega, 2017-05-17:
Thank you very much!
Added reset scanFlag.

reaper7, 2017-05-17:
done :)

closed 2017-05-20 by igrr