esp8266/Arduino #3252: SPI library on ESP8266 doesnt work

patriks51, 2017-05-17:
### Basic Infos

#### Hardware
Hardware:			ESP8266 nodeMCU lolin V3
Core Version:      	?2.1.0-rc2?

### Description

Hello guys, I have some problem with this library: https://github.com/esp8266/Arduino/tree/master/libraries/SPI
I want control this digital potentiometer http://ww1.microchip.com/downloads/en/DeviceDoc/11195c.pdf and after add the library a can't compile my project. Where is the problem, please?

### Settings in IDE

Module:  			Lolin V3 nodemcu
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:		?qio?
Flash Frequency:	?40Mhz?
Upload Using:		SERIAL
Reset Method:		?ck / nodemcu?

### Sketch

### Debug Messages
In file included from C:\Users\Patrik\Documents\Arduino\libraries\SPIn\SPIn.c:19:0:

C:\Users\Patrik\Documents\Arduino\libraries\SPIn\SPIn.h:41:1: error: unknown type name 'class'

 class SPISettings {

 ^

C:\Users\Patrik\Documents\Arduino\libraries\SPIn\SPIn.h:41:19: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token

 class SPISettings {

                   ^

C:\Users\Patrik\Documents\Arduino\libraries\SPIn\SPIn.h:50:1: error: unknown type name 'class'

 class SPIClass {

 ^

C:\Users\Patrik\Documents\Arduino\libraries\SPIn\SPIn.h:50:16: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token

 class SPIClass {

                ^

C:\Users\Patrik\Documents\Arduino\libraries\SPIn\SPIn.h:82:1: error: unknown type name 'SPIClass'

 extern SPIClass SPI;

 ^

In file included from C:\Users\Patrik\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.h:27:0,

                 from C:\Users\Patrik\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Stream.h:26,

                 from C:\Users\Patrik\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/HardwareSerial.h:31,

                 from C:\Users\Patrik\Documents\Arduino\libraries\SPIn\SPIn.c:20:

C:\Users\Patrik\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Printable.h:25:1: error: unknown type name 'class'

 class Print;

 ^

[... 196 lines cut ...]
              ^

C:\Users\Patrik\Documents\Arduino\libraries\SPIn\SPIn.c:421:14: error: expected '=', ',', ';', 'asm' or '__attribute__' before ':' token

 void SPIClass::writePattern(uint8_t * data, uint8_t size, uint32_t repeat) {

              ^

C:\Users\Patrik\Documents\Arduino\libraries\SPIn\SPIn.c:500:14: error: expected '=', ',', ';', 'asm' or '__attribute__' before ':' token

 void SPIClass::transferBytes(uint8_t * out, uint8_t * in, uint32_t size) {

              ^

C:\Users\Patrik\Documents\Arduino\libraries\SPIn\SPIn.c:514:14: error: expected '=', ',', ';', 'asm' or '__attribute__' before ':' token

 void SPIClass::transferBytes_(uint8_t * out, uint8_t * in, uint8_t size) {

              ^

C:\Users\Patrik\Documents\Arduino\libraries\SPIn\SPIn.c:553:1: error: unknown type name 'SPIClass'

 SPIClass SPI;

 ^

igrr, 2017-05-20:
The error message is pointing to `SPIn\SPIn.c`, which isn't the SPI library in this core. This one shoud be `SPI/SPI.cpp`.

patriks51, 2017-05-20:
Thank you. I renamed these files, because I had problem with colide library name.

igrr, 2017-05-21:
Well, in that case you may want to keep .cpp extension, because this is a C++ library.

closed 2017-05-22 by igrr