esp8266/Arduino #3254: calling delay()  2 times before calling Wifi.begin() will stop WiFi from connecting

lorneb, 2017-05-18:
### Basic Infos

#### Hardware
Hardware:			We Mos D1
Core Version:      	2.3.0

### Description

If delay(500) is called 2 times before the Wifi.begin() then WiFi connection never establishes

### Settings in IDE

Module:  			We Mos D1 R2 & Mini
Flash Size: 		4MB (3M SPIFFS)
CPU Frequency:	80Mhz
Upload Using:		SERIAL

### Sketch

```cpp
#include <ESP8266WiFi.h>

// Your WiFi credentials.
// Set password to "" for open networks.
char ssid[] = "xxxxx";
char pass[] = "xxxxx";

void setup()
{
  Serial.begin(115200);
  Serial.println("setup()");
  delay(500);
  //uncommenting out the next delay() causes Wifi to never connect
  //delay(500);  

  WiFi.begin(ssid, pass);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop()
{
   
}

```

### Debug Messages

```
--Good Connect
øsetup()
.
WiFi connected
IP address: 
192.168.0.93

--Failed Connect

øsetup()
....................................................................................................

```

igrr, 2017-05-18:
this is tracked as #2186

closed 2017-05-18 by igrr

alexvazquez, 2017-06-11:
Hi, I have the same problem, Is this already solved?

lorneb, 2017-06-12:
Yes it was a known issue see the linked (duplicate of ) issue

-- 

*Lorne Bonnell*
Solutions Architect

647 Flinders Street,
Townsville 4810 Australia
Telephone: +61 7 4444 4800  <+61+7+4444+4800>
Mobile: +61 418 7144 73

  <https://twitter.com/vix_technology>
<https://www.linkedin.com/company/2135174>
Vixtechnology.com <http://vixtechnology.com/>

<http://vixtechnology.com/>