esp8266/Arduino #3256: No response from device error

iAchyut, 2017-05-18:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:                ESP8266 module
Core Version:      	Arduino 1.8.2 and Python 2.7 installed (Path added to environment)

### Description
Hi
I uploaded the OTA bootloader serially and then uploaded blink led program via OTA, it worked for the first time. Now I am trying to upload the same program again but Arduino IDE shows error. 

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Reset Method:		ck

### Sketch

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

const char* ssid = "ABC";
const char* password = "123456789";

void setup() {
 Serial.begin(115200);
 Serial.println("Booting");
 WiFi.mode(WIFI_STA);
 WiFi.begin(ssid, password);
 while (WiFi.waitForConnectResult() != WL_CONNECTED) {
 Serial.println("Connection Failed! Rebooting...");
 delay(5000);
 ESP.restart();
 }

// Port defaults to 8266
 // ArduinoOTA.setPort(8266);

// Hostname defaults to esp8266-[ChipID]
 // ArduinoOTA.setHostname("myesp8266");

// No authentication by default
 // ArduinoOTA.setPassword((const char *)"123");

ArduinoOTA.onStart([]() {
 Serial.println("Start");
 });
 ArduinoOTA.onEnd([]() {
 Serial.println("\nEnd");
 });
 ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
 Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
 });
 ArduinoOTA.onError([](ota_error_t error) {
 Serial.printf("Error[%u]: ", error);
 if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
 else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
 else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
 else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
 else if (error == OTA_END_ERROR) Serial.println("End Failed");
 });
 ArduinoOTA.begin();
 Serial.println("Ready");
 Serial.print("IPess: ");
 Serial.println(WiFi.localIP());
 pinMode(2, OUTPUT);
}

void loop() {
 ArduinoOTA.handle();
 digitalWrite(2, HIGH); // turn the LED on (HIGH is the voltage level)
 delay(250); // wait for 2 second
 digitalWrite(2, LOW); // turn the LED off by making the voltage LOW
 delay(250); 
}

### Debug Messages
[19:02:36]:No Response from device

Although my ESP8266 is present in Tools>PORT option along with the local ip address. (My esp8266 is present in the network but still ArduinoIDE shows No response from Device error every time i upload the same program. )
This program worked for the very first time that too via OTA.

igrr, 2017-05-18:
Once you overwrite the OTA sketch with the Blink sketch, the board will no longer be able to receive OTA. In the sketch you upload you still need to initialize `ArduinoOTA` and call `ArduinoOTA.handle()` in the same way it is done in OTA examples.

iAchyut, 2017-05-18:
Sketch I uploaded above is my blink led sketch and it contains arduino Ota. Handle() and arduinoOTA. Moreover my esp8266 is still showing up in Port with local ip address that means my Ota is still working . Uploading same program again causes this error.  No response from Device.

igrr, 2017-05-20:
Please upload the sketch again via serial, this time enabling debug options in Tools menu:
Debug level: OTA + Updater
Debug port: Serial

Then keep serial monitor (not the one from Arduino, use some other software) open while uploading same sketch again over OTA. Once you reproduce the "no response from the device" issue, copy the output from the serial terminal, and attach it here.

devyte, 2017-09-06:
@iAchyut any progress with this?

odilonafonso, 2017-09-27:
I have same error.

Got this output from Putty:

[image]

odilonafonso, 2017-10-18:
it is definitely not possible to use OTA on a regular basis.

devyte, 2017-10-24:
@iAchyut @odilonafonso  after flashing over serial, did you physically reset the esp before doing OTA?

odilonafonso, 2017-10-24:
Hi @iAchyut. Yes.
I used OTA in my sketches earlier successfully. After implementing httpupdate, I stopped using OTA. Now I was trying to reuse OTA, as I made some changes to my application, but I can not re-record the firmware with OTA. Even the library examples do not work.
Thanks for help.

bcmears, 2017-11-01:
I am experiencing the same problem.  My debug output is essentially the same as odilonafonso's.  I installed the arduino IDE on a second laptop (and Python 2.7) and I find that I can successfully program OTA with the second machine, so it isn't a sketch problem, it isn't a board or board library problem, but there must somehow be something different in the installations on the two Win10 laptops.  I've done a clean install of Python and the IDE on my problem machine, but it has not helped.  Any suggestions would be appreciated.

dtbaker, 2017-11-01:
Following, Any ideas? First OTA worked great, now it shows up in the network list, the ESP is working fine, but I cannot OTA a second time. Have reset all the things. Is it to do with the chosen flash size when doing the OTA?

odilonafonso, 2017-11-04:
I gave up. I'm using httpUpdate.

supersjimmie, 2017-12-07:
I was using Arduino IDE 1.6.11 with esp8266-arduino v2.3.0 until now. 
The OTA always worked fine.

Today I switched to Arduino IDE 1.8.2 with esp8266-arduino Latest GIT.
First connection from the 1.8.2 IDE over OTA worked fine, at that moment the ESP was running the code compiled with 1.6.11/v2.3.0. So OTA, Wifi, network and the IDE (1.8.2) are fine. Note that I was connecting over OTA from IDE 1.8.2 and it worked.

After uploading the sketch where nothing ever changed to OTA or webserver or network, I was not able to use OTA anymore.  So to me it's clear that OTA is broke, not the IDE, my sketch or my network.

prasenjit-roy, 2018-01-17:
1. For ESP-01 need to set TOOLS>Flash Size>512 ( no SPIFFS)
2. Take it out of BOOT mode else WDT reset occurs and 2nd run may not work

gibo77, 2018-02-15:
My ArduinoOTA was working like a charm, until I incorporated it with WifiManager.h. The issue I have is "No Response from the device". I also got crash with Stack Dump, which another issue.
[image]

So I removed the WifiManager.h and went back to the old ArduinoOTA that worked before. But unfortunately, I am having that same problem. Do you think it has something to do with flash memory being occupied by the WifiManager.h? I did some research and said that if you do serial firmware download you basically erase the whole sketch memory space. I am not sure if this is true. Do you think I have to erase the EEPROM? Can you point me to a resource discussing about how the memory is split by Arduino ESP8266? Thanks guys.

Below is some information about flash space availability.
[image]

gibo77, 2018-02-16:
UPDATE: I found the problem!

My laptop that I use to run the Arduino IDE is connected to 5GHz Wifi that is why ArduinoOTA will return "No Response". 
I knew about this 5GHz Wifi issue before with ESP8266, but that was for Http. Now that I tried the OTA I forgot about it. I did all kind of stunts to figure out the problem. I did SPIFFS format and EEPROM clear 512bytes memory. But this snag had put me to be close to expert on Arduino ESP8266 memory management SPIFFS, PROGMEM and EEPROM.

bcmears, 2018-02-16:
Glad to hear you got yours working.  Still no joy for me.  I forced the
wifi to connect at 2.4GHz and I still get "No Response" from my esp8266.  I
even tried connecting my laptop through a 2.4GHz dongle.  Still won't
program.  I can program using an old laptop that doesn't have 5 GHz, and it
works fine.  I think you've gotten me closer to a solution but I don't have
it yet.

andrepalma, 2018-02-17:
Hi, I'm quite new with this OTA thing.. But i'm facing the same issues and is not about 5GHz, my esp is connected to 2.4 (different SSID) and i'm using a Desktop with physical connection to network..
First thing I've tried is to disable firewall. and still the same error. After that i was looking for the message and noticed that it seems that the programmer (I use Atom with platformio) send a message to 0.0.0.0 (it means a broadcast message) so I try, disable every network adapter I have, except of course, the one for Ethernet and it work just fine... 
After successful I enable the firewall again and voilá, everything is working.

P.S. - The adapter that i disable is the VirtualBox Host Adapter...

I hope this help other, mainly the people that still doesn't have it working yet.

prasenjit-roy, 2018-02-17:
For ESP-01 I had used 512 ( 64K SPIFFS) and it worked.
File size needs to be less than half of available Flash. 
Connect to serial port and you can see the fail reason if any.

andrepalma, 2018-02-17:
Yap, that is another thing we have to keep in mind, if the IC we are using have 1MB of memory our program cannot exceed 512KB. This is because it needs to kept the all new software while is still have the old software.

gibo77, 2018-02-19:
Do you have another esp8266 to try? I usually have two for comparison.

gibo77, 2018-02-19:
Yup. That is the rule now. When you change firmware the old one should still be in there, until the new one completely loads and run before wiping the old firmware. In the olden days they dont have this and if there is a power failure or comm error what you have is AN EXPENSIVE PAPER WEIGHT.

bcmears, 2018-02-21:
Thanks for your suggestion.  I did try a 2nd esp8266 module (same
manufacturer) and it won't program (does program with old laptop).  Also
tried disabling all wifi & ethernet drivers except for the built-in wifi.
Still no good.  Disabled internal wifi and used a 2.4 GHz dongle.  Still
won't program.  I even tried a different router without success.  Any ideas
are welcome.

LotusFly, 2018-02-28:
Try these settings on your Arduino>>Tools:

Board:  Generic ESP8266 Module
Flash Mode: DIO
Flash Size: 4M (1M SPIFFS)
Debug Port: Serial
Debug Level: OTA + Updater
Reset Method: ck

Use COM on first upload attempt.
Restart Arduino.
Change Port to Network Port you created.
Try uploading again.

Things i overlooked:
1. ArduinoOTA.setHostname("_yourHostnameHere_");     // _yourHostnameHere_ should include only alphabet characters
2. im using ESP8266MOD, it should use 4M (1M SPIFFS)

Verify if these will work for you.

bcmears, 2018-03-01:
Thank you for your help.  I have an ESP-12E.  Using your suggested settings
I received an EPROM sync failed message when trying to program via the USB.

This module will program OTA when using the Default ESP-12E settings when I
use an old laptop.  Even using this new laptop, communication starts as I
see the LED turn on solid when upload begins, but it appears to time out
and give the  no response message.

bcmears, 2018-03-03:
Finally, I've had some success.  Rather than use the Arduino IDE to program
OTA, I've used the web browser approach as described in the OTA github
project
https://github.com/esp8266/Arduino/tree/master/doc/ota_updates#web-browser

This works flawlessly for me.  I still would like to know why the IDE
approach won't work but at least I have a work-around.
Thank you all for your suggestions.

LotusFly, 2018-03-04:
That's a good thing.  I'll check the same and try if the workaround will also work for me.

gibo77, 2018-03-20:
Seems like using delay() could attribute to failure of OTA. I am not 100% sure. I have to counter check on this. But I throw in delay() just now and my OTA is unresponsive.

daebikelinz, 2018-05-08:
I've encountered the same problem. I only signed up to github to make a post here because it could help others.

In my case it was also the delay() function. After i deleted it, it works perfect.

Greets from Germany

dtbaker, 2018-05-08:
For what it's worth I just got a new computer and setup Arduino and the latest drivers from scratch, OTA is now working fine multiple times when I was previously getting these "no response from device" error messages. 

Just make sure to choose the right flash size, I chose 512M first when I should have chosen 4M.

el-basko, 2018-05-25:
Tenía el mismo problema que @bcmears. Lo solucioné agregando Python y Arduino IDE a las excepciones del firewall.

I had the same problem as @bcmears. I solved it by adding Python and Arduino IDE to the firewall exceptions.

ganan-s, 2018-06-26:
I have a wemos d1 mini pro, and solved it by using Auto Format (under tools menu) then re-uploaded my code and OTA worked. I was having an issue with OTA when I changed my SSID and password, where it would tell me the device did not respond.

ltvanderkrogt, 2018-11-14:
19:25:25 [ERROR]: No Answer

4M (3M SPIFFS) worked for me with WEMOS.

agraymd, 2019-02-10:
I was able to cure NO RESPONSE FROM DEVICE when trying to upload OTA to a Wemos D1 as well as a Wemos D1 Mini, by turning off the Windows firewall.   

Now I have to figure out how to add an exception so I can keep the firewall on.

AitorCM, 2019-04-09:
Same error here, @el-basko solution worked for me, just add firewall exceptions for Arduino IDE and Python exe's files.

devyte, 2019-04-11:
This discussion has become convoluted and there has been no response from OP. Closing.

closed 2019-04-11 by devyte

priyaru, 2019-06-10:
> Try these settings on your Arduino>>Tools:
>
> Board: Generic ESP8266 Module
> Flash Mode: DIO
> Flash Size: 4M (1M SPIFFS)
> Debug Port: Serial
> Debug Level: OTA + Updater
> Reset Method: ck
>
> Use COM on first [...]

priyaru, 2019-06-10:
For me , debug level is shown as OTA  not OTA+UPDATER..which cause any issues? and why it is shown like this to me only..? Moreover I didnt get the OTA output.. i doesnt upload a code even single time..  :(

yoosamui, 2020-03-21:
ESP8266-12E
OTA definitively don't work with static IP
  
Firewall is off!!!
  WiFi.mode(WIFI_STA);
  WiFi.config(IPADDRESS, GATEWAY, NETMASK);  // static ip, gateway, netmask
  WiFi.begin(credentials::m_credentials.m_wifi_ssid,           
 credentials::m_credentials.m_wifi_password);  
DON"T WORK!!!
-----------------------------------
Without static IP after enter the OTA password in the IDE the download over Wifi finish successfully.
  WiFi.mode(WIFI_STA);
//  WiFi.config(IPADDRESS, GATEWAY, NETMASK);  // static ip, gateway, netmask
  WiFi.begin(credentials::m_credentials.m_wifi_ssid,   
     credentials::m_credentials.m_wifi_password);  

Way is that?
the NTPClient lib also don't work with static IP
do i missing something?
Thanks!

khyarul, 2020-03-30:
My best config for OTA, more than 3 times upload with success
I'm using this in my Attandance Node project with **Scheduler.h** to handle much task like:
- buzzer & button
- lcd & rfid
- MQTT loop

[image]

yoosamui, 2020-03-30:
OTA is working now with static IP's  upload with success
I just use the OTABasic example and add an static IP.
i did not change anything in the tools settings

I don't know exactly  way now is working.
Anyway thanks to khyarul and all of you.

m4rc77, 2020-11-21:
I also had the infamous `No response from device error`. In my case (on macOS  Mojave, 10.14.6, with Plattform IO) it was definitively the firewall from macOS. Just tested it by quickly disabling the firewall, and it worked!

**As a final solution I had to add add `python 3.8` in the "Firewall Options" and `Allow incoming connections`. Which then solved the issue.**

ratnick, 2020-12-06:
For me, it was the firewall blocking Python for access to private networks (Windows 10).

siparker, 2021-03-01:
same for me. checked for python in my windows 10 firewall. change settings to allow the Arduino folder >  tools >  python 3.x to be allowed to communicate with the network.

chimeranzl, 2021-10-31:
Just for the record, I got this same error and realised that in my code, I forgot to call my "setup wifi" routine in setup{} prior to the arduinoota.onstart etc calls - of course, there was no response from the device coz wifi wasn't started first _facepalm_

sscholle, 2022-12-21:
Likely a Python/Firewall issue. You need to allow Python on your Firewall