esp8266/Arduino #3260: Hardware- ESP12 unable to communicate. Module able  to

Fireflaker, 2017-05-18:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:		ESP-12
Core Version:      	Do not know. Unable to flash.

### Description

Unable to upload sketch or flash firmware. However, if I reset the ESP12 by connecting reset to ground, I get a message saying Ai Thinker ready in serial monitor. No version code or boot mood included. It seems like my computer can not get MAC from the module. I tried a TTL converter and arduino UNO to upload. Eather works. 

### Settings in IDE

Module:  		Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:.    80M
Flash Mode:	        Tried all
Flash Frequency:	tried all
Upload Using:        USB to ttl converter
Reset Method:       ? Do not know.

### Sketch

The sketch does not have problem compiling or uploading to a nodemcu development board.

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

Fireflaker, 2017-05-18:
The sketch is just Blink example.

this is debugging info

warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

Settings are
80MHz,40MHz,DIO,115200,4M(1M),ck,disabled,None

jaromanda, 2017-05-19:
Do you know, unless you have one of those fancy modules with ck/nodemcu circuitry, you need to ground GPIO-0 while resetting to put the module into "flashing" mode?

Fireflaker, 2017-05-19:
Problem solved. It was a connection problem. I thought 5V data may damage the ESP8266 so I built a resister devider. I removed it and now everything is fine. 

Thank you for helping.

closed 2017-05-20 by igrr