esp8266/Arduino #3264: Inconsistent Post Requests

KyleHunter, 2017-05-19:
### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	Latest NodeMCU

### Description

So this issue has been infuriating to me for the last several weeks.. It is related to my prior issue: https://github.com/esp8266/Arduino/issues/3182

I had incorrectly assumed that it was fixed when it wasn't. So I am sending post requests to dweetpro.io every 15 seconds. I have the most inconsistent results with it. I have tried everything, from having a local WiFiClient instance, to a global one, that only connects at the start of the sketch. When I do WiFiClient.println(); sometimes it will go super fast, submit the post, return a response in less than half a second. Other times it will take seconds for each println(), but still get a valid post. Other times it will take 5+ seconds on each println() and won't submit a valid post..

But what is so frustrating, is that it will work for hours/days only failing on a few here and there, and then it will fail completely until a restart, but many times restarting it doesn't immediately solve it either. I've tried printing the current signal strength, and it's in the same room as the router so it is always -50dB or better.

Everything with the website, dweetpro is fine. Because when the esp is failing, I am able to manually make post requests from other sources and they work perfectly.

I've tried with a generic esp-12, nodemcu, same issues. I really don't know what else to try, as it is just bizarre how inconsistent this board is.

### Settings in IDE

Module:  			ESP-12
Flash Size: 		4MB
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		SERIAL
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

  bool send_post_to_server(WiFiClient client, String content, int content_length)
  {
    Serial.println(client.connected()); //always prints 1
    Serial.println("Here2.2"); // For debugging
    client.println("POST /v2/dweets HTTP/1.1");
    Serial.println("Here2.21"); // For debugging
    client.println("Host: dweetpro.io");
    Serial.println("Here2.22"); // For debugging
    client.println("X-DWEET-AUTH: xxx");
    Serial.println("Here2.23"); // For debugging
    client.println("Accept: application/json");
    Serial.println("Here2.24"); // For debugging
    client.println("Content-Type: application/json");
    Serial.println("Here2.25"); // For debugging
    client.print("Content-Length: ");
    Serial.println("Here2.36"); // For debugging
    client.println(content_length);
    client.println();
    Serial.println("Here2.4"); // For debugging
    client.print(content);
    Serial.println("Here2.5"); // For debugging
  }

  void send_to_server(String* time_sent, String status,
    bool, float* data, int data_size, WiFiClient client)
  {
    Serial.println("Here2"); // For debugging
    int content_length;
    String content = make_json_content(&content_length, status, data, data_size);
    Serial.println("Here2.1"); // For debugging
    send_post_to_server(client, content, content_length);
    Serial.println("Here3"); // For debugging
    delay(10);
    int timer = millis();
    while (client.available() == 0)
    {
      delay(10);
      if (millis() - timer > 3000)
      {
        Serial.println("2"); // When it fails, it is here. I have increased time to 10000, no difference
        return;
      }
    }

      Serial.println("Here4"); // For debugging
      String json = "";
      bool http_body = 0;

      led::turn_green_led(0);
      while(client.available())
      {
        String line = client.readStringUntil('\r');
        if (!http_body && line.charAt(1) == '{')
          http_body = 1;
        if (http_body)
          json += line;
      }

      Serial.println("Response: ");
      Serial.println(json);
  }

// In main loop I use:

  if (!client.connected())
  {
    client.stop();
    client.connect(IPAddress(54, 236, 198, 240), 80);
  }
//I have tried local instances, and global where I connect/disconnect each request.
//This way I have now seems slightly(?) more reliable

```

### Debug Messages

```
messages here
```

igrr, 2017-05-19:
Just to confirm, are you using the (latest) git version of the core, or the one installed via boards manager (2.3.0)?

Is there any specific reason for formatting the request manually using WiFiClient, instead of using the bundled ESP8266HTTPClient library? In version 2.3.0, every `println` you do causes 3 TCP segments to be sent — one for the line itself, and two more for `\r` and `\n`. This has been optimized in git version (so Nagle algorithm will prevent short writes from being sent immediately), but still it is better to prepare all the headers in a single buffer and send them in a single TCP segment — that's what the HTTPClient library does.

KyleHunter, 2017-05-19:
@igrr I'm using the latest version on git.

That's one thing I haven't tried yet, I'll give that a shot

KyleHunter, 2017-05-19:
@igrr using platformIO, my version is:
"Updating espressif8266                   @ 1.3.0          [Up-to-date]"

Using this new code:
```cpp

  void send_to_server(String* time_sent, String status, float* data, int data_size)
  {
    HTTPClient http;

    int content_length;
    String content = make_json_content(&content_length, data, data_size);

    http.begin("http://dweetpro.io/v2/dweets");

    http.addHeader("Content-Type", "application/json");
    http.addHeader("Accept", "application/json");
    http.addHeader("Accept", "application/json");
    http.addHeader("Content-Length", String(content_length));
    http.addHeader("X-DWEET-AUTH", "xxx");

    http.POST(content);
    http.writeToStream(&Serial);
    http.end();
  }

```
Same issues as before.. Still very inconsistent

KyleHunter, 2017-05-19:
Sorry, didn't mean to close..

KyleHunter, 2017-05-20:
@igrr if I do:
"int s = http.POST(content);"
and then print s, it returns -3 when it fails.

igrr, 2017-05-20:
As far as i can tell, platformio doesn't use the latest git version of the core, 1.3.0 was released in February (https://github.com/platformio/platform-espressif8266/commit/fc1b95adfcfb6420a23dca0763c869f8a70e0a1e).

Can you do a packet capture on the router side, or by sniffing the wifi network (using Wireshark or similar)? That would shed some light on the possible cause of the issue.

KyleHunter, 2017-05-20:
@igrr Here is the result of me using wireshark: https://drive.google.com/file/d/0B5-5-5KtayGhUDBIUUVpRU8zV28/view?usp=sharing

Data looked weird to me, it didn't use the IP assigned to it from my router's website.

igrr, 2017-05-20:
Yep this capture doesn't seem to have the POST from 192.168.1.177 recorded. If you are recording on the router side, check which interface you are using (eth/wifi). If recording from the air, check that your adapter is in monitor mode, on the right channel, on the right network.

KyleHunter, 2017-05-20:
@igrr Any idea how I can do that? I have a linksys wrt1900c and can't see any way to do it from that. Also, do you have a skype or some other form of communication that would be easier than this?
Thanks for all your help

KyleHunter, 2017-05-21:
@igrr Managed to set up a hotspot on my laptop, connect the esp to it, and sniff the data from it.

Here's what I got: https://drive.google.com/file/d/0B5-5-5KtayGhdTFISHZjSFZmYzg/view?usp=sharing

Contains successful posts, and failures.

Thank you so much again for all of your help

igrr, 2017-05-22:
There's one case around time=1108 where ESP8266 doesn't receive the SYN/ACK when connection is being established, even though the remote retransmits it 5 times. I don't have an immediate clue why that would be happening, but will try a few things.

The other cases, towards the bottom of the dump show that the remote replies with `HTTP/1.1 504 GATEWAY_TIMEOUT`. Whether this is intentional (=rate limiting) or unintentional (something wrong with their reverse proxy setup) I don't know.

KyleHunter, 2017-05-22:
@igrr Yeah I could see the 5 time retry live while it was failing to post also..

Found these issues which seem fairly similar:
http://www.esp8266.com/viewtopic.php?f=6&t=770&start=20
http://www.esp8266.com/viewtopic.php?f=6&t=1633&start=40

Wonder if it could be dependent on the wifi router I have? I'm using the dual band 5/2.4ghz wrt1900ac which is a pretty darn powerful router. I've noticed that if i'm not in the same room as it, i'm not able to connect to wifi, which seems odd to me.

KyleHunter, 2017-05-23:
@igrr So, not 100% sure that the problem is fixed, but it has been running for ~18 hours without any failures, so it seems like it very well may be.

What I did, is disabled the 5Ghz band on my dual band router. The SSID for the 2.4Ghz was in this format: "thisisthessid" and the 5Ghz was "thisis thessid". The same ssid, but one has a space in it. 

Maybe the esp would get confused with the two different ssids or something?

devyte, 2017-09-06:
Given that disabling the 5GHz radio seems to have fixed the described issue, I think it's safe to say that there is no issue with the core. I am therefore closing this.
A personal note here left here in case it may help anyone, I have 4 TPLink Archer C7 dual band 1900ac routers running dd-wrt, bridged via WPS, and the ESPs seems to be working fine with them.

closed 2017-09-06 by devyte