esp8266/Arduino #3265: Cant compile

christoferjh, 2017-05-19:
Seems like a recent commit according timeout broke the code.

C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\WiFiClient.cpp: In constructor 'WiFiClient::WiFiClient()':

C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\WiFiClient.cpp:51:15: error: class 'WiFiClient' does not have any field named '_timeout'

 : _client(0), _timeout(5000)

               ^

C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\WiFiClient.cpp: In constructor 'WiFiClient::WiFiClient(ClientContext*)':

C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\WiFiClient.cpp:57:20: error: class 'WiFiClient' does not have any field named '_timeout'

 : _client(client), _timeout(5000)

                    ^

exit status 1
Error compiling for board WeMos D1 R2 & mini.

christoferjh, 2017-05-19:
Checking out 28/5 makes it work.

igrr, 2017-05-20:
Thanks for reporting this, looks like after the recent update of the CI script, build errors stopped bubbling up and breaking the build, so this bug went in unnoticed.
This should be fixed in 3363be0 and 4812cae.

closed 2017-05-20 by igrr