esp8266/Arduino #3267: install of 2.3.0 using board manager fails, can't download esp8266-2.3.0.zip

jonhp, 2017-05-19:
even though i can download the "exact" same file manually, the arduino ide board manager just says error downloading https://github.com....

igrr, 2017-05-20:
See comment in #3269

closed 2017-05-20 by igrr

jonhp, 2017-05-20:
Error downloading https://github.com/esp8266/Arduino/releases/download/2.3.0/esp8266-2.3.0.zip
java.lang.RuntimeException: java.lang.Exception: Error downloading https://github.com/esp8266/Arduino/releases/download/2.3.0/esp8266-2.3.0.zip
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:176)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: Error downloading https://github.com/esp8266/Arduino/releases/download/2.3.0/esp8266-2.3.0.zip
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:131)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:69)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:102)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:173)
	... 1 more
Caused by: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
	at sun.security.ssl.Alerts.getSSLException(Alerts.java:192)
	at sun.security.ssl.SSLSocketImpl.fatal(SSLSocketImpl.java:1949)
	at sun.security.ssl.Handshaker.fatalSE(Handshaker.java:302)
	at sun.security.ssl.Handshaker.fatalSE(Handshaker.java:296)
	at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1509)
	at sun.security.ssl.ClientHandshaker.processMessage(ClientHandshaker.java:216)
	at sun.security.ssl.Handshaker.processLoop(Handshaker.java:979)
	at sun.security.ssl.Handshaker.process_record(Handshaker.java:914)
	at sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:1062)
	at sun.security.ssl.SSLSocketImpl.performInitialHandshake(SSLSocketImpl.java:1375)
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1403)
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1387)
	at sun.net.www.protocol.https.HttpsClient.afterConnect(HttpsClient.java:559)
	at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:185)
	at sun.net.www.protocol.http.HttpURLConnection.followRedirect0(HttpURLConnection.java:2701)
	at sun.net.www.protocol.http.HttpURLConnection.followRedirect(HttpURLConnection.java:2623)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1806)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1474)
	at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:338)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:171)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:129)
	... 4 more
Caused by: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
	at sun.security.validator.PKIXValidator.doBuild(PKIXValidator.java:387)
	at sun.security.validator.PKIXValidator.engineValidate(PKIXValidator.java:292)
	at sun.security.validator.Validator.validate(Validator.java:260)
	at sun.security.ssl.X509TrustManagerImpl.validate(X509TrustManagerImpl.java:324)
	at sun.security.ssl.X509TrustManagerImpl.checkTrusted(X509TrustManagerImpl.java:229)
	at sun.security.ssl.X509TrustManagerImpl.checkServerTrusted(X509TrustManagerImpl.java:124)
	at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1491)
	... 22 more
Caused by: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
	at sun.security.provider.certpath.SunCertPathBuilder.build(SunCertPathBuilder.java:141)
	at sun.security.provider.certpath.SunCertPathBuilder.engineBuild(SunCertPathBuilder.java:126)
	at java.security.cert.CertPathBuilder.build(CertPathBuilder.java:280)
	at sun.security.validator.PKIXValidator.doBuild(PKIXValidator.java:382)
	... 28 more

igrr, 2017-05-20:
Which OS version and Arduino IDE version are you getting this error with?

jonhp, 2017-05-20:
windows 10 (up-to-date) and Arduino IDE 1.8.2.

igrr, 2017-05-20:
These two reports might be related:
https://forums.adafruit.com/viewtopic.php?f=25&t=92357
https://github.com/arduino/Arduino/issues/4779

jonhp, 2017-05-20:
It could be the firewall, but when I put https://github.com/esp8266/Arduino/releases/download/2.3.0/esp8266-2.3.0.zip into the browser, it downloads just fine. 
Additional Boards Manager URLs: http://arduino.esp8266.com/stable/package_esp8266com_index.json

jonhp, 2017-05-20:
whoops, so I just read the 2nd one. didn't know the ide/jre had it's own certs and didn't use the system ones. I need to look into that. thanks, jon

jonhp, 2017-05-20:
That was the key. I used the instructions at: https://www.grim.se/guide/jre-cert to install the firewall certificate into the JRE cacerts. install worked fine. Now I just have to remember to reinstall after every update of the IDE. There should be a better way.

igrr, 2017-05-21:
Suggest posting this to https://github.com/arduino/Arduino/issues/4779, maybe IDE developers will take a look...