esp8266/Arduino #3268: Unable to compile to Lolin NodeMCU v3

nattgud, 2017-05-20:
### Basic Infos

Unable to compile anything to NodeMCU.

#### Hardware
Hardware:			Lolin NodeMCU v3 (esp8266mod)
Core Version:      	2.3.0

### Description

I'm unable to compile any arduino sketch at all. I get an error message saying it can' find "xtensa-lx106-elf-gcc" even though it does exist. I've tried to install through both the board manager and github (incl get.py) to no success. 

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:		-
Flash Frequency:	-
Upload Using:		Serial
Reset Method:		-

### Sketch

void setup() {
  Serial.begin(9600);
}

void loop() {

}

### Debug Messages

xtensa-lx106-elf-gcc: error: CreateProcess: No such file or directory

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

igrr, 2017-05-20:
Which operating system are you having this problem on? Is there any antivirus software installed? Which version of Arduino IDE?

nattgud, 2017-05-20:
Oh my God... I unblocked everything in Comodo before I tried installed and ran get.py. It runs without any problems now... Sorry for bothering with a stupid report ^^

closed 2017-05-20 by nattgud