esp8266/Arduino #3269: win32-xtensa-lx106-elf-gb404fb9-2.tar.gz very slow download

jonhp, 2017-05-20:
sometimes it fails. sometimes it takes a very long time. can this be hosted on a faster, more reliable server?

igrr, 2017-05-20:
We've recently moved downloads to Github Releases, which are hosted on AWS. Same files can still be downloaded from arduino.esp8266.com (i.e. http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz). You have an option of downloading from that URL and dropping the file into dist folder, then running `get.py` again to unpack it.

For most users, downloads from Github Releases are more reliable than the old server, which had a few hours of downtime every couple of months. In some regions, like here in China, for example, downloads from the AWS are less reliable, but that's the trouble with Chinese internet in general.
Out of curiosity, which region are you located in?

closed 2017-05-20 by igrr

jonhp, 2017-05-20:
I am in USA and I have had this problem using multiple systems over the last two days at different times. From a user perspective this issue and 3267, I think, are different problems.