esp8266/Arduino #3270: Send multiple message on the same connection

stefanoromanello, 2017-05-20:
Hi, I'm using this example https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiClientBasic/WiFiClientBasic.ino

I would like to send multiple messages without having to create a new connection. This is my current code but it works only for the first command. The server is working great I tested it with this https://sourceforge.net/projects/sockettest/ and I can send multiple messages.

`#include <ESP8266WiFi.h>

const int port = 9999;
const char * host = "192.168.4.2"; // ip or dns

void setup() {
  Serial.begin(9200);
  Serial.println("Iniziato");
  startWifi();
}

void loop() {
  delay(5000);
  
  Serial.print("connecting to ");
  Serial.println(host);

  // Use WiFiClient class to create TCP connections
  WiFiClient client;
  const int httpPort = 9999;
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }

    // This will send the request to the server

    for(int i=0;i<15;i++)
    {
      String msg = "CMD:" + String(i);
      client.print(msg);
      Serial.println(msg);
      //read back one line from server
      String line = client.readStringUntil('\r');
      client.println(line);
      delay(2000);
    }
    
    

    Serial.println("closing connection");
    client.stop();
    
    Serial.println("wait 5 sec...");
    delay(5000);
}
void startWifi() {
  Serial.println("\nStarting WiFi AP:");
  WiFi.mode(WIFI_STA);
  WiFi.softAP("RemoteControl", "123456789", 1, false);
  Serial.println("SSID     : 'RemoteControl'");
  Serial.println("Password : '123456789'");
  Serial.println("-----------------------------------------------");
}

void stopWifi() {
  Serial.println("stopping WiFi AP");
  Serial.println("-----------------------------------------------");
  WiFi.disconnect();
}
`

igrr, 2017-05-22:
Hi, i have tried your sketch and can't reproduce the issue.

Log from the ESP8266:
```
Started

Starting WiFi AP:
SSID     : 'RemoteControl'
Password : '123456789'
-----------------------------------------------
connecting to 192.168.4.2
connection failed
connecting to 192.168.4.2
connection failed
connecting to 192.168.4.2
connection failed
connecting to 192.168.4.2
connection failed
connecting to 192.168.4.2
connection failed
connecting to 192.168.4.2
connection failed
connecting to 192.168.4.2
CMD:0
CMD:1
CMD:2
CMD:3
CMD:4
CMD:5
CMD:6
```

On PC, i start netcat, and it receives all the lines:
```
$ nc -w 100 -vvv -l 9999
CMD:0
CMD:1
CMD:2
CMD:3
CMD:4
CMD:5
CMD:6
```

stefanoromanello, 2017-05-22:
Ok, so I don't know what is going on. This is my android server class. 
`import android.os.AsyncTask;
import android.util.Log;
import android.widget.TextView;

import java.io.BufferedReader;
import java.io.IOException;
import java.io.InputStreamReader;
import java.net.InetAddress;
import java.net.NetworkInterface;
import java.net.ServerSocket;
import java.net.Socket;
import java.net.SocketException;
import java.util.Enumeration;

public class Server2 extends AsyncTask<Void, Void, String> {

    final StackTraceElement se = Thread.currentThread().getStackTrace()[2];
    private String data = null;
    ServerSocket serverSocket;

    MainActivity activity;

    public Server2(MainActivity a)
    {
        this.activity = a;
    }

    @Override
    protected String doInBackground(Void... params) {
        String line="";
        try {
            serverSocket = new ServerSocket(9999);
            while (true) {
                Socket socket = serverSocket.accept();
                socket.setKeepAlive(true);

                BufferedReader in = new BufferedReader(new InputStreamReader(socket.getInputStream()));
                int readed = in.read();
                Log.d("Readed", "readed bytes : " + readed);

                while ((line = in.readLine()) != null) {
                    Log.i("line", "line : " + line);

                    final String toUI = line;
                    activity.runOnUiThread(new Runnable() {
                        @Override
                        public void run() {
                            TextView msg = (TextView)activity.findViewById(R.id.msg);
                            msg.setText(toUI);

                        }
                    });

                }

                socket.close();

            }
        } catch (IOException e) {
            e.printStackTrace();
        }
        return line;
    }

    @Override
    protected void onPostExecute(String result) {
        super.onPostExecute(result);
        Server2.this.data = result;

    }

    public void onDestroy() {
        if (serverSocket != null) {
            try {
                serverSocket.close();
            } catch (IOException e) {
                // TODO Auto-generated catch block
                e.printStackTrace();
            }
        }
    }

    public String getData() {
        return data;
    }

    public String getIpAddress() {
        String ip = "";
        try {
            Enumeration<NetworkInterface> enumNetworkInterfaces = NetworkInterface
                    .getNetworkInterfaces();
            while (enumNetworkInterfaces.hasMoreElements()) {
                NetworkInterface networkInterface = enumNetworkInterfaces
                        .nextElement();
                Enumeration<InetAddress> enumInetAddress = networkInterface
                        .getInetAddresses();
                while (enumInetAddress.hasMoreElements()) {
                    InetAddress inetAddress = enumInetAddress
                            .nextElement();

                    if (inetAddress.isSiteLocalAddress()) {
                        ip += "Server running at : "
                                + inetAddress.getHostAddress();
                    }
                }
            }

        } catch (SocketException e) {
            e.printStackTrace();
            ip += "Something Wrong! " + e.toString() + "\n";
        }
        return ip;
    }

}`

stefanoromanello, 2017-05-22:
I've tried to do the same. These are my results.

ESP

Starting WiFi AP:
SSID     : 'RemoteControl'
Password : '123456789'
-----------------------------------------------
connecting to 192.168.4.2
connection failed
connecting to 192.168.4.2
connection failed
connecting to 192.168.4.2
connection failed
connecting to 192.168.4.2
CMD:0
CMD:1
CMD:2
CMD:3
CMD:4
CMD:5
CMD:6
CMD:7
CMD:8
CMD:9
CMD:10
CMD:11
CMD:12
CMD:13
CMD:14
closing connection
wait 5 sec...

NETCAT

C:\Users\Stefano\Desktop\nc>nc -w 100 -vvv -l -p 9999
listening on [any] 9999 ...
192.168.4.1: inverse host lookup failed: h_errno 11004: NO_DATA
connect to [192.168.4.2] from (UNKNOWN) [192.168.4.1] 4100: NO_DATA
 sent 0, rcvd 0

C:\Users\Stefano\Desktop\nc>

stefanoromanello, 2017-05-22:
I've also tried to change the code to void loop() {
  
```
  Serial.print("connecting to ");
  Serial.println(host);

  // Use WiFiClient class to create TCP connections
  WiFiClient client;
  const int httpPort = 9999;
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }

    // This will send the request to the server

    /*for(int i=0;i<15;i++)
    {
      String msg = "CMD:" + String(i);
      client.print(msg);
      Serial.println(msg);
      //read back one line from server
      //String line = client.readStringUntil('\r');
      //client.println(line);
      delay(2000);
    }*/
    client.print("TEST");
    client.print("TEST");
    client.print("TEST");
    client.print("TEST");
    client.print("TEST");
    

    Serial.println("closing connection");
    //client.stop();
    
    Serial.println("wait 5 sec...");
    delay(5000);
}
```
and after testing it with  http://www.hw-group.com/products/hercules/index_en.html sometimes I received 2 messages, sometimes 3. For me there something like a timeout on the client side.

devyte, 2019-10-31:
This issue is ancient, and there hasn't been feedback in a very long time. A lot has happened since this was reported.
Closing. If the problem can be reproduced with latest git, please open a new issue.

closed 2019-10-31 by devyte