esp8266/Arduino #3275: NodeMcu v1.0 (ESP8266 12E) Store/Read an (int) using the EEPROM

SurtrTch, 2017-05-22:
Hello, I want to store a value (int) of four numbers like (int a=1111) in the NodeMcu v1.0 EEPROM then when I turn off and on again the board the int i will stay equal to 1111.
I tried this code on my Arduino UNO board and it works fine, to store:
`#include <EEPROM.h>

void setup() {

  Serial.begin(9600);
  
  int a=1111;
  int eeAddress = 0;

  EEPROM.put(eeAddress, a);

  eeAddress += sizeof(int);

  EEPROM.put(eeAddress, a);
  Serial.print("Written");
}

void loop() {
  /* Empty loop */
} `

And this one to read the data when I unplug and plug the board again:

`#include <EEPROM.h>

void setup() {

  int a;
  int eeAddress = 0; 

  Serial.begin(9600);
  while (!Serial) {
    ; 
  }
  EEPROM.get(eeAddress, a);
 
  Serial.println(a);

}

void loop() {
  /* Empty loop */
} `

Now when I use it on the NodeMcu board I always get the same value whatever the int I store

SurtrTch, 2017-05-22:
I found the solution It works now.
To store a value
```
#include <EEPROM.h>

void setup() {

  Serial.begin(115200);
  EEPROM.begin(512);
  
  int a=5412;
  int eeAddress = 0;

  EEPROM.put(eeAddress, a);

  eeAddress += sizeof(int);

  EEPROM.put(eeAddress, a);
  EEPROM.commit();
  Serial.print("Written");
}

void loop() {
  /* Empty loop */
} 
```

To read that value

```
#include <EEPROM.h>

void setup() {

  int a;
  int eeAddress = 0; 

  Serial.begin(115200);
  EEPROM.begin(512);
  while (!Serial) {
    ; 
  }
  EEPROM.get(eeAddress, a);
 
  Serial.println(a);

}

void loop() {
  /* Empty loop */
}
```

d-a-v, 2017-05-22:
You need EEPROM.commit() after all your EEPROM.put() in the first sketch.
see [example](https://github.com/esp8266/Arduino/blob/master/libraries/EEPROM/examples/eeprom_write/eeprom_write.ino)
reason: EEPROM is emulated in ram

SurtrTch, 2017-05-22:
I did it, now it's fine.
Thanks

closed 2017-05-23 by igrr