esp8266/Arduino #3277: Ethernet and SSL/TLS?

Humancell, 2017-05-22:
I'm beginning to experiment with the Ethernet support in the library, and I am curious if there is any way to leverage the SSL/TLS work that was done for WiFi over the Ethernet client connections?

Right now, there does not seem to be any support in the examples ... but can this be done?

igrr, 2017-05-23:
TLS support currently uses some of the WiFiClient internals (`ClientContext`) directly, so it can not be abstracted away easily. With some refactoring it would be possible to convert TLS support into a mixin class (`TLSWrapper`), so you could inherit WiFiClientSecure from `TLSWrapper<WiFiClient>` and EthernetClientSecure from `TLSWrapper<EtherenetClient>` to get the TLS features.

I'll keep this issue open, but be advised that I'm not planning to work on this.

Humancell, 2017-05-23:
Hello ... thank you for the response.  This confirmed my suspicions, and your response is both helpful and expected.

I know this is not the ESP32 repo/issues ... but my guess would also be that any work done on this would not be reused in the new ESP32 versions of code as the ESP32 has a completely different Ethernet hardware implementation, and hardware support for SSL/TLS?

igrr, 2017-05-23:
The reason why this work will not be reused for the ESP32 is mostly due to the fact that different TLS libraries are used — on the ESP8266 we use axTLS and on ESP32 we use mbedTLS.

But the overall pattern of introducing a mixin class for TLS features and then using it to create TLS-enabled versions of Client class for each interface (WiFi, Ethernet) can be taken there.

gjt211, 2017-11-08:
@Humancell, did you get any further with this? I have ethernet connected to an ESP8266 and it would be beyond amazing if I could use the TLS support that I currently use with WiFi.

@igrr, I know you are extremely busy and the ESP8266 is probably not that high on the priority list as it used to be. Can you provide any updates on 'if' TLS might ever get into Ethernet for the ESP8266?

I would attempt to do what you mentioned in your first post, but my code skill level is not that good. I would not know where to start. Do you have, or is there some sort of guide that could help me with this?

igrr, 2017-11-08:
I don't have plans for something like EthernetClientSecure myself, but i don't see why it wouldn't be possible to implement. If you are willing to look into this, I would suggest starting by looking at the differences between WiFiClient and WiFiClientSecure, and seeing how SSL support is added by the latter. Then see if that "SSL support" feature can be extracted out of WiFiClientSecure into a separate class template. This class template would then be used by both WiFiClientSecure and EthernetClientSecure.

The constraints to consider are 1) compatibility with the older code, 2) maintainability of the final code (avoiding code duplication).

gjt211, 2017-11-08:
@igrr, thank you for your reply. Your explanation of the process actually makes sense to me, considering that I don't yet understand what classes are.  I have tried to learn them in php without success, even though I use them I have trouble creating my own.

I am certainly willing to look into this, and now I have a starting point. I will start with a comparative check between WiFiClient & WiFiClientSecure.

May I ask, if I need to ask questions about implementing this, are you able to provide guidance from time to time?

GerogeRomy4, 2017-12-18:
Hi guys,
I have been looking at this issue and I am very interested in doing something similar.

Basically, I am able to add a certificate to a WINC1500 and get my WiFiSSLClient working. 
My next step would be to add RJ45 Ethernet support via SPI to a separate board.

I would like to use the TLS support currently using the WiFiSSLClient internals to the RJ45 Ethernet board.

Any advise / thoughts?
Thanks,
Al

devyte, 2019-10-31:
This should come for free with #6680 .

PR #6680 (2020-12-22): more lwIP physical interfaces
19-10-28:

* Ethernet: W5100, W5500 and ENC28J60, thanks to quality drivers from @njh repositories
https://github.com/njh/EtherSia/tree/master/src/enc28j60.cpp
https://github.com/njh/W5100MacRaw
https://github.com/njh/W5500MacRaw
SPI needs to be initialized before using these drivers.

* PPP server with example

* DHCP server moved out from lwip2 directories is now a per-interface instanciable class
  (and more easily improvable)

* Every services / libraries should work with ethernet 
  ~~- mDNS is [a work in progress](https://github.com/Labor-Et-Ars/ESP8266-Ardu [... cut]
closed 2020-12-22 by d-a-v