esp8266/Arduino #3279: WiFiEvent, disconnection not detected

Banzai00, 2017-05-23:
### Basic Infos

#### Hardware
Hardware:	      Wroom-02
Core Version:      	2.3.0-rc2

I have two Wroom-02 : one is in the soft AP mode and the other in station mode. And i want to detect when the second one connect or disconnect to/from the AP of the first one. So, i am using WiFiEvents example.
When i power on the wroom 02 in station mode, the onStationConnected works and the connection is detected. But my problem is when i power off the wroom 02, the onStationDisconnected don't get called and the disconnection is not detected.
This problem only occur with two esp. When, i connect and disconnect using a smartphone it works great.
The code of the esp in ap mode : 
 
```
#include <ESP8266WiFi.h>
#include <stdio.h>

const char *ssid = "APssid";
const char *password = "APpassword";

WiFiEventHandler stationConnectedHandler;
WiFiEventHandler stationDisconnectedHandler;

void setup() {
  Serial.begin(115200);

  // Don't save WiFi configuration in flash - optional
  WiFi.persistent(false);

  // Set up an access point
  WiFi.mode(WIFI_AP);
  WiFi.softAP(ssid, password);

  // Register event handlers.
  // Callback functions will be called as long as these handler objects exist.
  // Call "onStationConnected" each time a station connects
  stationConnectedHandler = WiFi.onSoftAPModeStationConnected(&onStationConnected);
  // Call "onStationDisconnected" each time a station disconnects
  stationDisconnectedHandler = WiFi.onSoftAPModeStationDisconnected(&onStationDisconnected);  
}

void onStationConnected(const WiFiEventSoftAPModeStationConnected& evt) {
  Serial.print("Station connected: ");
  Serial.println(macToString(evt.mac));
}

void onStationDisconnected(const WiFiEventSoftAPModeStationDisconnected& evt) {
  Serial.print("Station disconnected: ");
  Serial.println(macToString(evt.mac));
}

void loop() {
  
}

String macToString(const unsigned char* mac) {
  char buf[20];
  snprintf(buf, sizeof(buf), "%02x:%02x:%02x:%02x:%02x:%02x",
           mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
  return String(buf);
}

```
The code of the esp in station mode : 

```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>

IPAddress staticIP(192,168,4,222);
IPAddress gateway(192,168,4,1);
IPAddress subnet(255,255,255,0);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println();
  
  WiFi.mode(WIFI_STA);
  WiFi.begin("APssid", "APpassword");
  WiFi.config(staticIP, gateway, subnet);
  Serial.print("Connecting");
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");   
    
  }
  Serial.println();

  Serial.print("Connected, IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  // put your main code here, to run repeatedly:

}

```

igrr, 2017-05-23:
Question, when you disconnect the smartphone, how do you do this? Do you pull the battery out of the smartphone?

Banzai00, 2017-05-23:
No, i simply connect to another Wifi network. 
I tried to close my phone without disconnecting from the AP. And i got the same behavior (like with the 2 esp) and the disconnection was not detected.
So, it seems that if the disconnection is not done properly, the esp can't detect it ???

Banzai00, 2017-05-23:
After some test, i realized that the esp do detect the disconnection but approximately 5 minutes after i power off the phone or the other esp.

igrr, 2017-05-23:
Right. Because when you switch to another network on the mobile phone, it first tells its current AP (in a simplified terms) "hey AP, i'm leaving". Then the AP knows that the client is about to disconnect. On the other hand if you yank the power cord, the client doesn't have time to notify the AP that it's leaving. There is a certain timeout that AP has before it will deauthenticate the station which hasn't sent even an empty frame in a while. This timeout depends on the AP. AFAIK, on the ESP8266 this timeout can not be modified by the application. If you need this feature, please post the enhancement request on bbs.espressif.com, asking to support this in the non-OS SDK.

closed 2017-05-23 by igrr

FelixStiepermann, 2018-03-07:
Sry to comment on a closed issue, but i use the onSoftAPModeStationConnected function in exactly the same way, but the build fails with the error message 

> "Description	Resource	Path	Location	Type
> no matching function for call to 'ESP8266WiFiClass::onSoftAPModeStationConnected(void (*)())'	ESP_RTC_WRITEREAD.cpp	/ESP_Tests	line 28	C/C++ Problem"