esp8266/Arduino #3284: Why http POST form allows only 32 postArgs elements?

drschlaumeier, 2017-05-23:
### Basic Infos

My Webserver Configuration Page needs to POST a form with >32 elements.  Why  ESP8266WebServer::_parseForm allows only 32 postArgs elements? 

#### Hardware
Hardware:	ESP-8285
Core Version:      2.4.0 ...

### Description
I'm building a WebServer around my home light control based on Sonoff Touch device. The flexible configuration page of the webserver uses a html page with form and tables filled with some postArgs elements (some input - type text, some select, some input - type checkbox).  All together I have 37 postArgs elements to POST back to my WebServer. After  long debugging the esp CRASH I found that the ESP8266WebServer::_parseForm allows only 32 postArgs elements. ... see below code from "Parsing.cpp"
Why only 32? Can someone change to higher value? Any other idea what I can do? Is splitting the config page into several pages the right way?

Thanks in advance

```
bool ESP8266WebServer::_parseForm(WiFiClient& client, String boundary, uint32_t len){
  (void) len;
#ifdef DEBUG_ESP_HTTP_SERVER
  DEBUG_OUTPUT.print("Parse Form: Boundary: ");
  DEBUG_OUTPUT.print(boundary);
  DEBUG_OUTPUT.print(" Length: ");
  DEBUG_OUTPUT.println(len);
#endif
  String line;
  int retry = 0;
  do {
    line = client.readStringUntil('\r');
    ++retry;
  } while (line.length() == 0 && retry < 3);

  client.readStringUntil('\n');
  //start reading the form
  if (line == ("--"+boundary)){
    RequestArgument* postArgs = new RequestArgument[32];
    int postArgsLen = 0;
    while(1){
      String argName;
      String argValue;
      String argType;
      String argFilename;
      bool argIsFile = false;

.......
```
### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB
CPU Frequency:	80Mhz
Flash Mode:		DOUT
Flash Frequency:	40Mhz
Upload Using:		OTA / SERIAL
Reset Method:		ck

################################################

igrr, 2017-05-24:
The value of 32 was chosen as a compromise between supporting forms with a lot of fields and having to allocate large postArgs array.

Possible solutions:

- introduce a run-time setting (e.g. `ESP8266WebServer::setMaxPostArgs(size_t)`)
- do a two-pass parsing: count the number of arguments on the first pass, allocate the array, fill it on the second pass
- replace an array with a vector which will grow dynamically as the arguments are added

drschlaumeier, 2017-05-24:
Many thanks. 
I personally prefer the third solution since i do not need to care on setting the max postArgs and not wasting time with two-pass. 
However, I manually changed to postArgs array to 64 for now. Waiting for enhancement ...

igrr, 2017-05-24:
Please feel free to submit a PR with the 3rd solution :)

drschlaumeier, 2017-05-24:
Hmm, I somehow expected that comment :-) 
I'm not a good c programmer but I can try ...The first solution is easy :-) The third is tricky ... e.g. to catch all the possible errors.
Can I PR the first solution ?

devyte, 2017-09-06:
@drschlaumeier using a std::vector is far easier than an array that gets allocated/deallocated. For this, you would need something along these lines:
#include <vector>
...

replace:
  int              _currentArgCount;
  RequestArgument* _currentArgs;

with
std::vector<RequestArgument> _currentArgs;

Don't init _currentArgs(0) in the constructor, don't alloc it anywhere, clear it wherever it gets deleted.

Replace references to _currentArgCount with _currentArgs.size().
 
When a new RequestArgument is added, do something like this:
_currentArgs.emplace_back(key, value);

Likely some additional changes, but the above should cover the important stuff.

Be aware that there is currently a bug in Arduino.h that makes #include <vector> not compile sometimes. To make certain:
#ifdef min
#undef min
#endif
#ifdef max
#undef max
#endif
#include <vector>

yoursunny, 2018-08-18:
> Be aware that there is currently a bug in Arduino.h that makes #include not compile sometimes. To make certain:
> #undef min
> #undef max

I wonder why `Arduino.h` `#define`s min and max, instead of using standard library's version?
```
#include <algorithm>
using std::min;
using std::max;
```

devyte, 2018-08-18:
@yoursunny that has already been fixed, and 3rd party libs have also been fixed to comply.