esp8266/Arduino #3286: ESP12F Zombie mode after deepSleep

entropia1ac, 2017-05-24:
Hardware
suggested a 4.7k pull up each for pin RST, CH_PD,0,2  and pull down for pin 15 as insurance.
Hardware: ESP8266 E12F
Core Version: 2.3.0
USB /TTL 3,3V  - CP2101
Upow=3,3V/2A

From time to time (between xx hours and a few days)  the chip isn't reliably returning from deep sleep mode ,the board does not restart automatically
During this mode, the GPIO0 is oscillating from 0.0 to 3.3 V at 26 MHz 

### Settings in IDE

Module:  			ESP8266 E12E
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		 SERIAL
Reset Method:		ck 

Sketch
void setup()
{
Serial.begin(9600);
delay(100);

}

void loop()
{
delay(1000);
ESP.deepSleep(1000000, WAKE_RF_DEFAULT); // Sleep for 1 second

}

 Debug Messages
Zombie mode:
23:51:12.917>  ets Jan  8 2013,rst cause:0, boot mode:(3,6)
23:51:12.917> 
23:51:12.917> unknown reset
ets_main.c 

when board  restart automatically OK:
23:51:09.547>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
23:51:09.547> 
23:51:09.547> load 0x4010f000, len 1384, room 16 
23:51:09.609> tail 8
23:51:09.609> chksum 0x2d
23:51:09.609> csum 0x2d
23:51:09.609> v09f0c112

igrr, 2017-05-24:
See http://bbs.espressif.com/viewtopic.php?t=562, how do you connect GPIO16 to RST?

entropia1ac, 2017-05-24:
-GPIO16 connect  only to nRESER pin
- CHIP_PD connect to  pin1 4,7k  ,pin2 4,7k connect to Vcc=3,3V

entropia1ac, 2017-06-01:
Update to SDK2.1.0 solved the  issue.

closed 2017-06-01 by entropia1ac