esp8266/Arduino #3289: The problem about sprintf or string of 2.4.0-rc1 lead the system hang and reboot

countrysideboy, 2017-05-24:
float humi;
sprintf(message, "%s%", String(humi, 0).c_str());
----------------------------------------------------
In the version of 2.3.0, the code above works ok, but when i switch to 2.4.0-rc1, the system hang and reboot.

igrr, 2017-05-24:
```c++
void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println();
  delay(1000);
  char message[50] = {0};
  float val;
  sprintf(message, "%s%", String(val, 0).c_str());
  Serial.println(message);
}

void loop() {
}
```
The following doesn't crash in my test. Can you give the comlete code which shows the issue?

countrysideboy, 2017-05-24:
Thanks for your prompt reply.
I think my problem maybe related to stack overflow.
And I temporarily solve my problem with strcpy(message, String(humi, precision).c_str()).
I try to use the code below to compare v2.3.0 with v2.4.0-rc1.
`char message[50] = {0};
float val = 66.66;

void displayHumi(float humi, int precision)
{
  if (precision > 0) sprintf(message, "%s%", String(humi, precision).c_str());
  else sprintf(message, "^%s%", String(humi, precision).c_str());
}

void setup() {
  Serial.begin(9600);
  Serial.println();
  Serial.println();
  Serial.println("Thanks!");//"Thanks!" should print onece
  delay(1000);
}

void loop() {
  displayHumi(val, 0);
  Serial.println(message);
  delay(1000);
}`
"Thinks!" shoud print onece, but it prints with "message" every time.(version 2.3.0 works ok)

devyte, 2017-10-24:
@countrysideboy will you be posting a clean MCVE sketch that reproduces your issue?