esp8266/Arduino #3291: ESP8266 writing jsonObject to FFS returns empty List {}

Pfannex, 2017-05-24:
Hi,
I want to combine FFS and json in one Class.
So far it works as expected.
Right now I'm hanging while writing back the jsonObject to the FFS.

Every time I call the `void FFSjsonFile::writeItem(String itemName, String value)` function my json-File returns an empty ObjectList --> `{}`.

Where I'm wrong?
Did anybody got an idea?

Greatings
Pf@nne

**config.json**
```c++
{
  "webUser":        "ESPuser",
  "webPassword":    "ESPpass",
  "apName":         "ESP8266_1032096",
  "apPassword":     "ESP8266config",
  "wifiSSID":       "Pf@nne-NET",
  "wifiPSK":        "xxxxx",
  "wifiIP":         "",
  "mqttServer":     "192.168.1.203",
  "mqttPort":       "1883",
  "mqttDeviceName": "ESP8266_1032096",
  "updateServer":   "192.168.1.3",
  "filePath":       "/bin/ESP8266_AktSen.ino.bin"
}     
```
**FFS.h**
```c++
#pragma once
//FileSystem
  #include <FS.h>                     //this needs to be first, or it all crashes and burns...
  #include <ArduinoJson.h>
  #include <Arduino.h>
  #include "Setup.h"
  #include "I2C.h"
 
//===============================================================================
//  jsonFile
//===============================================================================
class FFSjsonFile{
public:  
  FFSjsonFile(String filePath, int type);
  String filePath;
  size_t size;                               //filled in readJsonString()
  int type;                                  //0=Object; 1=Array
  String root;
 
  String readItem(String itemName);
  String readItem(int item);
  void writeItem(String itemName, String value);

private: 
  String readJsonString();
  DynamicJsonBuffer JsonBuffer;
  JsonObject& jsonRootObject;
  JsonArray& jsonRootArray;

  void parseJsonObject(JsonObject& jsonObject);
  void parseJsonArray(JsonArray& jsonArray);
};
 

//===============================================================================
//  FFS 
//===============================================================================
class FFS{
public:
  FFS(I2C& i2c);
  I2C& i2c;
  
  FFSjsonFile cfg;
  FFSjsonFile testArray;

  void TEST();  
  
private:
};

```
**FFS.cpp**
```c++
#include "FFS.h"

//===============================================================================
//  FFS 
//===============================================================================
FFS::FFS(I2C& i2c):
    i2c(i2c),
    cfg(CFG_PATH, TYPE_OBJECT),
    testArray(TESTARRAY_PATH, TYPE_ARRAY){

}

//===> TEST <------------------------------------------------------
void FFS::TEST(){
  Serial.println(cfg.readItem("apName"));
  Serial.println(cfg.readItem("wifiSSID"));
  Serial.println(cfg.readItem("wifiPSK"));

  //cfg.writeItem("wifiSSID", "HelloWorld");
  //Serial.println(cfg.readItem("wifiSSID"));
}
  

//===============================================================================
//  FFSjsonFile public
//===============================================================================
FFSjsonFile::FFSjsonFile(String filePath, int type) :
    filePath       (filePath),
    type           (type),
    root           (readJsonString()),
    jsonRootObject (JsonBuffer.parseObject(root)),
    jsonRootArray  (JsonBuffer.parseArray(root)){
}

//===> readItem from jsonString <-------------------------------------
String FFSjsonFile::readItem(String itemName){
  return jsonRootObject[itemName].asString();
}

String FFSjsonFile::readItem(int item){
  return jsonRootArray[item].asString(); 
}

//===> writeItem to jsonString <---------------------------------------
void FFSjsonFile::writeItem(String itemName, String value){
  jsonRootObject[itemName] = value;

  SPIFFS.begin();
  File jsonFile = SPIFFS.open(filePath, "w");
  if (!jsonFile) {
    Serial.println("failed to open File for writing");
    //Serial.print("format file System.. ");
    //SPIFFS.format();
  }

  jsonRootObject.printTo(Serial);
  jsonRootObject.printTo(jsonFile);
  jsonFile.close(); 
}

//  FFSjsonFile private 
//===============================================================================

//===> Read json String from File <------------------------------------
String FFSjsonFile::readJsonString(){    
  File jsonFile;
  String jsonData = "NIL";
 
  if (SPIFFS.begin()) {
    Serial.println("mounted file system...OK");
    if (SPIFFS.exists(filePath)) {
      jsonFile = SPIFFS.open(filePath, "r");
      if (jsonFile) {
        size = jsonFile.size();
        std::unique_ptr<char[]> buf(new char[size]);

        jsonFile.readBytes(buf.get(), size);
        if (type == TYPE_ARRAY){
        //Array 
          DynamicJsonBuffer _jsonBufferArray;
          JsonArray& jsonArray = _jsonBufferArray.parseArray(buf.get());             
          if (jsonArray.success()) {
            char buffer[size];
            jsonArray.printTo(buffer, sizeof(buffer));
            jsonData = String(buffer); 
          }
        }else{
        //Object
          DynamicJsonBuffer _jsonBuffer;
          JsonObject& json = _jsonBuffer.parseObject(buf.get());
          if (json.success()) {         
            char buffer[size];
            json.printTo(buffer, sizeof(buffer));
            jsonData = String(buffer);
          }
        }
      }
    }
  }
  jsonFile.close();
  return jsonData;  
}

//#########################################################################################

void FFSjsonFile::parseJsonObject(JsonObject& jsonObject){
  //Serial.println("parseJsonObject");
  for (auto &element : jsonObject){
    String strKey = element.key;
    String strValue = element.value;
    Serial.print(strKey);Serial.print("      ");Serial.println(strValue);
  }
}

void FFSjsonFile::parseJsonArray(JsonArray& jsonArray){
  for (auto &element : jsonArray){
    if (element.is<JsonArray&>()){
      Serial.println("-->>");//Serial.println(i);
      JsonArray& nestedArray = element;
      parseJsonArray(nestedArray);
    }else{
      Serial.println(element.asString()); //Serial.print(" | ");Serial.println(i);
    }
  }
}

```

devyte, 2017-05-25:
Hi,

This is not the right place to ask questions like "what's wrong with my
code" or "how do I blah". This issue tracker is meant for reporting issues
with the core code and for requesting/proposing enhancements.

Also, your code is about json, so you may wantwant toto look at the
arduinojson repository and the documentation there.

A quick look at your code shows a bunch of SPIFFS.begin()s, which isn't how
it's meant to be used. That call is meant for mounting the filesystem,
which means it should be called once at bootup, and that's it.
Then, you seem to be trying to serialize a json object. The arduinojson
already has low level serializer functions, at least for streams. I suggest
you look at their exemplars.

I suggest studying the documentation and examples of the relevant parts of
your code a bit more, testing out those sketches, and then if you still
have questions, ask in a discussion forum rather than an issue tracker.

Pfannex, 2017-05-25:
Sorry,
also the completely wrong github repository.....my fault

closed 2017-05-25 by Pfannex