esp8266/Arduino #3292: Can't receive udp broadcast packet from 255.255.255.255

bestpika, 2017-05-25:
### Basic Infos

#### Hardware
Hardware: NodeMCU 1.0
Core Version: 2.3.0

### Description

I send packet to 172.16.255.255, client can receive and print.
But I send packet to 255.255.255.255, client can't receive.

### Sketch

```cpp
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>

#define WIFI_STA_SSID "..."
#define WIFI_STA_PSWD "..."

const uint8_t ipAddr[] = { 172, 16, 63, 254 };
const uint8_t subnet[] = { 255, 255, 0, 0 };
const uint8_t bcAddr[] = { 255, 255, 255, 255 };
const int dstPort = 5000;

unsigned long thisMs = 0;

WiFiUDP udp;

void setup() {
  Serial.begin(115200);
  Serial.flush();
  Serial.setDebugOutput(true);
  Serial.println();
  Serial.println();

  WiFi.config(ipAddr, ipAddr, subnet);
  WiFi.begin(WIFI_STA_SSID, WIFI_STA_PSWD);
  thisMs = millis();
  while (!WiFi.isConnected() && (unsigned long)(millis() - thisMs) <= 10000) {
    Serial.print(".");
    delay(1000);
  }
  udp.begin(dstPort);
}

void loop() {
  int pktSize = udp.parsePacket();
  if (pktSize) {
    byte pktBuf[pktSize];
    Serial.print(udp.remoteIP());
    Serial.print(":");
    Serial.println(udp.remotePort());
    udp.read(pktBuf, pktSize);
    udp.stop();
    udp.begin(dstPort);
    for (int i = 0; i < pktSize; i++) {
      Serial.print(pktBuf[i], HEX);
      Serial.print("\t");
    }
    Serial.println();
  }
}
```

igrr, 2017-05-25:
So the ESP8266 and PC are connected to the same router?
I might be mistaken, but in think that in this case packets sent to 255.255.255.255 will not be forwarded to the ESP by the router. Quoting Wikipedia,

> A special definition exists for the IP broadcast address 255.255.255.255. It is the broadcast address of the zero network or 0.0.0.0, which in Internet Protocol standards stands for this network, i.e. the local network. Transmission to this address is limited by definition, in that it is never forwarded by the routers connecting the local network to other networks.

Makuna, 2017-05-25:
@igrr If both the ESP and the computer both have the same local network ip addressing (lets say in this case 172.16.*.*); then they are on the same local network and it "should" be forwarded.  BUT, I have personally found many low cost WiFi routers (probably based on the same faulty code) have problems following UDP broadcast standards.  One such would only forward the first few broadcast packets and then it would stop.  Use a different WiFi router and it worked fine.

bestpika, 2017-05-26:
I try a NodeMCU send to 255.255.255.255, then client can receive.
But I don't known why...

closed 2017-05-26 by bestpika

alex0freeman, 2019-02-03:
WiFi.config(ipAddr, ipAddr, subnet);