esp8266/Arduino #3294: ESP8266 Upload issue: espcomm_upload_mem failed

architmuchhal12, 2017-05-25:
Hello,

I am using a **NodeMCU 1.0 (ESP - 12E Module)** with Arduino 1.2.1 and Windows 7. The board was working fine until it started showing up the following error. I referred the troubleshooting guide completely, but could not resolve the error. 

```
esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 921600
	setting port from COM1 to COM6
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
An error occurred while uploading the sketch
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
Uploading 228336 bytes from C:\Users\ARCHIT~1.MUC\AppData\Local\Temp\arduino_build_777326/Blink.ino.bin to flash at 0x00000000
	erasing flash
	size: 037bf0 address: 000000
	first_sector_index: 0
	total_sector_count: 56
	head_sector_count: 16
	adjusted_sector_count: 40
	erase_size: 028000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting serial port timeouts to 15000 ms
	read 0, requested 1
	setting serial port timeouts to 1000 ms
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
closing bootloader
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
error: espcomm_upload_mem failed

```

I can't even load a basic Blink program.
Please help me regarding the necessary steps that should be taken.

Thanks

devyte, 2017-05-26:
Your baudrate looks too high. Try with 115kbps. Not all boards work well
with high baudrates, and even those that do when new can degrade over time.

architmuchhal12, 2017-05-26:
I dd try uploading the program using all the possible baud rate values. Yet, the problem still persists. 
Is it a board failure? Do I need to change the board?

Thanks

closed 2017-06-02 by architmuchhal12

JesusCarrascoCarrillo, 2017-09-19:
Al final no se como se ha solucionado este problema, yo tengo el mismo problema y me gustaria saber si se ha resuelto de alguna forma. Gracias.