esp8266/Arduino #3297: Help a newbie

joshuakaleb0424, 2017-05-26:
### Basic Infos
 I am using a hardware from  [http://www.e-gizmo.com/KIT/esp11shield.html](url). I am new at using arduino and i badly need to use this wifi shield.
#### Hardware
Hardware:			ESP-11
Core Version:      	??

### Description

I am using a arduino clone called gizduino and the wifi shield sits on top of it.  
Since it sits on top of a arduino I dont know which board in the board menu to choose from.

pipi61, 2017-05-26:
Try generic

joshuakaleb0424, 2017-05-30:
Sorry for the late reply! It returns an error when compiling :(

warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

duncan-a, 2017-05-30:
If the shield is on top of an Arduino (clone) then, presumably the ESP on
the shield is communicating with the Arduino using AT Commands - i.e. you
don't need to program it, you just need to program the Arduino to make use
of it. In which case, you need to select the Arduino board option that's
applicable to the gizduino.

What exactly are you trying to do?

joshuakaleb0424, 2017-05-31:
the arduino clone (gizduino) is communicating with the ESP8266 module using SoftwareSerial. 
I like your library it seems that it'll help me with my project.

lrmoreno007, 2017-05-31:
As duncan-a told you, your board makes use of the AT commands, which is another platform totally different from this. Read:
Https://drive.google.com/drive/folders/0BxdLxDCD6HidLTczRGVfRjQwaHc

To guide you a little: there are several programming platforms for ESP8266:
AT commands (default in your board)
Arduino
LUA
MicroPython
Smart.js
...

You use the ESP8266 attached to an arduino clone, so the communication between both is very special. The "normal" thing in these platforms is to program the ESP8266 directly and not through an Arduino board.  In the previous manual explains how to install the platform LUA and in a similar way can install this platform. But before doing so make sure you do it and why, and how to go back.

You will need to learn many concepts and this is not the best place to learn them. For help of this type is the web: www.esp8266.com 

Greetings.

joshuakaleb0424, 2017-05-31:
Thanks guys! Sorry if i didn't understand it earlier >.< Thanks for the tips though! 👍  :D

closed 2017-05-31 by joshuakaleb0424