esp8266/Arduino #3298: Error in documentation for ESP8266WiFiGenericClass::hostByName

UlliBien, 2017-05-26:
### Basic Infos

#### Hardware
Core Version:      	2.3.0

### Description
Ducomentation says:

>  * @return 1 if aIPAddrString was successfully converted to an IP address,
>  *          else error code

In case of an error 0 is returned. No error code

devyte, 2017-09-06:
Where in the docs? Or do you mean the comments in the source?

UlliBien, 2017-09-06:
Sorry for my bad explanation.
It's in ESP8266WiFiGeneric.cpp, line 410 ff.

```
/**
 * Resolve the given hostname to an IP address.
 * @param aHostname     Name to be resolved
 * @param aResult       IPAddress structure to store the returned IP address
 * @return 1 if aIPAddrString was successfully converted to an IP address,
 *          else error code
 */
int ESP8266WiFiGenericClass::hostByName(const char* aHostname, IPAddress& aResult) {
...
```

The code of the method ends with
`    return (err == ERR_OK) ? 1 : 0;`

which returns 1 or 0, but not an error code.

earlephilhower, 2018-11-08:
Fixed via #5325

closed 2018-11-08 by earlephilhower