esp8266/Arduino #3303: ESPHTTPUpdate: HTTP_UPDATE_FAILD Error (-1): HTTP error: connection refused

malebuffy, 2017-05-28:
### Basic Infos

#### Hardware
Hardware:			WEMOS D1 Mini 
Core Version:      	         2.3.0-rc2

### Description

HTTP_UPDATE_FAILD Error (-1): HTTP error: connection refused

### Settings in IDE

Module:  			WEMOS D1 Mini 
Flash Size: 		4M (3M Spiffs)
CPU Frequency:	80Mhz
Upload Using:		SERIAL

### Debug Messages

HTTP_UPDATE_FAILD Error (-1): HTTP error: connection refused

### Sketch (Part of it)

```
void httpupdate() {
    webupdate = 1;
    if((WiFi.status() == WL_CONNECTED)) {

        t_httpUpdate_return ret = ESPhttpUpdate.update("http://localhost/file.bin");
        //t_httpUpdate_return  ret = ESPhttpUpdate.update("https://server/file.bin");

          switch(ret) {
            case HTTP_UPDATE_FAILED:
                USE_SERIAL.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
                break;

            case HTTP_UPDATE_NO_UPDATES:
                USE_SERIAL.println("HTTP_UPDATE_NO_UPDATES");
                break;

            case HTTP_UPDATE_OK:
                USE_SERIAL.println("HTTP_UPDATE_OK");
                break;
        }
    }
}

```

Hello,

I have a problem updating my WEMOS board over ESPhttpUpdate. I am getting a connection refused message although the webserver is local and  normal http. If I try to use a webhosting webserver I get the same error. Both http and not https.

I am calling this function after a MQTT message that tells the ESP to update. This however works in a way. I came across it when searching for a solution.

```
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();

  WiFi.begin("testbox1", "testbox123");
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
  }
  Serial.println("Connected, updating");
  delay(1000);

  t_httpUpdate_return ret = ESPhttpUpdate.update("https://raw.githubusercontent.com/Zyxmn/esp8266_BlinkyBin/master/Blinky.cpp.bin", "", "21 99 13 84 63 72 17 13 B9 ED 0E 8F 00 A5 9B 73 0D D0 56 58");
}
void loop() {
}
```
Although I am not 100% sure. The Serial Monitor has this info:

```
scandone
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 

connected with dlink-6E30, channel 11
dhcp client start...
ip:192.168.0.165,mask:255.255.255.0,gw:192.168.0.1
Connected, updating
please start sntp first !
cert FP: CC AA 48 48 66 46 0E 91 53 2C 9C 7C 23 2A B1 74 4D 29 9D 33 
test FP: 21 99 13 84 63 72 17 13 B9 ED 0E 8F 00 A5 9B 73 0D D0 56 58 

```

So why does the simple http doenst work and the https works?

malebuffy, 2017-05-28:
I have some new info. If I use the previous mentioned code:

```
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();

  WiFi.begin("dlink-6E30", "abymu80918");
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
  }
  Serial.println("Connected, updating");
  delay(1000);

  t_httpUpdate_return ret = ESPhttpUpdate.update("http://localhost/file.bin");
}
void loop() {
}
```

 and use the local webserver address, I get to the point where the Serial monitor says:

```
scandone
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 

connected with dlink-6E30, channel 11
dhcp client start...
ip:192.168.0.165,mask:255.255.255.0,gw:192.168.0.1
Connected, updating
pm open,type:2 0
```

and hangs there.

devyte, 2017-05-28:
Your core version says 2.3.0-rc2, which is old, prior to 2.3.0 (rc means
release candidate). Please try with 2.3.0, or a more recent version, like a
github pull.

or mute
the thread

malebuffy, 2017-05-28:
Unfortunately still same error. Removed 2.3.0 RC and installed 2.3.0. Same error

HTTP_UPDATE_FAILD Error (-1): HTTP error: connection refused

devyte, 2017-05-28:
Are you saying that updating from an external server over https works, but
from a local server over http does not?

devyte, 2017-05-28:
Also, did you try the latest github code? While 2.3.0 is the latest stable,
it's still pretty old.

malebuffy, 2017-05-28:
No basically with 2.3.0 I always get the same -1 error.

Will try 2.4.0, however using PC an I am not sure about the steps. Unzipping everything to a folder 2.4.0 where 2.3.0 lies works or do I have to do something else as well?

devyte, 2017-05-28:
There's a link on how to use git from the Readme page. I can't link it
here, I'm on my phone on a roadtrip and it won't work for some reason.

malebuffy, 2017-05-28:
Installed 2.4.0 Still nothing. Same error.

malebuffy, 2017-05-28:
Anyone has a simple code or example where this works? I have to rule out one parameter one by one....

malebuffy, 2017-05-31:
Was the fileserver and insufficient power from ftdi.

closed 2017-05-31 by malebuffy

fischerjade, 2019-04-10:
I don't know if this helps but I had the same error.
This worked for me.  I had changed the http.begin() to use my endpoint.  I didn't add the SSL fingerprint though.
Old code read as 
http.begin("https://192.168.1.12/test.html", "7a 9c f4 db 40 d3 62 5a 6e 21 bc 5c cc 66 c8 3e a1 45 59 38");

I found my fingerprint by adding the full domain at this website.

https://www.grc.com/fingerprints.htm.  Just enter this in the second argument for the http.begin() method like

http.begin("https://yahoo.com","DE:9E:8E:5B:C0:D9:AD:A9:E8:C4:68:F4:76:DD:57:75:12:90:EB:D6").

rs-benatti, 2021-01-29:
Localhost is only accessible on the machine that is hosting, so if your computer is hosting a website you can't access it via localhost with yout ESP.

stevem21m, 2021-12-08:
I know this is old, and closed, but thought this might be useful. In the original post, the serial log output is: 
----------
please start sntp first !
cert FP: CC AA 48 48 66 46 0E 91 53 2C 9C 7C 23 2A B1 74 4D 29 9D 33 
test FP: 21 99 13 84 63 72 17 13 B9 ED 0E 8F 00 A5 9B 73 0D D0 56 58 
---------

This log is telling you the fingerprint is wrong. "test FP" is your fingerprint, but the server fingerprint, "cert FP," is different. You need to update the fingerprint to the servers Fingerprint. 

To get this level of logging, Arduino > tools > SSL+TLS_MEM+HTTP_CLIENT+HTTP_SERVER. 

Hope this helps someone.