esp8266/Arduino #3304: Trouble using strcat_P()

milkpirate, 2017-05-28:
This isnt really an issue its more a request for support. I would like to do the following:

```
const char* flashString PROGMEM = "flashString to print";
...

char buff[32];
uint16_t num = 1234;

sprintf_P(buff, PSTR("%s=%u %s"), F("value1"), num, flashString);
Serial.println(buff);
```
But I couldnt get it to work. I am sure it is possible but I am not sure about all these `PGM_P`, `PSTR`, `FPSTR`, `F` and so on (the only thing I'm sure is `PSTR` at the beginning of the format string) but with the rest I dont know where to put which? Could you help me out with some hints?

Makuna, 2017-05-29:
The core thing I believe you are looking for is the ability to use a PROGMEM string as a argument value to be embedded by the format string.   With AVR, if the string you are embedding is PROGMEM, the format qualifier is capital S, not s.   I am not sure this is supported for Esp though.
```
sprintf_P(buff, PSTR("%S=%u %S"), F("value1"), num, flashString);
```

pieman64, 2017-05-29:
@Makuna so there are definitely some qualifiers that are not supported by the ESP then?

Over the last few months I have tried and failed to use the float qualifier.

I finally cracked it yesterday with a hack to use integers so 33.1234567 becomes integer 33 a dot and integer 1234567. Not nice but it works (remembering to cover 0 padding for numbers like 33.0123456).

Makuna, 2017-05-29:
use ftoa instead of printf for floats.  I don't think AVR Arduino supports floats either in printf/sprint.

Try the upper case S and let us know if it works ;-)

milkpirate, 2017-05-29:
Nope. Doesnt work (strangely it doesnt even correctly print out the unsigned int): 

```%S=1076053752 %S```

@pieman64 as far as I know the Arduino `printf` on the ESP is just kind of a wrapper for the `os_printf` function from Espressif, which lacks the ability to print floats. You need to use the Arduinos own functions as Makuna suggeted.

igrr, 2017-05-29:
With the latest version, floats should be supported, but uppercase S is certainly not supported. Printf and friends are provided by newlib library.

milkpirate, 2017-06-05:
@igrr is it possible to add the, like in the initial post, desired functionality to `Xprintf_P` / `Xprintf`?

igrr, 2017-06-05:
It is possible, here's the part of newlib code responsible for '%s' formats:
https://github.com/igrr/newlib-xtensa/blob/xtensa-2_2_0-lock-arduino/newlib/libc/stdio/nano-vfprintf_i.c#L213

You would need to handle the case of 'S' and then copy the string into the output buffer in 4-byte units. or use pgm_read_byte in a loop.

closed 2017-06-05 by milkpirate

milkpirate, 2017-09-28:
@igrr I'm now in the mood to try to implement the functionality... not sure if my knowledge will suffice. so here what I have figured out so far and made up myself by the `case s:` part:

```
	...
case 's':
	cp = GET_ARG (N, *ap, char_ptr_t);			// get ptr to next argument
	char *p = memchr (cp, 0, pdata->prec);			// find \0 in cp-pointed str until (max num of chars)=pdata->prec
	if (p != NULL) pdata->prec = p - cp;			// compute string size
	pdata->size = pdata->prec;				// write size to pdata
	goto non_number_nosign;
case 'S':
	cp = (char*)GET_ARG (N, *ap, char_ptr_t);		// get ptr to next argument (guess its const char)

	char *p = (char*)memchr_P(cp, 0, pdata->prec);		// find \0 in cp-pointed str, cast neccessary for next lines?
	if (p != NULL) pdata->prec = p - cp;   			// compute string size
	pdata->size = pdata->prec;			 	// write size to pdata

	// shouldnt the following be equivalent to the above 3 lines?
	//pdata->size = strlen_P(cp);				// get length of \0 term. string
	//pdata->prec = pdata->size;

	pdata->l_buf[0] = '\0';
	n = _printf_common (data, pdata, &realsz, fp, pfunc);
	if (n == -1)
		goto error;
	while(pgm_read_byte( (const char*) cp))
		PRINT(pgm_read_byte( (const char*) cp++));
	if (pdata->flags & LADJUST)
		PAD (pdata->width - realsz, pdata->blank);

	return (pdata->width > realsz ? pdata->width : realsz);
default:
	...
```

Cant see where the buffer is, you were talking about. Where is `PRINT` defined in line [241](https://github.com/igrr/newlib-xtensa/blob/e6743695d217592a298c51ecb410cfab1dcdc167/newlib/libc/stdio/nano-vfprintf_i.c#L241)? Does it use the whole char array or just a single char? Would the above code work? What do you think?