esp8266/Arduino #3306: DHT11 + ESP8266 D1 mini - accuracy of readings ?

cvbkf, 2017-05-28:
### Basic Infos

#### Hardware
Hardware:			?ESP-8266 D1 mini + DHT11 Shield?
Core Version:      	?2.3.0?

### Description

I'm working on some wireless, battery-powered temperature sensors and currently i'm trying different types of sensors. I tried a simple sketch from the web with the DHT11 sensor, and noticed some weird behavior while graphing the values with munin. (sensor is read every 5 minutes). while the ds18b20 sensors show some nice curves without big aberrations, the dht11 readings are "jumping" not very accurate. is this usual ? i know that the DHT11 has a accuracy ~2°C , so maybe i'll try the DHT22. Or is it some problem with the sketch or the library ? 

### Settings in IDE

Module:  			?WeTek D1 mini?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?default?
Flash Frequency:	?default?
Upload Using:		?SERIAL?
Reset Method:		?default?

### Sketch

```/*********
  Rui Santos
  Complete project details at http://randomnerdtutorials.com  
*********/

// Including the ESP8266 WiFi library
#include <ESP8266WiFi.h>
#include "DHT.h"

// Uncomment one of the lines below for whatever DHT sensor type you're using!
#define DHTTYPE DHT11   // DHT 11
//#define DHTTYPE DHT21   // DHT 21 (AM2301)
//#define DHTTYPE DHT22   // DHT 22  (AM2302), AM2321

// Replace with your network details
const char* ssid = "the.asylum.guest";
const char* password = "test#1234";

// Web Server on port 80
WiFiServer server(80);

// DHT Sensor
const int DHTPin = 2;
// Initialize DHT sensor.
DHT dht(DHTPin, DHTTYPE);

// Temporary variables
static char celsiusTemp[7];
static char fahrenheitTemp[7];
static char humidityTemp[7];

// only runs once on boot
void setup() {
  // Initializing serial port for debugging purposes
  Serial.begin(115200);
  delay(10);

  dht.begin();
  
  // Connecting to WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  
  // Starting the web server
  server.begin();
  Serial.println("Web server running. Waiting for the ESP IP...");
  delay(10000);
  
  // Printing the ESP IP address
  Serial.println(WiFi.localIP());
}

// runs over and over again
void loop() {
  // Listenning for new clients
  WiFiClient client = server.available();
  
  if (client) {
    Serial.println("New client");
    // bolean to locate when the http request ends
    boolean blank_line = true;
    while (client.connected()) {
      if (client.available()) {
        char c = client.read();
        
        if (c == '\n' && blank_line) {
            // Sensor readings may also be up to 2 seconds 'old' (its a very slow sensor)
            float h = dht.readHumidity();
            // Read temperature as Celsius (the default)
            float t = dht.readTemperature();
            // Read temperature as Fahrenheit (isFahrenheit = true)
            float f = dht.readTemperature(true);
            // Check if any reads failed and exit early (to try again).
            if (isnan(h) || isnan(t) || isnan(f)) {
              Serial.println("Failed to read from DHT sensor!");
              strcpy(celsiusTemp,"Failed");
              strcpy(fahrenheitTemp, "Failed");
              strcpy(humidityTemp, "Failed");         
            }
            else{
              // Computes temperature values in Celsius + Fahrenheit and Humidity
              float hic = dht.computeHeatIndex(t, h, false);       
              dtostrf(hic, 5, 2, celsiusTemp);             
              float hif = dht.computeHeatIndex(f, h);
              dtostrf(hif, 5, 2, fahrenheitTemp);         
              dtostrf(h, 5, 2, humidityTemp);
            }
            client.println("HTTP/1.1 200 OK");
            client.println("Content-Type: text/html");
            client.println("Connection: close");
            client.println();
            // your actual web page that displays temperature and humidity
            client.print(celsiusTemp);
            client.print(";");
            client.print(humidityTemp);
            break;
        }
        if (c == '\n') {
          // when starts reading a new line
          blank_line = true;
        }
        else if (c != '\r') {
          // when finds a character on the current line
          blank_line = false;
        }
      }
    }  
    // closing the client connection
    delay(1);
    client.stop();
    Serial.println("Client disconnected.");
  }
}   
```

### Debug Messages

fake-name, 2017-05-29:
The DHT11 only supports temperature sensing in [1°C increments](https://github.com/adafruit/DHT-sensor-library/blob/master/DHT.cpp#L41). You're basically seeing the quantization effects of the sensor.

The only way to get around this is to buy a better sensor.

cvbkf, 2017-05-29:
Ah, thanks for clarifying that. I didn't notice that when reading the specs. So by the code i assume the DHT22 is better ? Or maybe i'll stay with the DS18B20.

beicnet, 2017-05-29:
@cvbkf The DHT22 sensor is way better than the DHT11 sensor.

cvbkf, 2017-05-29:
Ok, thank you for your help & input. Ordering one for testing now.

closed 2017-05-29 by cvbkf