esp8266/Arduino #3308: Unable to make http requests from esp8266 (STA) to esp8266 (AP)

renarsvilnis, 2017-05-30:
Hi,

I have a setup where I have 2x esp8266-01 where on is a STA (client) and the other one is an AP with a web server on port 80.

I'm unable to debug with serial monitor, this is why I built debugging endpoints on the AP web server - that marks if requests were made, how many clients are connected to WiFI etc.

I'm only unable to make any HTTP requests to a web server if it's on the AP (192.168.4.1:80)

I connected to the AP network created a node.js web server and was able to make requests to it from the esp8266 client. I also successfully was able to make requests to the esp8266 AP server through node.js.

Is there a technical reason it doesn't work? configuration? What alternatives could I try out? I only need one-way communication - client --> server.

## Client code

```cpp
#include <ESP8266WiFi.h>

const char* WIRELESS_SSID = "invisible-env";
const char* WIRELESS_PASSWORD = "kakiskakis";
const char* SERVER_HOST = "192.168.4.1";
const int   SERVER_PORT = 80;
const char* SERVER_URL  = "/";

void setup() {
  WiFi.mode(WIFI_STA);

  WiFi.begin(WIRELESS_SSID, WIRELESS_PASSWORD);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
}

void loop() {
  // Open connection
  WiFiClient client;
  if (!client.connect(SERVER_HOST, SERVER_PORT)) {
    delay(1000);
    return;
  }

  client.print(
    String("GET ") + SERVER_URL + " HTTP/1.1\r\n" +
    "Host: " + SERVER_HOST + ":" + SERVER_PORT + "\r\n" +
     "Connection: close\r\n" +
     "\r\n"
  );

  unsigned long timeout = millis();
  while (client.available() == 0) {
    if (millis() - timeout > 5000) {
      client.stop();
      // End loop as we don' lister for respond of the message
      return;
    }
  }

  delay(1000);
}
```

## Server code

```cpp
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

const char* WIRELESS_SSID = "invisible-env";
const char* WIRELESS_PASSWORD = "kakiskakis";

ESP8266WebServer server(80);

void handleRoot() {
  String message = "Hello world!";
  server.send(200, "text/plain", message);
}

void setup() {
  WiFi.softAP(WIRELESS_SSID, WIRELESS_PASSWORD);

  server.on("/", handleRoot);
  // Other routes..
  server.begin();
}

void loop(void) {
  server.handleClient();
}
```

devyte, 2017-09-08:
@renarsvilnis both pieces of code look correct at first glance, so this should work. I know you ended up with a workaround, but please try with latest git and report back here.

devyte, 2017-10-10:
No feedback in over 30 days, closing.

closed 2017-10-10 by devyte

renarsvilnis, 2017-10-10:
I eventually solved it.

Solved by writing the entire thing from scratch (no copy-pasting) part-by-part. And came to a realization, that the issue was with my code. I opened the HTTP connection with the client, then did other work for 2-3 seconds and only then sent the data. By moving the connection after the processing data, everything worked. I guess the connection timeous out, as nothing is sent for a few seconds.