esp8266/Arduino #3310: Feature Request: Update sketch and Spiffs over ethernet (W5100)

danriches, 2017-05-30:
Hi,
I use an ESP8266-07 with external antenna and with a W5100 WizNet Ethernet chip and need to be able to update the sketch and SPIFFS via Ethernet as well. Is there an easy way to implement this or is there going to be any updates to include this functionality??

Thanks,

Dan

devyte, 2017-05-30:
Sketch is already possible to update over wifi, either from an onboard
webserver or via http client (or other in house method you could
implement). Ethernet shoukd bee the same thing.

SPIFFS is being added now, or was added very recently. Not sure what the
current status is.

devyte, 2017-09-06:
In the latest git, the Updater class supports updating the SPIFFS (U_FLASH argument, per Updater.h). I'm not sure about frontends, such as a webserver module, but at least the basic functionality is there. 
Closing.

closed 2017-09-06 by devyte

devyte, 2017-09-06:
For completeness, the capabality is also added to ESP8266HTTPUpdateServer in PR #3234, although at the time of this writing, it is not merged yet.