esp8266/Arduino #3311: Is possible to User I2C in ISR ? (Exception (0):)

far5893, 2017-05-30:
Hi,
i'm testing this simple sketck

`#include "Arduino.h"
#include <Wire.h>
#include <FS.h>
#include <EEPROM.h>
#include <bma180.h>

BMA180 bma180;

volatile byte newData=0;
volatile unsigned int countISR=0;
unsigned int served=0;

void BMAISR(void)
{
   countISR++;
   newData=1;
   bma180.readAccel();
}

//The setup function is called once at startup of the sketch
void setup()
{
  delay(5000);
  EEPROM.begin(4096);
  Serial.begin ( 115200 );
  Serial.println("Start....");
  delay(3000);

  Wire.begin(12,13); //I2C pin
    Wire.setClockStretchLimit(400000);
    bma180.SetAddress((int)BMA180_DEFAULT_ADDRESS);
    bma180.SoftReset();
    bma180.enableWrite();
    int sversion;
    int id;
    bma180.getIDs(&id,&sversion);
    Serial.print("Id = ");
    Serial.print(id,DEC);
    Serial.print(" v.");
    Serial.println(sversion,HEX);
    bma180.SetFilter(bma180.F40HZ);
    bma180.setGSensitivty(bma180.G1);
    //bma180.SetResolution(bma180.R14BIT);
    bma180.SetSMPSkip();
    bma180.SetISRMode();
    bma180.disableWrite();
    bma180.readAccel();
    pinMode(4, INPUT);
    attachInterrupt(4, BMAISR, RISING);
    Serial.println("End Setup");

}

// The loop function is called in an endless loop
void loop()
{   static unsigned int loopCount,loopLast;

//newData = 1;
 //   Serial.print("Pin4: ");
 //   Serial.print(digitalRead(4));
  //  Serial.print(" ISR CALL:");
  //  Serial.println(countISR);

  if (newData)
    {
        newData=0;
        served++;
        if (served % 2==0) // skip some data...
        {
            Serial.print(" [");
            Serial.print(bma180.x,DEC);
            Serial.print(" ");
            Serial.print(bma180.y,DEC);
            Serial.print(" ");
            Serial.print(bma180.z,DEC);
            Serial.print("] ");
            Serial.println(countISR);
        }
    }
   #ifdef DEBUGON
    if (millis() > tAlive)
    {
       bma180.readAccel();
        Serial.print("ISR's=");
        Serial.print(countISR);
        Serial.print(" served=");
        Serial.println(served);
        Serial.print("t=");
        Serial.print(bma180.temp);
        Serial.print("[");
        Serial.print(bma180.x,DEC);
        Serial.print(" ");
        Serial.print(bma180.y,DEC);
        Serial.print(" ");
        Serial.print(bma180.z,DEC);
        Serial.println("]");

       tAlive=millis()+5000;
       Serial.print("filter reg=");
       Serial.println(bma180.getRegValue(0x20),BIN);
       Serial.print("status_reg1=");
       Serial.println(bma180.getRegValue(0x09),BIN);
       Serial.print("status_reg2=");
       Serial.println(bma180.getRegValue(0x0A),BIN);
       Serial.print("status_reg3=");
       Serial.println(bma180.getRegValue(0x0B),BIN);
       Serial.print("status_reg4=");
       Serial.println(bma180.getRegValue(0x0C),BIN);
       Serial.print("ctrl_reg0=");
       Serial.println(bma180.getRegValue(0x0d),BIN);
       Serial.print("ctrl_reg1=");
       Serial.println(bma180.getRegValue(0x0e),BIN);
       Serial.print("ctrl_reg2=");
       Serial.println(bma180.getRegValue(0x0f),BIN);
       Serial.print("ctrl_reg3=");
       Serial.println(bma180.getRegValue(0x21),BIN);
       Serial.print("ctrl_reg4=");
       Serial.println(bma180.getRegValue(0x22),BIN);
   }
   #endif
   delay(10);

   loopCount++;
//Add your repeated code here
}`

but when attach bma180 i obtain:
`0x40106b4c: interrupt_handler at /home/monition/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_wiring_digital.c line 119
0x40106b14: interrupt_handler at /home/monition/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_wiring_digital.c line 113
0x402020a4: BMAISR() at /home/monition/Arduino/bma180seism/bma180seism.ino line 19
0x40106bb0: interrupt_handler at /home/monition/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_wiring_digital.c line 135
0x40106b9a: interrupt_handler at /home/monition/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_wiring_digital.c line 131
0x4010103d: ets_timer_disarm at ?? line ?
0x40106b14: interrupt_handler at /home/monition/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_wiring_digital.c line 113
0x4021b77c: pm_get_sleep_type at ?? line ?
0x401015c6: spi_flash_read at ?? line ?
0x40106fec: pvPortZalloc at /home/monition/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/heap.c line 33
0x4021b24d: pm_set_sleep_time at ?? line ?
0x4021b6e2: pm_get_sleep_type at ?? line ?
0x4021b78f: pm_get_sleep_type at ?? line ?
0x40203d05: ets_timer_handler_isr at ?? line ?
0x40203d4a: ets_timer_handler_isr at ?? line ?
0x40202d37: loop_task at /home/monition/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_main.cpp line 56
`
from : 
`Exception (0):
epc1=0x4020207c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc30 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffdd0:  40106b4c 00000000 3ffedcd8 0000000a  
3ffffde0:  ffffffff 00000020 00000001 3ffee317  
3ffffdf0:  00000000 00000000 00000000 00000022  
3ffffe00:  3fffc200 40106b14 3fffc258 4000050c  
3ffffe10:  4000437f 00000030 00000016 ffffffff  
3ffffe20:  60000200 00000008 00000000 80000000  
3ffffe30:  20000000 3fff1558 80000000 203fc040  
3ffffe40:  80000000 3fffc6fc 3ffe8d8c 3fff155c  
3ffffe50:  000002b4 003fc040 60000600 00000030  
3ffffe60:  e0037035 00000004 00000000 402020a4  
3ffffe70:  40106bb0 00080000 3fffc200 40106b9a  
3ffffe80:  ffffffff 00000020 3ffefdbc 00000000  
3ffffe90:  00000000 00000000 0000001f 4010103d  
3ffffea0:  4000050c 40106b14 3fffc258 4000050c  
3ffffeb0:  40000f68 00000030 0000001c ffffffff  
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  00000013 4021b77c 3ffe8d8c 00000001  
3ffffee0:  ffffffff 3ffe9214 3ffe8d8c 3fffdab0  
3ffffef0:  00000000 3fffdcb0 3ffe8928 00000030  
3fffff00:  00000000 400042db 00000064 60000600  
3fffff10:  40004b31 3fff151c 000002f4 003fc000  
3fffff20:  401015c6 3ffe8910 3ffe90c0 40106fec  
3fffff30:  4021b24d 3ffe90c0 3ffe8910 05cbea8a  
3fffff40:  3fff151c 00001000 4021b6e2 00000008  
3fffff50:  00000000 00000000 4021b78f 3ffe9174  
3fffff60:  3ffe8910 01cf7e25 3ffe8910 60000600  
3fffff70:  40203d05 3ffe9174 3ffe8910 05cbdc43  
3fffff80:  40203d4a 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffe8930 3fffdad0 3ffee468 40202d37  
3fffffa0:  40000f49 00028048 3fffdab0 40000f49  
<<<stack<<<
`

library function read does  :  
`unsigned int BMA180::readAccel()
{
  unsigned int result;

  Wire.beginTransmission(address);
  Wire.write(0x02);
  Wire.endTransmission();
  Wire.requestFrom((int)address, 7);
  if(Wire.available()==7)
  {
    int lsb = Wire.read()>>2;
    int msb = Wire.read();
    x=(msb<<6)+lsb; 
    if (x&0x2000){
    	x= x-0b11111111111111-1;

    }
    lsb = Wire.read()>>2;
    msb = Wire.read();
    y=(msb<<6)+lsb;
    if (y&0x2000){
    	y= y-0b11111111111111-1;
    }
    lsb = Wire.read()>>2;
    msb = Wire.read();
    z=(msb<<6)+lsb;
    if (z&0x2000){
    	z= z-0b11111111111111-1;

    }
    temp = Wire.read();
    if (temp&0x80) temp|=0xff00;
  }
  result = Wire.endTransmission();
  return result;
}`

devyte, 2017-05-30:
Without looking too closely at this, I can say: you shouldn't be doing
complex or lengthy operations from an ISR. An ISR should be as tiny as
possible, because its execution preempts everything else.

Set a flag or something within the ISR, and then check the flag within the
loop() to handle your comms.

devyte, 2017-09-06:
Closing this due to user error (short answer is no, you can't).

closed 2017-09-06 by devyte