esp8266/Arduino #3315: Same source different binaries compiling on macOS Sierra and Windows

piflav, 2017-05-31:
Hi
compiling the very same source on macOS and Windows I get different binaries and while the output on Windows works pefectly the one on macOS crash with exception (9) on Wroom-2 (PRODino Board).
Both  Arduino/esp8266 enviroment are fresh and updated installations.

Investigating I found the crash happens on line 463 of SPI.cpp:

`*fifoPtr = *dataPtr;`

calling void SPIClass::transferByte.
The source reports:
```
/**
 * Note:
 *  in and out need to be aligned to 32Bit
 *  or you get an Fatal exception (9)
 * @param out uint8_t *
 * @param in  uint8_t *
 * @param size uint32_t
 */
```

Is that the problem?
Why does it happen on macOS and not on Windows?

Thanks
Flavio

piflav, 2017-06-05:
Can you please suggest further investigations?
Thanks
Flavio

igrr, 2017-06-05:
Do you have the minimal sketch required to reproduce this issue? Please attach it to the first post, along with the exception output decoded using ESPExceptionDecoder tool.

igrr, 2017-06-19:
Based on the backtrace, indeed that library is passing a non-aligned pointer to SPI::transferBytes.

I would call this a bug in SPI library, because on other Arduino platforms transferBytes doesn't have such alignment restrictions.

d-a-v, 2018-10-02:
Fixed by #4925

closed 2018-10-02 by d-a-v