esp8266/Arduino #3316: 10 ms ack delay

Jorgegarve, 2017-05-31:
Yesterday I sniffed the communication between my raspberry server (sending TCP packets with 1460 bytes data, so a fully complete TCP packet) and my ESP8266 which receives them and I saw that every ack the ESP8266 send is delayed more or less 10 ms (time from the last TCP packet received). Can someone explain me why or how to fix it? I need it to be inmediate ACK.

The attached file shows what I am talking about. In delta time column you can see these around 10 ms of delay, and you can also see a delay of 100 ms aprox which I don't really know why it is there...

Thanks for all.

[image]

igrr, 2017-05-31:
Could you post the code you have on the ESP8266 side?

Jorgegarve, 2017-05-31:
It is difficult to post the code because I am working with things related to patents and it is a huge code. Which part would you like to see?

igrr, 2017-05-31:
Well, since you are reporting an issue I was wondering whether you have an [mcve](https://stackoverflow.com/help/mcve) to provide along with the report... Since the issue is about receiving TCP data, i suppose that the example would contain some loop in which you receive data using WiFiClient.

WiFiClient instance will not acknowledge incoming packets until the application calls its `read` (or `readBytes`) method to extract the received data. Once the data is taken by the application, WiFiClient informs TCP stack that given amount of data has been received and the TCP stack can now send an ACK. Depending on what the application does in between the reads, there may be some delay before the next chunk of data is 'read' by the application.

Also take a look at [espasynctcp](https://github.com/me-no-dev/espasynctcp) library which allows you to acknowledge data reception explicitly.

Jorgegarve, 2017-05-31:
Ok, I understand, so the first fast fix to this issue should be to use espasyntcp, am I wrong? How does the asynctcp manage acks? In my code I use a WiFiClient for all the communications and the loop starts with:

void loop() {
  
  if ( client.available() ) { 
      c = client.read();
      //actions to do

Thanks in advance for being so fast!

igrr, 2017-05-31:
I'm pretty sure that if you read 1 byte at a time, it will take a fairly long time to consume the whole 1460 byte segment (only then will the ACK be sent). Using `int WiFiClient::read(uint8_t *buf, size_t size)` to read the whole segment would be more efficient. I suggest you do some benchmarking of your code, to see how much time it takes your `loop` to consume 1460 bytes from `client`.

Jorgegarve, 2017-05-31:
So, for example, in order to read the fully 1460 bytes tcp packet data stored only once, I can do:

c = client.read(buffer, 1460);

Thanks for all the support, and thanks for this amazing work.

Jorgegarve, 2017-06-04:
So, igrr, is there a way to force checking for a complete received TCP packet without losing the information in order to send ack? I mean, when you use read() if the TCP packet is completely received, you told me that you will send an ack. But this c = client.read() will take the byte received, and you will be forced to use it somewhere. I am looking towards something that force the ACK to be sent or just checked.

igrr, 2017-06-04:
If you want to send an ACK before consuming (processing) the data, that's fine — you can buffer it by doing `client.read(buffer, 1460)`. That places data into a buffer owned by the application, and tells the TCP stack: "data has been passed to the application layer; application layer is now responsible for whatever happens to the data". At this point TCP stack can acknowledge delivery by sending an ACK and increasing the receive window.
We can't let the stack increase receive window before data is passed to the application, as that would defeat TCP flow control and allow the remote end to flood the ESP until it runs out of RX buffers.

Jorgegarve, 2017-06-05:
Thanks for the answer, it is very self-explanatory! I'm gonna try it and I will tell you something! I'm just afraid of consuming 1460 bytes in order to get a full TCP data packet when the full TCP packet is not received yet... Could this happen?

Jorgegarve, 2017-06-05:
Also, for a better adaptability, should I store before the return from howManyBytes = client.available() and use it in client.read(buffer, howManyBytes)?

igrr, 2017-06-05:
You can query `available()` first, sure. Also if you request to `read` certain number of bytes but there are less bytes available, the bytes which are actually available will be put into the buffer and the number of bytes will be returned by `read` function.

```c++
size_t bytes_read = client.read(buffer, TCP_MSS);
if (bytes_read > 0) {
    // do something with the contents of buffer
}
```

Jorgegarve, 2017-06-05:
Now I want to check if I receive what I should receive, but I get instead this error:

> Exception (29):
epc1=0x4000dfd9 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff0680 end: 3fff0890 offset: 01a0

>>>stack>>>
3fff0820:  401059a8 001614fb 3ffef6e4 00000000  
3fff0830:  4020722d 3ffef6e4 3ffef870 3ffef6e4  
3fff0840:  00000000 3ffef860 4020a734 3ffef870  
3fff0850:  402010ae 000003e8 000003e8 3ffef860  
3fff0860:  3fffdad0 3ffef2d0 3ffef330 402067fe  
3fff0870:  3fffdad0 00000000 3ffef858 4020a780  
3fff0880:  feefeffe feefeffe 3ffef870 40100718  
<<<stack<<<

My code is the following:
`void loop() {
  
  if (client.available() > 0) {
    client.read(rxBuffer, sizeBuffer);
    Serial.println((char)* rxBuffer);
  }

  delay(1000);

}`

igrr, 2017-06-05:
Could you please decode the stack dump and paste the result here? Thanks.
https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html#decode

Also please check if the data you receive actually contains a null-terminated string. `client.read` will not add a null character at the end of the buffer for you.

Jorgegarve, 2017-06-05:
The data I receive does not contain a null at the end, it is just a huge amount of pictures compressed in JPEG, so the TCP data will contain "random data", but never terminated in null. I though when you call client.read(buffer, size) it will read an amount of size bytes from the TCP buffer.

igrr, 2017-06-05:
`client.read` is no problem, but what do you expect `Serial.println` to do when you pass it a buffer full of binary data?

Jorgegarve, 2017-06-05:
I expect to check if I receive a 0xffd8 (beginning of the JPEG file).

Jorgegarve, 2017-06-05:
Decoding 7 results
0x4020a580: esp_yield at C:\Users\Pc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x402071fd: ClientContext::read(char*, unsigned int) at C:\Users\Pc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
:  (inlined by) WiFiClient::read(unsigned char*, unsigned int) at C:\Users\Pc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 230
0x4020a5f8: __yield at C:\Users\Pc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x4020a638: optimistic_yield at C:\Users\Pc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x402067ef: loop at C:\Users\Pc\Dropbox\TFM_JorgeGarciaVega\Proyecto\HUDmjpegViewer_readModified/HUDmjpegViewer_readModified.ino line 252
0x4020a5cc: loop_wrapper at C:\Users\Pc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\Pc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109

igrr, 2017-06-06:
> I expect to check if I receive a 0xffd8 (beginning of the JPEG file).
Please post the full sketch which exhibits the issue.

Jorgegarve, 2017-06-06:
`#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP.h>
#include <SPI.h>
#include <SSD_13XX.h>
#include <JPEGDecoder.h>

//Pin definitions for display
#define __CS1  2  //(D4)
#define __DC  5   //(D1)
#define TFT_BLACK 0
#define MAXSIZE_ARRAYJPEG  20000
// Return the minimum of two values a and b
#define minimum(a,b)     (((a) < (b)) ? (a) : (b))

// WiFi information
const char WIFI_SSID[] = "******";
const char WIFI_PSK[] = "*******";
const char http_site[] = "******";
const int http_port = 8080;

// Pin definitions
const int LED_PIN = 2;

// Global variables
WiFiClient client;
SSD_13XX tft = SSD_13XX(__CS1, __DC);
uint32_t tTime, tTimeImage;
String requestPageCommand;
//Debugging
int flag = 0;
uint8_t arrayJpeg[MAXSIZE_ARRAYJPEG];
uint32_t arrayJpeg_currentSize;
uint8_t* rxBuffer;
size_t sizeBuffer = 1460;
boolean decoded;
int state=0;   //  0: init,    1: 0xFF received   2: 0xFF 0xD8 received

//Functions

//====================================================================================
//   Plots information about the image received
//====================================================================================

void jpegInfo() {
}

//====================================================================================
//   Decode and render the Jpeg image onto the TFT screen
//====================================================================================
void jpegRender(int xpos, int ypos) {

  // retrieve infomration about the image
  uint16_t  *pImg;
  uint16_t mcu_w = JpegDec.MCUWidth;
  uint16_t mcu_h = JpegDec.MCUHeight;
  uint32_t max_x = JpegDec.width;
  uint32_t max_y = JpegDec.height;

  // Jpeg images are draw as a set of image block (tiles) called Minimum Coding Units (MCUs)
  // Typically these MCUs are 16x16 pixel blocks
  // Determine the width and height of the right and bottom edge image blocks
  uint32_t min_w = minimum(mcu_w, max_x % mcu_w);
  uint32_t min_h = minimum(mcu_h, max_y % mcu_h);

  // save the current image block size
  uint32_t win_w = mcu_w;
  uint32_t win_h = mcu_h;

  // record the current time so we can measure how long it takes to draw an image
[... 149 lines cut ...]

  requestPageCommand = String("GET /?action=stream HTTP/1.1\r\nHost: ") + http_site + "\r\n" +
               "Connection: keep-alive\r\n\r\n";

  requestPage();

  //requestPageCommand = String("GET /xxx1.jpg HTTP/1.1\r\nHost: ") + http_site + "\r\n";

  
  Serial.println("First page requested");

} // End of setup()

void loop() {

if (client.available() > 0) {
client.read(rxBuffer, sizeBuffer);
Serial.println((char)* rxBuffer);
}

delay(1000);

}`

We are not using the full code because I am just trying to receive correctly the data.

Jorgegarve, 2017-06-06:
```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP.h>
#include <SPI.h>
#include <SSD_13XX.h>
#include <JPEGDecoder.h>

//Pin definitions for display
#define __CS1  2  //(D4)
#define __DC  5   //(D1)
#define TFT_BLACK 0
#define MAXSIZE_ARRAYJPEG  20000
// Return the minimum of two values a and b
#define minimum(a,b)     (((a) < (b)) ? (a) : (b))

// WiFi information
const char WIFI_SSID[] = "******";
const char WIFI_PSK[] = "*******";
const char http_site[] = "******";
const int http_port = 8080;

// Pin definitions
const int LED_PIN = 2;

// Global variables
WiFiClient client;
SSD_13XX tft = SSD_13XX(__CS1, __DC);
uint32_t tTime, tTimeImage;
String requestPageCommand;
//Debugging
int flag = 0;
uint8_t arrayJpeg[MAXSIZE_ARRAYJPEG];
uint32_t arrayJpeg_currentSize;
uint8_t* rxBuffer;
size_t sizeBuffer = 1460;
boolean decoded;
int state=0;   //  0: init,    1: 0xFF received   2: 0xFF 0xD8 received

//Functions

//====================================================================================
//   Plots information about the image received
//====================================================================================

void jpegInfo() {
}

//====================================================================================
//   Decode and render the Jpeg image onto the TFT screen
//====================================================================================
void jpegRender(int xpos, int ypos) {

  // retrieve infomration about the image
  uint16_t  *pImg;
  uint16_t mcu_w = JpegDec.MCUWidth;
  uint16_t mcu_h = JpegDec.MCUHeight;
  uint32_t max_x = JpegDec.width;
  uint32_t max_y = JpegDec.height;

  // Jpeg images are draw as a set of image block (tiles) called Minimum Coding Units (MCUs)
  // Typically these MCUs are 16x16 pixel blocks
  // Determine the width and height of the right and bottom edge image blocks
  uint32_t min_w = minimum(mcu_w, max_x % mcu_w);
  uint32_t min_h = minimum(mcu_h, max_y % mcu_h);

  // save the current image block size
  uint32_t win_w = mcu_w;
  uint32_t win_h = mcu_h;

[... 151 lines cut ...]
  requestPageCommand = String("GET /?action=stream HTTP/1.1\r\nHost: ") + http_site + "\r\n" +
               "Connection: keep-alive\r\n\r\n";

  requestPage();

  //requestPageCommand = String("GET /xxx1.jpg HTTP/1.1\r\nHost: ") + http_site + "\r\n";

  
  Serial.println("First page requested");

} // End of setup()

void loop() {

if (client.available() > 0) {
client.read(rxBuffer, sizeBuffer);
Serial.println((char)* rxBuffer);
}

delay(1000);

}
```

We are not using the full code, just trying to check if we receive data correctly...

igrr, 2017-06-06:
It seems that you aren't in fact initializing `rxBuffer` to point to a valid buffer before calling client.read.

Jorgegarve, 2017-06-06:
I am currently at work, so I will try to solve it this afternoon. Again, thanks for the amazing support and work you do, and sorry about my insufficient knowledge...

Jorgegarve, 2017-06-06:
I am sorry but I am not able to make it work. I don't find any example over the internet in order to have some examples to work around with. I just want to read the full TCP packet and save it in a variable and make things with these bytes, but I just can't do it.

Jorgegarve, 2017-06-08:
I finally did it yesterday. It works like a charm... My NodeMCU is able to manage the packets so fast that I could even increase frame rate of my project. Thanks for the support, igrr, and if someone needs some help about using client.read(buffer, size); just say it.

Keep strong!

closed 2017-09-06 by devyte