esp8266/Arduino #3317: Flash exported bin file

DcloudGit, 2017-05-31:
Hi all! I'm able to upload a sketch from the Arduino IDE, but if I export the bin file and flash it with the Espressif tool: https://espressif.com/en/support/download/other-tools?keys=&field_type_tid%5B%5D=14
it won't works. These are my settings: 
[image]
[image]

I knew that there's the esptool.py but why this does not work with the official tool? Am I doing something wrong?

Thank you

reaper7, 2017-05-31:
Your FLASH SIZE setting is wrong
4Mbytes flash must be set as 32Mbit

DcloudGit, 2017-06-06:
!!!What a stupid mistake!
now it's working... thanks :-)

closed 2017-09-06 by devyte