esp8266/Arduino #3318: exception(29)

BurakYasin, 2017-06-01:
### Basic Infos

#### Hardware
Hardware:			ESP-01
Core Version:      	2.3.0-rc2

### Description

Exception (29) when I call the client.connect(host, port);

As I debug from the wificlientsecure.cpp, It seems to error from this
while (millis() - t < timeout_ms && ssl_handshake_status(_ssl) != SSL_OK) {
            uint8_t* data;
            int rc = ssl_read(_ssl, &data);
            if (rc < SSL_OK) {
                break;
            }
        }
Could you help in this exception ?

I tried this also in version 2.3.0, 2.2.0 stable versions and 2.3.0-rc2 and rc1 versions. But still error...

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512kB/64kB
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck 

### Sketch

#include <ArduinoJson.h>
#include <SoftwareSerial.h>
SoftwareSerial esp8266(0,1); //RX,TX
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <WiFiClientSecure.h>
#include <ESP8266WebServer.h>
#include <Ticker.h>
#include <EEPROM.h>
#include <WiFiUdp.h>
#include "helpers.h"
#include "global.h"
#include "stdio.h"

const char* hostInt= "**.com.tr";
const int httpPort2 = 443;
WiFiClientSecure client;

void setup() {
  Serial.begin(115200);
  delay(500);
  Serial.println("Starting ESP8266");
  esp8266.begin(115200);
}

void loop() {
 if(Serial.available()){
      dataLength=Serial.readBytes(processedData,999);
      Serial.println("dataLength:");
      Serial.println(dataLength);
      for(i=0;i<192;i++){ 
        Serial.println(processedData[i]);
      }
  sendData=1;
  }
 if(sendData){
    DynamicJsonBuffer JSONbuffer;
    JsonObject& root = JSONbuffer.createObject();
  
    JsonArray& data2 = root.createNestedArray("Data");
    for(unsigned int i=0;i<150;i++){
      data2.add(processedData[i]);
    }

    Serial.println("ok1");
    if(!client.connect(hostInt,httpPort2)){
      Serial.println("not respond");
      return;
    }else{
      Serial.println("respond");
      String out;
      root.printTo(out);

      client.println("POST /ApplianceRecognition/ApplianceRecognitionService.svc/Save HTTP/1.1"); 
      client.println("Host: ***.com.tr");
      client.println("Content-Type: application/json");
      client.println("Cache-Control: no-cache");
      client.println();
      client.println(out);
      Serial.println("POST /ApplianceRecognition/ApplianceRecognitionService.svc/Save HTTP/1.1"); 
      Serial.println("Host: seasdemo.innova.com.tr");
      Serial.println("Content-Type:application/json");
      Serial.println("Cache-Control:no-cache");
      Serial.println();
      Serial.println(out);
      Serial.println("Innova response:");
    }
    
    while(client.available()){
      String line = client.readStringUntil('\r');
      Serial.print(line);
    }
    Serial.println();
    Serial.println("closing connection");
}
[image]

}
```

### Debug Messages

Sketch uses 345,789 bytes (79%) of program storage space. Maximum is 434,160 bytes.
Global variables use 50,164 bytes (61%) of dynamic memory, leaving 31,756 bytes for local variables. Maximum is 81,920 bytes.

electronicsguy, 2017-06-02:
I've also gotten many errors exactly at the same point as you (more_comps or bi_divide in bigint.c). I wasn't able to narrow it down to the exact cause. However, timing seems to be a problem, after memory (RAM) consumption.  Try to reduce the number of operations you do within loop(). Also, insert a delay(1500) as the last line of the loop. experiment with the delay number. The loop needs free operations after sometime else the watchdog timer resets the chip.

aj-ptw, 2017-06-19:
Having the same issue... @electronicsguy i have the client secure in it's own function and so i'm confused why adding a delay to loop would cause any solution. This seems like an out of memory situation from using both ESPWebServer and WifiClientSecure at the same time.

electronicsguy, 2017-06-19:
@aj-ptw  the delay() in the loop helps return control to reset the watchdog timer. otherwise it reset the chip. Some discussion here: [https://github.com/esp8266/Arduino/issues/1532](https://github.com/esp8266/Arduino/issues/1532)

BurakYasin, 2017-06-20:
Actually I solved this not adding delay. I turn off the wifi when waiting data in serial bus. Also, I put the Esp on deep sleep and it work without sleep. I think problem was the energy (It's weird). Only problem is Esp connect my wifi too late and I am losing lots of time for this. But it works.

devyte, 2017-06-20:
I suggest timing the loop() body. You have a lot going on there, and there
is a limit on the total time you are allowed without servicing the wifi
stack (which happens after loop returns, on yield(), on delay() ).

Consider you're running the serial port at 115kbps. That's about 10KB/s, or
10bytes in 1ms. That's not even counting code overhead. Send a bunch of
msgs, and you go without servicing the wifi stack for way too long.

I would add yield()s or delay(1)s between each of those lines, see if that
helps.

devyte, 2017-09-08:
@BurakYasin I failed to understand your last comment when I posted my previous one. Now that I rearead it, I suspect the original issue was available mem.
Could you please post the modified sketch that works for you?

devyte, 2018-01-08:
Closing due to lack of feedback in several months.

closed 2018-01-08 by devyte