esp8266/Arduino #3324: AT command implementation location?

hoangnh0194, 2017-06-02:
Hi,
i'm trying to understand your lib, could you show me where 'd you used AT command in this lib.
thanks,

me-no-dev, 2017-06-02:
We do not use AT commands at all. We use the chip as Arduino, not as a WiFi chip attached through UART to another Arduino.

closed 2017-06-02 by me-no-dev

hoangnh0194, 2017-06-02:
@me-no-dev  :if we dont use AT command,how it can send/receive data through ESP8266,bro?

me-no-dev, 2017-06-02:
you receive it on the ESP8266. Imagine that you miraculously got a WiFi in your Arduino and that it somehow became 10 times faster. That is what ESP8266 is. It's a MCU with WiFi and you program it using the Arduino IDE the same way you do your other Arduinos

hoangnh0194, 2017-06-02:
oops,, finaly understand. thank you!