esp8266/Arduino #3327: Missing #include <avr/pgmspace.h> for Adafruit FONA

architmuchhal12, 2017-06-02:
I am using an Arduino ESP8266 along with Adafruit FONA 808,
While compiling, it gives me an error as - 

```
 #include <avr/pgmspace.h>

                          ^

compilation terminated
```.

Do I need to add this file? And if yes, where would I need to make changes?
Thanks!

pipi61, 2017-06-02:
[http://lmgtfy.com/?q=esp8266+avr%2Fpgmspace.h](url)
first in the list:
[https://github.com/SodaqMoja/Sodaq_DS3231/issues/5](url)

architmuchhal12, 2017-06-05:
@pipi61 Thanks a lot! It helped. I could compile the program.

After compiling the program to the ESP8266 board (NodeMCU 1.0), this is what I get on the Serial Monitor of Arduino IDE -

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
```
I tried to test the FONA 808 module by uploading the FONA_Test example by making the following changes in the code -

```
#define FONA_RX D7
#define FONA_TX D8
#define FONA_RST D4
```

I am not sure as to why I am seeing the output of that kind on the Serial Monitor.

Please help me in this regard.

Thanks!

devyte, 2017-09-05:
#1749 may help. Also, be very careful about which pins you use, and how you configure them, and with your power supply. I use a 5V 20A power supply for development (yes, that's 20 amps) to make sure that any resets or whatever aren't due to dips in supply voltage due to power spikes (and this is a cell module, so it could well have power spikes).

Having said that, this is not the correct place to ask for support questions regarding an external lib. This is an issue tracker, meant to track issues in the core libs. Please ask at esp8266.com or stackoverflow for assistance with getting the FONA code to work with the ESP.
If you eventually do find an issue in the core libs, such as a missing/incorrect function or #define, feel free to open a new issue with the explanation, and add a full minimal sketch that reproduces it, so that action can be taken.
Closing.

closed 2017-09-05 by devyte

Rhanem, 2018-03-25:
i downloaded this file from here :
https://github.com/mikalhart/galileo-Pgmspace.h/
next i edit the line to: #include <pgmspace.h>
is working..