esp8266/Arduino #3330: ssl_handshake_status not working on SDK 2.4.0-rc1

capedra, 2017-06-04:
Hello, I've been trying to use the sketch from the following URL on Wemos D1 Mini Pro: (https://github.com/electronicsguy/ESP8266/blob/master/HTTPSRedirect/GoogleDocs.ino)

It happens that this same sketch works well on SDK 2.3.0 but, on the other hand, on SDK 2.4.0-rc1 the SSL handshake doesn't work for the host script.googleusercontent.com when it's redirecting, because SSL.h is returning -1 when the function ssl_handshake_status is called on _connectSSL method.

Would like to know why, @igrr . Thanks.

igrr, 2017-06-04:
Can you set "debug level" to "Core + SSL", add `Serial.setDebugOutput(true);`, and post the resulting serial log?

electronicsguy, 2017-06-04:
@capedra  also, please post the sketch that you're using, in case you've made changes.

capedra, 2017-06-05:
@electronicsguy It's the same sketch from there. I didn't make any changes.

capedra, 2017-06-05:
@igrr Sure, here's the output:
`
Free heap before: 34880
unmodified stack   = 3888

Connecting to wifi: APJ15
sta config unchangedscandone
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 
.
connected with APJ15, channel 11
dhcp client start...
..............ip:192.168.254.76,mask:255.255.255.0,gw:192.168.254.254
.
WiFi connected
IP address: 
192.168.254.76
Connecting to script.google.com
:ref 1
please start sntp first !
State:	sending Client Hello (1)
:wr 104 104 0
:wrc 104 104 0
:sent 104
:rn 1452
:rd 5, 1452, 0
:rdi 1452, 5
:rd 74, 1452, 5
:rdi 1447, 74
State:	receiving Server Hello (2)
:rd 5, 1452, 79
:rdi 1373, 5
:rch 1452, 596
:rch 2048, 1452
:rch 3500, 596
:rch 4096, 80

:rd 4083, 4176, 84
:rdi 1368, 1368
:c 1368, 1452, 4176
:rdi 596, 596
:c 596, 596, 2724
:rdi 1452, 1452
:c 1452, 1452, 2128
:rdi 596, 596
:c 596, 596, 676
:rdi 80, 71
State:	receiving Certificate (11)
:rd 5, 80, 71
:rdi 9, 5
:rd 4, 80, 76
:rdi 4, 4
:c0 4, 80
State:	receiving Server Hello Done (14)
State:	sending Client Key Exchange (16)
:wr 267 267 0
:wrc 256 267 0
:wrc 11 11 0
:wr 6 6 0
:wrc 6 6 0
State:	sending Finished (16)
:wr 69 69 0
:wrc 69 69 0
:sent 267
:sent 75
:rn 75
:rd 5, 75, 0
:rdi 75, 5
:rd 1, 75, 5
:rdi 70, 1
:rd 5, 75, 6
:rdi 69, 5
:rd 64, 75, 11
:rdi 64, 64
:c0 64, 75
State:	receiving Finished (16)
domain name: 'script.google.com'
Certificate match.
:wr 357 357 0
:wrc 256 357 0
:wrc 101 101 0
:sent 357
:rn 1452
:rd 5, 1452, 0
:rdi 1452, 5
:rd 1447, 1452, 5
:rdi 1447, 1447
:c0 1447, 1452
:rn 596
:rch 596, 106
:rd 121, 702, 0
:rdi 596, 121
:wcs ra 1516
:ur 1
:close
:del
:ref 1
:wr 53 53 0
:wrc 53 53 0
Alert: close notify
please start sntp first !
State:	sending Client Hello (1)
:wr 115 115 0
:wrc 115 115 0
:sent 53
:rn 7
:rd 5, 7, 0
:rdi 7, 5
:rd 2, 7, 5
:rdi 2, 2
:c0 2, 7
Alert: decode error
Error: SSL error 50
:wr 7 7 0
:wrc 7 7 0
Alert::rcl
:abort
 unexpected message
Alert: close notify
Connection to re-directed URL failed!
Error! Not connected to host.

GET: Write into cell 'A1'
=========================
Error! Not connected to host.
Error! Not connected to host.

GET: Fetch Google Calendar Data:
================================
Error! Not connected to host.
Error! Not connected to host.

Series of GET and POST requests
===============================
Free heap: 32984
unmodified stack   = 2032
:ur 1
:del
:ref 1
please start sntp first !
State:	sending Client Hello (1)
:wr 104 104 0
:wrc 104 104 0
:sent 104
:rn 1452
:rd 5, 1452, 0
:rdi 1452, 5
:rd 74, 1452, 5
:rdi 1447, 74
State:	receiving Server Hello (2)
:rd 5, 1452, 79
:rdi 1373, 5
:rch 1452, 596
:rch 2048, 1452
:rch 3500, 596

:rd 4012, 4096, 84
:rdi:rch 4096, 80
 1368, 1368
:c 1368, 1452, 4176
:rdi 596, 596
:c 596, 596, 2724
:rdi 1452, 1452
:c 1452, 1452, 2128
:rdi 596, 596
:c 596, 596, 676
:rd 71, 80, 0
:rdi 80, 71
State:	receiving Certificate (11)
:rd 5, 80, 71
:rdi 9, 5
:rd 4, 80, 76
:rdi 4, 4
:c0 4, 80
State:	receiving Server Hello Done (14)
State:	sending Client Key Exchange (16)
:wr 267 267 0
:wrc 256 267 0
:wrc 11 11 0
:wr 6 6 0
:wrc 6 6 0
State:	sending Finished (16)
:wr 69 69 0
:wrc 69 69 0
:sent 267
:sent 75
:rn 75
:rd 5, 75, 0
:rdi 75, 5
:rd 1, 75, 5
:rdi 70, 1
:rd 5, 75, 6
:rdi 69, 5
:rd 64, 75, 11
:rdi 64, 64
:c0 64, 75
State:	receiving Finished (16)
:wr 357 357 0
:wrc 256 357 0
:wrc 101 101 0
:sent 357
:rn 1452
:rd 5, 1452, 0
:rdi 1452, 5
:rd 1447, 1452, 5
:rdi 1447, 1447
:c0 1447, 1452
:rn 121
:rch 121, 581
:rd 121, 702, 0
:rdi 121, 121
:c 121, 121, 702
:wcs ra 1516
:ur 1
:close
:del
:ref 1
:wr 53 53 0
:wrc 53 53 0
Alert: close notify
please start sntp first !
State:	sending Client Hello (1)
:wr 115 115 0
:wrc 115 115 0
:sent 53
:rn 7
:rd 5, 7, 0
:rdi 7, 5
:rd 2, 7, 5
:rdi 2, 2
:c0 2, 7
Alert: decode error
Error: SSL error 50
:wr 7 7 0
:wrc 7 7 0
Alert: un:rcl
:abort
expected message
Alert: close notify
Connection to re-directed URL failed!
GET Data from cell 'A1':
Error! Not connected to host.
POST append memory data to spreadsheet:
Error! Not connected to host.
:ur 1
:del
:ref 1
please start sntp first !
State:	sending Client Hello (1)
:wr 104 104 0
:wrc 104 104 0
:sent 104
:rn 1452
:rd 5, 1452, 0
:rdi 1452, 5
:rd 74, 1452, 5
:rdi 1447, 74
State:	receiving Server Hello (2)
:rd 5, 1452, 79
:rdi 1373, 5
:rch 1452, 596
:rch 2048, 1452
:rch 3500, 596
:rch 4096, 80

:rd 4083, 4176, 84
:rdi 1368, 1368
:c 1368, 1452, 4176
:rdi 596, 596
:c 596, 596, 2724
:rdi 1452, 1452
:c 1452, 1452, 2128
:rdi 596, 596
:c 596, 596, 676
:rdi 80, 71
State:	receiving Certificate (11)
:rd 5, 80, 71
:rdi 9, 5
:rd 4, 80, 76
:rdi 4, 4
:c0 4, 80
State:	receiving Server Hello Done (14)
State:	sending Client Key Exchange (16)
:wr 267 267 0
:wrc 256 267 0
:wrc 11 11 0
:wr 6 6 0
:wrc 6 6 0
State:	sending Finished (16)
:wr 69 69 0
:wrc 69 69 0
:sent 267
:sent 75
:rn 75
:rd 5, 75, 0
:rdi 75, 5
:rd 1, 75, 5
:rdi 70, 1
:rd 5, 75, 6
:rdi 69, 5
:rd 64, 75, 11
:rdi 64, 64
:c0 64, 75
State:	receiving Finished (16)
:wr 357 357 0
:wrc 256 357 0
:wrc 101 101 0
:sent 357
:rn 1452
:rd 5, 1452, 0
:rdi 1452, 5
:rd 1447, 1452, 5
:rdi 1447, 1447
:c0 1447, 1452
:rn 121
:rch 121, 581
:rd 121, 702, 0
:rdi 121, 121
:c 121, 121, 702
:wcs ra 1516
:ur 1
:close
:del
:ref 1
:wr 53 53 0
:wrc 53 53 0
Alert: close notify
please start sntp first !
State:	sending Client Hello (1)
:wr 115 115 0
:wrc 115 115 0
:sent 53
:rn 7
:rd 5, 7, 0
:rdi 7, 5
:rd 2, 7, 5
:rdi 2, 2
:c0 2, 7
Alert: decode error
Error: SSL error 50
:wr 7 7 0
:wrc 7 7 0
A:rcl
:abort
lert: unexpected message
Alert: close notify
Connection to re-directed URL failed!
GET Data from cell 'A1':
Error! Not connected to host.
POST append memory data to spreadsheet:
Error! Not connected to host.
Halting processor...
:ur 1
:del
Final free heap: 33536
Final unmodified stack   = 1776
state: 5 -> 0 (0)
rm 0
del if0
bcn 0
del if1
usl
`

electronicsguy, 2017-06-05:
@capedra  I have seen users get errors due to wifi/network configuration too, which is difficult to debug. Could you try using your setup with on a completely different network if possible, and report?

capedra, 2017-06-05:
@electronicsguy I just tested with another wifi network from a completely different ISP but the same problem persists. Maybe @cameronrich could help us either.

I think this part describes the problem:
```
Alert: decode error
Error: SSL error 50
:wr 7 7 0
:wrc 7 7 0
Aler:rcl
:abort
t: unexpected message
Alert: close notify
Connection to re-directed URL failed!
Error! Not connected to host.
```

igrr, 2017-06-05:
Thanks for the log, i think i understand what's going on now.

igrr, 2017-06-05:
This should be fixed in 8c3bb69. I wasn't able to verify this using GoogleDocs script, probably because I have not set something up on Google Docs side correctly. Nevertheless, in the simulated case when i call `WiFiClientSecure::connect` twice in a row, the second call no longer results in the connection being immediately aborted.

electronicsguy, 2017-06-05:
@igrr   Great work Ivan. What was the problem? By the way, you could use the googledocs script without any changes. It'll add lines to the (shared) spreadsheet in my account. You will still be able to see if it is able to make the connection.

igrr, 2017-06-05:
See description of the commit linked above for explanation. Ok, will try with the unmodified sketch.

igrr, 2017-06-05:
There's an explanation in the description of the commit linked above.

Tried again just to realize that connecting to script.google.com ain't going to work from China, for reasons entirely unrelated :) Please test latest master on your side, if possible.

electronicsguy, 2017-06-05:
@igrr   Sure I'll try. I use Arduino IDE on Windows 7. Is there a document explaining how I can load the git version side-by-side and switch to using that version for now?

igrr, 2017-06-05:
https://www.arduino.cc/en/Guide/PortableIDE
Follow this guide to create a "portable" installation of the IDE, so that it doesn't use the cores you have installed in your user directory. Then follow this part of Readme to install the latest git version:
https://github.com/esp8266/Arduino#using-git-version

This way, the portable installation will be entirely separate from the other copies of Arduino IDE in your system, and will use the version of the core installed inside its hardware directory.

capedra, 2017-06-06:
Hello, my friend Ivan. I can confirm that it's currently working after these changes that you've made on [8c3bb69](https://github.com/esp8266/Arduino/commit/8c3bb69530d24c638514d9e191906685e24e6d99). Thank you very much!

capedra, 2017-06-06:
Now the following issue happens from time to time:

```
Fatal exception 29(StoreProhibitedCause):
epc1=0x4020cc16, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000004, depc=0x00000000

Exception (29):
epc1=0x4020cc16 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: cont 
sp: 3fff0fb0 end: 3fff14a0 offset: 01a0

>>>stack>>>
3fff1150:  00820000 00000082 3fff85dc 4021914a  
3fff1160:  3fff0394 3fff89dc 3fff85dc 402192dc  
3fff1170:  00000040 00000082 3fff85c4 4021955d  
3fff1180:  00000104 00009d70 00000000 00000041  
3fff1190:  0002d30b 0000003f 0002d30b 00000041  
3fff11a0:  00800000 00000080 3fff34bc 3fff927c  
3fff11b0:  3fffbb20 3fff89dc 3fff89dc 402192dc  
3fff11c0:  00000040 3fff89dc 3fff85c4 3fff89dc  
3fff11d0:  00000040 3fff89dc 3fff85c4 4021a013  
3fff11e0:  3fffad8c 0000003f 7441c2a6 00000001  
3fff11f0:  3fff89dc 3fff51ec 3fff85c4 00000001  
3fff1200:  00000010 3fff89dc 0000000f 4021a268  
3fff1210:  3fff34bc 3fff328c 0000000f 00000001  
3fff1220:  00000001 3fff3534 3fff3534 402191ac  
3fff1230:  00000100 3fff6fdc 3fff3617 00000000  
3fff1240:  00000100 3fff6fdc 3fff3617 4021b399  
3fff1250:  3fff1280 00000000 000000d0 00000030  
3fff1260:  7983451a 3fff50e8 00000006 3fff3611  
3fff1270:  00000004 3fff7074 3fff3611 40216e10  
3fff1280:  d9de0303 24c36380 4dbbc9c4 ac676556  
3fff1290:  39f86a14 3a4857be 33c7b48b e3e574c8  
3fff12a0:  d891f5c4 2c9ff8da a8f6c658 451aa706  
3fff12b0:  00000004 00000004 3fff7074 402172a5  
3fff12c0:  3fff34bc 738d4478 21da0e0e 3fff9044  
3fff12d0:  00000000 600011f0 3fff50e4 3fff9044  
3fff12e0:  00000000 00000004 00000004 402034f0  
3fff12f0:  00000019 00000f59 00000064 40205418  
3fff1300:  0000004a 00000000 00000038 3fff3611  
3fff1310:  3fff3611 00000004 3fff7074 3fff3611  
3fff1320:  00000004 00000004 3fff7074 40216c18  
3fff1330:  0000006e 3fff3611 3fff7074 40216ff4  
3fff1340:  0000001c 3fff3644 3fff7074 401004d8  
3fff1350:  3fff13a0 0000000e 00000010 00000000  
3fff1360:  00000000 3fff334c 3fff7074 01000000  
3fff1370:  3fff13a0 3fff806c 3fff7074 40216d80  
3fff1380:  3fff913c 00000000 3fff7074 00001387  
3fff1390:  000c52fb 3fff806c 3fff3164 4020398d  
3fff13a0:  3fff3164 00000000 3fff9044 40202ffe  
3fff13b0:  3fff9044 00000001 3fff3164 000000c8  
3fff13c0:  000001bb 3fff3164 3fff5e84 000000c8  
3fff13d0:  000001bb 3fff3164 3fff5e84 40203d15  
3fff13e0:  3ffe94f8 41c03ad8 3ffe94f8 41c03ad8  
3fff13f0:  3fff31b8 0000012e 3fff3164 40204460  
3fff1400:  00000001 3fff0190 3fff3164 3ffe8514  
3fff1410:  00000001 3fff0190 3fff3164 40204518  
3fff1420:  3ffe8d0c 3fff0480 3fff03b8 40204944  
3fff1430:  3fff0160 3fff03b8 3fff0174 4020453a  
3fff1440:  3fff0178 3fff03b8 3fff0174 402024a6  
3fff1450:  00000000 00000000 00000000 feefeffe  
3fff1460:  3fff0178 3ffe8514 3fff0164 feefeffe  
3fff1470:  00000000 00000000 00000001 3fff0478  
3fff1480:  3fffdad0 00000000 3fff0470 402053ec  
3fff1490:  feefeffe feefeffe 3fff0480 4010070c  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vf6d232f1
~ld
Free heap before: 36520
unmodified stack   = 3728
```

electronicsguy, 2017-06-06:
Please increase the value of the delay in the loop() to say 6000 and check.

capedra, 2017-06-06:
@electronicsguy Alright. Currently testing.

igrr, 2017-06-06:
Could you please decode the stack dump and paste the result here? Thanks.
https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html#decode

capedra, 2017-06-06:
@igrr Here it goes:
```
Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
Decoding 29 results
0x4020cc16: strstr at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/string/../../../.././newlib/libc/string/strstr.c line 57
0x4020cc16: strstr at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/string/../../../.././newlib/libc/string/strstr.c line 57
0x4021914a: trim at crypto/bigint.c line 672
0x402192dc: bi_clear_cache at crypto/bigint.c line 672
0x4021955d: regular_multiply at crypto/bigint.c line 672
0x402192dc: bi_clear_cache at crypto/bigint.c line 672
0x4021a013: bi_mod_power at crypto/bigint.c line 1348
0x4021a268: precompute_slide_window at crypto/bigint.c line 1323
:  (inlined by) bi_mod_power at crypto/bigint.c line 1372
0x402191ac: comp_right_shift at crypto/bigint.c line 672
0x4021b399: SHA1ProcessMessageBlock at crypto/sha1.c line 132 (discriminator 2)
0x40216e10: send_client_hello at ssl/tls1_clnt.c line 210
:  (inlined by) do_client_connect at ssl/tls1_clnt.c line 167
0x402172a5: process_cert_req at ssl/tls1_clnt.c line 455
:  (inlined by) do_clnt_handshake at ssl/tls1_clnt.c line 135
0x402034f0: ClientContext::_consume(unsigned int) at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 616
:  (inlined by) ClientContext::read(char*, unsigned int) at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WiFi\src\include/ClientContext.h line 253
0x40205418: do_global_ctors at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/core_esp8266_main.cpp line 57
:  (inlined by) init_done() at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/core_esp8266_main.cpp line 152
0x40216c18: basic_read at ssl/tls1.c line 2005
0x40216ff4: ssl_client_new at ssl/tls1_clnt.c line 76
0x401004d8: malloc at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266\umm_malloc/umm_malloc.c line 1664
0x40216d80: send_client_key_xchg at ssl/tls1_clnt.c line 409
0x4020398d: WiFiClientSecure::_connectSSL(char const*) at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 616
0x40202ffe: WiFiClient::connect(IPAddress, unsigned short) at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WiFi\src/WiFiClient.cpp line 324
0x40203d15: HTTPSRedirect::~HTTPSRedirect() at C:\Users\APJ\Documents\Arduino\libraries\HTTPSRedirect/HTTPSRedirect.cpp line 435
0x40204460: HTTPSRedirect::GET(String const&, char const*, bool const&) at C:\Users\APJ\Documents\Arduino\libraries\HTTPSRedirect/HTTPSRedirect.cpp line 435
0x40204518: HTTPSRedirect::POST(String const&, char const*, String const&) at C:\Users\APJ\Documents\Arduino\libraries\HTTPSRedirect/HTTPSRedirect.cpp line 435
0x40204944: Print::println(Printable const&) at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/Print.cpp line 99
0x4020453a: EspClass::deepSleep(unsigned int, RFMode) at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/Esp.cpp line 151
0x402024a6: loop at C:\Users\APJ\Documents\Arduino\HTTPS_redirect_example/HTTPS_redirect_example.ino line 193
0x402053ec: optimistic_yield at ?? line ?
0x4010070c: cont_norm at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/cont.S line 109
```

capedra, 2017-06-06:
@electronicsguy, @igrr Now it's giving me some exception related to malloc after changing the delay to 6000 millis. Here it's the decode of the new exception:
```
Decoding 25 results
0x4010068c: free at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266\umm_malloc/umm_malloc.c line 1733
0x401004d8: malloc at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266\umm_malloc/umm_malloc.c line 1664
0x40205200: operator new(unsigned int) at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/abi.cpp line 84
0x40203bc3: ClientContext::write(unsigned char const*, unsigned int) at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 616
:  (inlined by) __ax_port_write at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 645
0x40216198: send_raw_packet at ssl/tls1.c line 2005
:  (inlined by) send_packet at ssl/tls1.c line 1267
0x40216db8: send_client_key_xchg at ssl/tls1_clnt.c line 419
0x40216258: send_change_cipher_spec at ssl/tls1.c line 2005
0x4021722f: do_clnt_handshake at ssl/tls1_clnt.c line 126 (discriminator 1)
0x4020350c: ClientContext::_consume(unsigned int) at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 616
:  (inlined by) ClientContext::read(char*, unsigned int) at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WiFi\src\include/ClientContext.h line 253
0x40205398: __yield at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/core_esp8266_main.cpp line 57
0x40216b98: do_handshake at ssl/tls1.c line 2005
:  (inlined by) basic_read at ssl/tls1.c line 1481
0x40216f74: do_client_connect at ssl/tls1_clnt.c line 168
0x401004d8: malloc at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266\umm_malloc/umm_malloc.c line 1664
0x40216d00: ssl_read at ssl/tls1.c line 2005
0x402039a9: SSLContext::connect(ClientContext*, char const*, unsigned int) at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 616
:  (inlined by) WiFiClientSecure::_connectSSL(char const*) at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 328
0x40202ffe: WiFiClient::connect(IPAddress, unsigned short) at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WiFi\src/WiFiClient.cpp line 324
0x40203c95: WiFiClientSecure::connect(char const*, unsigned short) at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 320
0x402043e0: HTTPSRedirect::printRedir() at C:\Users\APJ\Documents\Arduino\libraries\HTTPSRedirect/HTTPSRedirect.cpp line 435
0x40204506: HTTPSRedirect::POST(String const&, char const*, String const&, bool const&) at C:\Users\APJ\Documents\Arduino\libraries\HTTPSRedirect/HTTPSRedirect.cpp line 435
0x4010068c: free at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266\umm_malloc/umm_malloc.c line 1733
0x4020452a: HTTPSRedirect::POST(String const&, char const*, String const&) at C:\Users\APJ\Documents\Arduino\libraries\HTTPSRedirect/HTTPSRedirect.cpp line 435
0x40202529: loop at C:\Users\APJ\Documents\Arduino\HTTPS_redirect_example/HTTPS_redirect_example.ino line 204
0x4020536c: loop_wrapper at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/core_esp8266_main.cpp line 57
0x4010070c: cont_norm at C:\Users\APJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/cont.S line 109
```

aj-ptw, 2017-06-19:
I'm having the same error with [OpenBCI Wifi](http://github.com/openbci/OpenBCI_Wifi)

closed 2018-01-02 by igrr

brnyza, 2018-11-13:
Could you inform how can I update?
I'm working with "2.3.0"
C:\Users\b.yuzo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0