esp8266/Arduino #3331: Getting Exception 3 while trying to write to a screen (Nokia 5110)

witnessmenow, 2017-06-04:
### Basic Infos

I'm getting a Exception (3) when i try write text to the Nokia 5110 screen. This screen worked for me a good few months ago when I was on an older version of Arduino IDE (I'm now on 1.8.2) and Esp8266 core.

Does not appear to be a hardware issue as drawing shapes works fine, crash only happens when writing text. I've tried several different sample sketches and they all crash when writing text.

Any ideas how to troubleshoot any further?

#### Hardware
Hardware:			Wemos D1 Mini
Core Version:      	2.4.0 (According to platform.txt - Installed from Github)

### Description

Required Libraries:
https://github.com/bbx10/Adafruit-PCD8544-Nokia-5110-LCD-library/tree/esp8266 (make sure its the esp8266 branch)
Adafruit-GDX from library manager

### Settings in IDE

Module:  			Wemos D1 Mini
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			N/A
Flash Frequency:	N/A
Upload Using:		 SERIAL
Reset Method:		N/A

### Sketch

https://github.com/bbx10/Adafruit-PCD8544-Nokia-5110-LCD-library/blob/esp8266/examples/pcdtest/pcdtest.ino

https://github.com/mcauser/WeMos-D1-mini-Nokia-5110-PCD8544/blob/master/examples/hello-world/hello-world.ino

### Debug Messages

WiFi connected
IP address: 
192.168.1.35

Exception (3):
epc1=0x40203a6e epc2=0x00000000 epc3=0x00000000 excvaddr=0x40236170 depc=0x00000000

ctx: cont 
sp: 3ffefe50 end: 3fff00a0 offset: 01a0

>>>stack>>>
3ffefff0:  3ffef048 0000000a 00000003 402042ca  
3fff0000:  00000080 00000001 3ffe90dd 40204e14  
3fff0010:  00000080 00000000 3ffe8a37 40204e14  
3fff0020:  3ffeee50 0000000c 3ffef048 3ffef074  
3fff0030:  3ffeee50 3ffef048 3ffeee08 40204109  
3fff0040:  3ffe8a37 3ffef048 3ffef048 40204134  
3fff0050:  2301a8c0 00ffffff 3ffeee08 40204179  
3fff0060:  3fffdad0 3ffef048 3ffeee08 40202032  
3fff0070:  3ffe90d8 2301a8c0 feefeffe feefeffe  
3fff0080:  3fffdad0 00000000 3ffef06c 40204678  
3fff0090:  feefeffe feefeffe 3ffef080 4010070c  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

Decoded Stack:

Decoding 9 results
0x402042ca: Print::print(unsigned long, int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 87
0x40204e14: Print::write(unsigned char const*, unsigned int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 87
0x40204e14: Print::write(unsigned char const*, unsigned int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 87
0x40204109: Print::write(char const*) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 87
0x40204134: Print::println() at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 87
0x40204179: Print::print(char const*) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 87
:  (inlined by) Print::println(char const*) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 172
0x40202032: setup at /Users/brian/Documents/Arduino/bitcoin/bitcoin.ino line 165
0x40204678: loop_wrapper at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 57
0x4010070c: cont_norm at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/cont.S line 109

Can't figure out how to get debug messages, even following the page link up top

witnessmenow, 2017-06-04:
More info: The above is on OSX 

I loaded up an old windows install on another computer and that worked fine:

Arduino 1.81
Core Version 2.3.0 (According to platform.txt - Installed from Github)

igrr, 2017-06-06:
`0x40202032: setup at /Users/brian/Documents/Arduino/bitcoin/bitcoin.ino line 165`

What's at line 165 of bitcoin.ino? The example link you have posted doesn't have line 165...

witnessmenow, 2017-06-06:
Sorry about that, here are lines 164 and 165, the first time display.println is used

```
  display.println("  Powered by");
  display.println("   CoinDesk");
```

I went to try do a simple example and I'm pretty confused to whats going on! I'm getting the same error (3), but the stack refers to libraries I am not using in the example (ArduinoJson for example). I'm sure it's the right sketch as the serial monitor shows the appropriate messages

## Code:
```
#include <Arduino.h>

#include <SPI.h>
#include <Adafruit_GFX.h>
// Make sure to install the esp8266 version linked to in the first post
#include <Adafruit_PCD8544.h>

// Pins
const int8_t RST_PIN = D4;
const int8_t CE_PIN = D8;
const int8_t DC_PIN = D6;

Adafruit_PCD8544 display = Adafruit_PCD8544(DC_PIN, CE_PIN, RST_PIN);

void setup() {
  Serial.begin(9600);
  Serial.println("\n\nWeMos D1 Mini + Nokia 5110 PCD8544 84x48 Monochrome LCD\nUsing Adafruit_PCD8544 and Adafruit_GFX libraries\n");

  display.begin();
  display.setContrast(60);  // Adjust for your display
  Serial.println("Show Adafruit logo bitmap");

  // Show the Adafruit logo, which is preloaded into the buffer by their library
  // display.clearDisplay();
  delay(2000);

  display.clearDisplay();
  display.setTextSize(1);
  display.setTextColor(BLACK);
  display.setCursor(0,0);
  display.println("Hello, world!");
  display.display();
  Serial.println("You should now see Hello, world! on the display");
}

void loop() {
}
```
## Serial output:
```
WeMos D1 Mini + Nokia 5110 PCD8544 84x48 Monochrome LCD
Using Adafruit_PCD8544 and Adafruit_GFX libraries

Show Adafruit logo bitmap

Exception (3):
epc1=0x4020259d epc2=0x00000000 epc3=0x00000000 excvaddr=0x4023402c depc=0x00000000

ctx: cont 
sp: 3ffefa70 end: 3ffefd10 offset: 01a0

>>>stack>>>
3ffefc10:  feefeffe feefeffe feefeffe feefeffe  
3ffefc20:  feefeffe 00000000 feefeffe 00000006  
3ffefc30:  feefeffe 00000000 00000001 00000001  
3ffefc40:  feefeffe feefeffe 00000168 00000008  
3ffefc50:  3ffeeafc feefeffe feefeffe feefeffe  
3ffefc60:  feefeffe feefeffe feefeffe 3ffe8ab1  
3ffefc70:  3ffeeaa8 00000000 3ffeeaa8 40202824  
3ffefc80:  00000001 00000001 3ffe8c99 40203660  
3ffefc90:  3ffe84cc 00000000 3ffe8aa4 40203660  
3ffefca0:  40101068 000c1746 3ffeeb24 3ffeece8  
3ffefcb0:  3fffdad0 3ffeecc4 3ffeeaa8 40202fcd  
3ffefcc0:  3ffe8aa4 3ffeece8 40203164 3ffeecf0  
3ffefcd0:  402010b2 000007d0 3ffeeaa8 40203005  
3ffefce0:  3fffdad0 3ffeecc4 3ffeeaa8 40201f0f  
3ffefcf0:  feefeffe 00000000 3ffeece0 402031a8  
3ffefd00:  feefeffe feefeffe 3ffeecf0 4010070c  
<<<stack<<<

```

## Decoded Stack:
```
Decoding 13 results
0x4020319c: Stream at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 332
:  (inlined by) Client at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Client.h line 26
:  (inlined by) WiFiClient::WiFiClient() at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 51
0x40203190: WiFiClient::read(unsigned char*, unsigned int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 332
0x40202824: _GLOBAL__sub_I__ZN16ESP8266WiFiClass9printDiagER5Print at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp line 92
0x40203660: SPIClass::begin() at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/libraries/SPI/SPI.cpp line 515
0x40203660: SPIClass::begin() at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/libraries/SPI/SPI.cpp line 515
0x40101068: RC_GetAckTime at ?? line ?
0x40202fcd: ClientContext::_write_from_source(DataSource*) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 332
0x40203164: ClientContext::read(char*, unsigned int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 332
:  (inlined by) WiFiClient::read(unsigned char*, unsigned int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 235
0x402010b2: uart_write_char_d at ?? line ?
0x40203005: ClientContext::_write_from_source(DataSource*) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 332
0x40201f0f: ArduinoJson::JsonObject::invalid() at /Users/brian/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObject.hpp line 287 (discriminator 1)
0x402031a8: SList ::_add(WiFiClient*) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 332
:  (inlined by) WiFiClient::WiFiClient() at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 53
0x4010070c: cont_norm at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/cont.S line 109

```

witnessmenow, 2017-06-06:
I restarted Arduino and tried again got a better looking decoded stack:

## Serial output:
```
WeMos D1 Mini + Nokia 5110 PCD8544 84x48 Monochrome LCD
Using Adafruit_PCD8544 and Adafruit_GFX libraries

Show Adafruit logo bitmap

Exception (3):
epc1=0x4020259d epc2=0x00000000 epc3=0x00000000 excvaddr=0x4023402c depc=0x00000000

ctx: cont 
sp: 3ffefa70 end: 3ffefd10 offset: 01a0

>>>stack>>>
3ffefc10:  feefeffe feefeffe feefeffe feefeffe  
3ffefc20:  feefeffe 00000000 feefeffe 00000006  
3ffefc30:  feefeffe 00000000 00000001 00000001  
3ffefc40:  feefeffe feefeffe 00000168 00000008  
3ffefc50:  3ffeeafc feefeffe feefeffe feefeffe  
3ffefc60:  feefeffe feefeffe feefeffe 3ffe8ab1  
3ffefc70:  3ffeeaa8 00000000 3ffeeaa8 40202824  
3ffefc80:  00000001 00000001 3ffe8c99 40203660  
3ffefc90:  3ffe84cc 00000000 3ffe8aa4 40203660  
3ffefca0:  40101068 000c1388 3ffeeb24 3ffeece8  
3ffefcb0:  3fffdad0 3ffeecc4 3ffeeaa8 40202fcd  
3ffefcc0:  3ffe8aa4 3ffeece8 40203164 3ffeecf0  
3ffefcd0:  402010b2 000007d0 3ffeeaa8 40203005  
3ffefce0:  3fffdad0 3ffeecc4 3ffeeaa8 40201f0f  
3ffefcf0:  feefeffe 00000000 3ffeece0 402031a8  
3ffefd00:  feefeffe feefeffe 3ffeecf0 4010070c  
<<<stack<<<

```

## Decoded Stack:
```
Decoding 11 results
0x40202824: Adafruit_GFX::write(unsigned char) at /Users/brian/Documents/Arduino/libraries/Adafruit_GFX_Library/Adafruit_GFX.cpp line 625
0x40203660: Print::write(unsigned char const*, unsigned int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 38
0x40203660: Print::write(unsigned char const*, unsigned int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 38
0x40101068: ets_timer_arm_new at ?? line ?
0x40202fcd: Print::write(char const*) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 169
0x40203164: esp_yield at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 57
0x402010b2: delay at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring.c line 54
0x40203005: Print::print(char const*) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 106
:  (inlined by) Print::println(char const*) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 172
0x40201f0f: setup at /Users/brian/Documents/Arduino/test/test.ino line 33
0x402031a8: loop_wrapper at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 57
0x4010070c: cont_norm at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/cont.S line 109

```

igrr, 2017-06-06:
Looks similar to https://github.com/esp8266/Arduino/issues/3140#issuecomment-304667491. This was fixed in #3325 yesterday. Can you try again using the latest git version?

witnessmenow, 2017-06-06:
> Looks similar to #3140 (comment)

Sure, will give it a try after work

witnessmenow, 2017-06-06:
@igrr that looks to be working for me now, thanks a lot for your help!

witnessmenow, 2017-06-06:
Sorry... so I started on the next part of my project and I am getting a Exception 3 when I try connect to coinmarketcap.com's api using WiFiClientSecure. Should I create a new issue for this? Details below:

## Sketch:
https://github.com/witnessmenow/arduino-coinmarketcap-api/blob/master/examples/ESP8266/GetTickerInfo/GetTickerInfo.ino

Last part of code of mine before the exception is:
https://github.com/witnessmenow/arduino-coinmarketcap-api/blob/master/src/CoinMarketCapApi.cpp#L15
(the serial print below this line does not get hit if its not commented out)

## Serial Output
```
 1384, room 16 
tail 8
chksum scandone
Connecting Wifi: Mikrotik
.....scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 8
cnt 

connected with Mikrotik, channel 1
dhcp client start...
.ip:192.168.1.35,mask:255.255.255.0,gw:192.168.1.1

WiFi connected
IP address: 
192.168.1.35
Getting ticker data for bitcoin
:ref 1
please start sntp first !
State:	sending Client Hello (1)
:wr 108 108 0
:wrc 108 108 0
:sent 108
:rn 1440
:rd 5, 1440, 0
:rdi 1440, 5
:rd 80, 1440, 5
:rdi 1435, 80
State:	receiving Server Hello (2)
:rd 5, 1440, 85
:rdi 1355, 5
:rd 1350, 1440, 90
:rdi 1350, 1350
:c0 1350, 1440
:rn 1440
:rd 1440, 1440, 0
:rdi 1440, 1440
:c0 1440, 1440
:rn 1440
:rd 1440, 1440, 0
:rdi 1440, 1440
:c0 1440, 1440
:rn 1252
:rd 1243, 1252, 0
:rdi 1252, 1243
State:	receiving Certificate (11)
Fatal exception 3(LoadStoreErrorCause):
epc1=0x4020ce03, epc2=0x00000000, epc3=0x00000000, excvaddr=0x40244450, depc=0x00000000

Exception (3):
epc1=0x4020ce03 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40244450 depc=0x00000000

ctx: cont 
sp: 3fff0a00 end: 3fff1360 offset: 01a0

>>>stack>>>
[... 152 lines cut ...]
0x40216dcc: ssl_read at ssl/tls1.c line 2005
0x40204a21: SSLContext::connect(ClientContext*, char const*, unsigned int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp line 615
:  (inlined by) WiFiClientSecure::_connectSSL(char const*) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp line 327
0x40204116: WiFiClient::connect(IPAddress, unsigned short) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 324
0x40204ba5: WiFiClientSecure::connect(char const*, unsigned short) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp line 319
0x402022a5: CoinMarketCapApi::SendGetToCoinMarketCap(String) at /Users/brian/Documents/Arduino/libraries/arduino-coinmarketcap-api/src/CoinMarketCapApi.cpp line 15
0x402052b3: String::changeBuffer(unsigned int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 439 (discriminator 4)
0x40205331: String::copy(char const*, unsigned int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 439 (discriminator 4)
0x402053f0: String::operator=(String const&) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 439 (discriminator 4)
0x40203538: CoinMarketCapApi::GetTickerInfo(String, String) at /Users/brian/Documents/Arduino/libraries/arduino-coinmarketcap-api/src/CoinMarketCapApi.cpp line 71
0x402052b3: String::changeBuffer(unsigned int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 439 (discriminator 4)
0x40205331: String::copy(char const*, unsigned int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 439 (discriminator 4)
0x4020537e: String::String(char const*) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 439 (discriminator 4)
0x4020209a: printTickerData(String) at /var/folders/sl/1b3tnsqj3t9301xzfkh9g1900000gn/T/arduino_modified_sketch_875138/GetTickerInfo.ino line 51
0x4010053d: _umm_realloc at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1491
:  (inlined by) realloc at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1709
0x40204f9e: Print::print(unsigned long, int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 99
0x40201900: uart_start_isr at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/uart.c line 461
0x402052b3: String::changeBuffer(unsigned int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 439 (discriminator 4)
0x402052ff: String::reserve(unsigned int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 439 (discriminator 4)
0x40205331: String::copy(char const*, unsigned int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 439 (discriminator 4)
0x40202161: loop at /var/folders/sl/1b3tnsqj3t9301xzfkh9g1900000gn/T/arduino_modified_sketch_875138/GetTickerInfo.ino line 94
0x402056e8: loop_wrapper at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 57
0x4010070c: cont_norm at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/cont.S line 109
```

witnessmenow, 2017-06-07:
Going to close this one as the issue in the title is fixed.

New issue opened at #3335

closed 2017-06-07 by witnessmenow